gokrazy/gokrazy #2: smsc95xx: accept Raspberry Pi bootloader’s MAC address

stapelberg, 2017-02-04:


stapelberg, 2017-02-04:
A patch to fix this has been rejected upstream 7 years ago:
http://netdev.vger.kernel.narkive.com/Fi2roLel/patch-smsc95xx-add-module-parameter-to-override-mac-address

Here’s a patch against linux 4.9.7:

```diff
diff --git i/drivers/net/usb/smsc95xx.c w/drivers/net/usb/smsc95xx.c
index 831aa33..68a5321 100644
--- i/drivers/net/usb/smsc95xx.c
+++ w/drivers/net/usb/smsc95xx.c
@@ -62,6 +62,7 @@
 					 SUSPEND_SUSPEND2 | SUSPEND_SUSPEND3)
 
 #define CARRIER_CHECK_DELAY (2 * HZ)
+#define MAC_ADDR_LEN                    (6)
 
 struct smsc95xx_priv {
 	u32 chip_id;
@@ -82,6 +83,10 @@ static bool turbo_mode = true;
 module_param(turbo_mode, bool, 0644);
 MODULE_PARM_DESC(turbo_mode, "Enable multiple frames per Rx transaction");
 
+static char *macaddr = ":";
+module_param(macaddr, charp, 0);
+MODULE_PARM_DESC(macaddr, "MAC address");
+
 static int __must_check __smsc95xx_read_reg(struct usbnet *dev, u32 index,
 					    u32 *data, int in_pm)
 {
@@ -910,6 +915,53 @@ static int smsc95xx_ioctl(struct net_device *netdev, struct ifreq *rq, int cmd)
 	return generic_mii_ioctl(&dev->mii, if_mii(rq), cmd, NULL);
 }
 
+/* Check the macaddr module parameter for a MAC address */
+static int smsc95xx_is_macaddr_param(struct usbnet *dev, u8 *dev_mac)
+{
+       int i, j, got_num, num;
+       u8 mtbl[MAC_ADDR_LEN];
+
+       if (macaddr[0] == ':')
+               return 0;
+
+       i = 0;
+       j = 0;
+       num = 0;
+       got_num = 0;
+       while (j < MAC_ADDR_LEN) {
+               if (macaddr[i] && macaddr[i] != ':') {
+                       got_num++;
+                       if ('0' <= macaddr[i] && macaddr[i] <= '9')
+                               num = num * 16 + macaddr[i] - '0';
+                       else if ('A' <= macaddr[i] && macaddr[i] <= 'F')
+                               num = num * 16 + 10 + macaddr[i] - 'A';
+                       else if ('a' <= macaddr[i] && macaddr[i] <= 'f')
+                               num = num * 16 + 10 + macaddr[i] - 'a';
+                       else
+                               break;
+                       i++;
+               } else if (got_num == 2) {
+                       mtbl[j++] = (u8) num;
+                       num = 0;
+                       got_num = 0;
+                       i++;
+               } else {
+                       break;
+               }
+       }
+
+       if (j == MAC_ADDR_LEN) {
+               netif_dbg(dev, ifup, dev->net, "Overriding MAC address with: "
+               "%02x:%02x:%02x:%02x:%02x:%02x\n", mtbl[0], mtbl[1], mtbl[2],
+                                               mtbl[3], mtbl[4], mtbl[5]);
+               for (i = 0; i < MAC_ADDR_LEN; i++)
+                       dev_mac[i] = mtbl[i];
+               return 1;
+       } else {
+               return 0;
+       }
+}
+
 static void smsc95xx_init_mac_address(struct usbnet *dev)
 {
 	const u8 *mac_addr;
@@ -931,6 +983,10 @@ static void smsc95xx_init_mac_address(struct usbnet *dev)
 		}
 	}
 
+	/* Check module parameters */
+	if (smsc95xx_is_macaddr_param(dev, dev->net->dev_addr))
+		return;
+
 	/* no useful static MAC address found. generate a random one */
 	eth_hw_addr_random(dev->net);
 	netif_dbg(dev, ifup, dev->net, "MAC address set to eth_random_addr\n");

```

stapelberg, 2017-03-04:
As per https://github.com/raspberrypi/firmware/issues/613, we should upgrade the firmware, then drop the kernel patch.

massar, 2017-03-13:
Michael++++ :)

Thanks for pushing, having just received two RPi3's, I run into this too.

Your image at https://wiki.debian.org/RaspberryPi3 boots fine.

I'll be looking into using GoKrazy for my needs too, but first need to experiment with the toys that I got which needs a bit bigger platform and debuggability, but for deployment GoKrazy looks like the way to go.

stapelberg, 2017-03-17:
firmware was updated in https://github.com/gokrazy/firmware/commit/ff326c20c58ee276801e18a2559c256530b74564

closed 2017-03-17 by stapelberg