gokrazy/gokrazy #7: authentication and transport layer security

stapelberg, 2017-02-07:
This issue describes which authentication mechanisms and transport layer security configurations we’d like to support.

num | TLS key source | listening IP | authentication | decision
-------|-------------------|----------------|--------------------|--------
1 | no TLS | private IPs | password | default
2 | self-signed | private+public | password | using `-tls=self-signed`
3 | self-signed | private+public | TLS client cert | _if there’s interest_
4 | user-supplied | private+public | password | using `-tls=<path>[,<path>]`
5 | user-supplied | private+public | TLS client cert | _if there’s interest_
6 | Let’s Encrypt | private+public | password | _if there’s interest_

details:

* for all password authentication scenarios, the password which is stored in `~/.config/gokrazy` the first time an image is created
* in scenario ①, we make the assumption that TLS is not necessary for wired connections (low risk of eavesdropping) or wireless connections (the wifi encryption is good enough)
* in scenario ②, users need to add the self-signed certificate to their browser (we should have instructions) (skilled users could click away the security warning when the risk of MITM is low, but we shouldn’t encourage that behavior)
* in scenario ④, we load combined PEM files, or keys/certs from separate files

discarded approaches:
* adding a self-signed certificate to the system certificate store (`/etc/ssl/certificates`) and/or browser-specific certificate stores (using `certutil` for chrome and firefox): browsers don’t pick up changes unless you cause them via their UI (see e.g. [cr issue 489956#c18](https://bugs.chromium.org/p/chromium/issues/detail?id=489956#c18)). Notably, on Fedora, this works better: https://bugs.launchpad.net/ubuntu/+source/nss/+bug/1647285 — possibly we can revisit this approach in the future.
* https://git.daplie.com/Daplie/localhost.daplie.me-certificates or similar (publicly available key material)

janisstreib, 2019-12-05:
Hi,
I'm currently working on this feature and implemented ② and ④ (proof of concept).
I decided to store the generated self-signed certificates in the gokrazy config folder and use those in the gokr-packer/updater (with SSL stripping detection).
I therefore introduced host-specific folders in the gokrazy config folder, which may also allow to store host-specific passwords in the future.
My code still needs some testing but I will PR it as soon as possible.

stapelberg, 2019-12-10:
Thanks for letting us know!

Out of curiosity, can I ask what your motivation is for implementing this? Are you doing updates remotely? In which setup are you using gokrazy? :)

janisstreib, 2019-12-11:
I'm planning on using it on multiple PIs (about 20) for laundry controllers (payment) in student dormitories across the city (currently evaluating/developing hardware and software) with a central "controller", which manages API access, metrics (like machine status or washing time), updates, etc.
Instead of implementing own update and debug mechanisms, using the existing gokrazy userland might be a quite nice option :)

stapelberg, 2019-12-11:
Sounds very cool :)

Looking forward to your PR!

hazcod, 2020-09-29:
Interesting, how would we push the keypair to the gokrazy firmware?

stapelberg, 2020-09-29:
The packer includes the necessary files. See https://gokrazy.org/userguide/tls-for-untrusted-networks/

stapelberg, 2020-09-30:
In fact, this whole issue should be closed as we now have enough TLS support: https://gokrazy.org/userguide/tls-for-untrusted-networks/ :)

closed 2020-09-30 by stapelberg

hazcod, 2020-09-30:
Except for the authentication part I presume @stapelberg ?

stapelberg, 2020-10-01:
Yeah, we don’t have TLS client certificate authentication right now, but I’m also not convinced yet that it’s required.

We support user-provided certificates, and the HTTP auth password that gokrazy requires (independent of TLS) should be good enough for typical gokrazy use-cases.

hazcod, 2020-10-01:
Can only agree 👍