gokrazy/gokrazy #10: Implement SquashFS support

stapelberg, 2017-02-11:
Currently, we use a FAT file system for both boot (required) and root (convenient, because we already have an implementation).

FAT comes with a few disadvantages when used as a root file system: file names are truncated to 8.3 characters and symlinks are not supported.

The most promising candidate for a simple root file system seems to be SquashFS.

joeblew99, 2017-03-16:
thats a great idea.
i had heaps of SD cards dying, and used SquashFS in the past and it worked

joeblew99, 2017-03-22:
hey @stapelberg 
The rpi3 has a new boot mechanism so that it can boot from storage other than the SD Card.
I am wondering if you plan is to make SquashFS work on the USB stick or the SD card.

stapelberg, 2017-03-22:
The type of file system doesn’t matter for selecting the boot device, so your question is tangential to this issue. That said, there’s no plan to support any other boot medium than an SD card.

joeblew99, 2017-03-22:
ok. SO if the SD card has squash FS on it, where to store non immutable data ? For example, data that is generated from the App, like a database etc ?

stapelberg, 2017-03-22:
The fourth partition is used (now and in the future) for the permanent file system.

The documentation currently contains examples which use `mkfs.ext4` to make that an ext4 file system, but you are free to format it with any file system which the Linux kernel from the https://github.com/gokrazy/kernel repository supports.

If you need to use a file system which is contained in the mainline Linux kernel but not enabled in our config, please file an issue on the kernel repository and we can enable it.

joeblew99, 2017-03-22:
thanks -  makes sense. 
ext looks good for me.

thanks. I will try it out..

stapelberg, 2018-06-21:
This is done with https://github.com/gokrazy/tools/commit/1450d301b4476a99387dde2c08ebadf9c2a243c4

closed 2018-06-21 by stapelberg