gokrazy/gokrazy #13: Implement wifi setup

phifty, 2017-03-04:
It might a good thing, if a wifi network can be configured. The SSID and password would be used at boottime to connect to the local WLAN.

stapelberg, 2017-03-04:
Thanks for the suggestion.

I considered this but decided against it, at least for the initial release. A WPA supplicant seems fairly complicated and intricate. Just recently, [a security vulnerability](https://ftp.openbsd.org/pub/OpenBSD/patches/6.0/common/018_net80211.patch.sig) was found in the OpenBSD wifi implementation.

Hence, personally, I will not contribute a supplicant implementation and will just use wired ethernet for the time being.

I haven’t yet made up my mind as to whether a contributed supplicant implementation would be merged or not — perhaps we’d need to look at an actual implementation to better understand the complexity involved. My suspicion is that the complexity will be rather high (compared to the rest of the project code).

dangerhuss, 2017-03-12:
I was hoping to use this with a Raspberry Pi Zero W. I don't that will be possible without WiFi.

Drache93, 2017-03-17:
I'd recommend looking into DietPi for their implementation of WiFi configuration on install if you decide to attempt it. Personally, don't know anything on the security side (except ease of password accessibility) - but it works. 
I might look into it myself, as your project matches an idea i was thinking about for easily deploying to robots. Robots don't need all the OS crap everything else comes with!! :D

fd0, 2017-04-16:
Supporting wifi is (at least for me) very desirable, I have various Pi3 in the house that I'd like to use with gokrazy. Due to some rather remote locations, most of them are connected via wifi.

What about statically compiling wpa_supplicant and interface it with Go for the time being?

Merovius, 2017-04-16:
Having C-Software (statically compiled or not) exposed to the network seems very much opposed to the goals of gokrazy (specifically to get rid of unsafe C-code as much as possible).

stapelberg, 2017-04-16:
@Merovius captured the philosophical part pretty well.

@fd0, you can of course implement your suggestion either locally or even publish it in a separate repository — technically speaking, you’d just need to bundle the statically compiled wpa_supplicant, then write it to a file in /tmp and exec it. Of course, you’d be responsible for updating and maintaining that setup, then.

kaofishy, 2017-07-03:
Have you looked into directly using the nl80211 interface from the Linux kernel and bypassing wpa_supplicant? I've done some search and these projects seem promising and actively maintained:

https://github.com/hkwi/nlgo
https://github.com/mdlayher/wifi

stapelberg, 2017-07-03:
Thanks for the pointers. The wifi package you referred to does not seem to implement authentication, which is the biggest pain point with regards to wifi support. To be clear: I wasn’t saying that we need `wpa_supplicant(8)`, I was saying that we need a program which acts as a WPA supplicant.

One could argue that access to open wifi networks would be a start, and I’d agree. However, users who’d be interested in using open wifi networks will need to set up some alternative way of authentication than gokrazy’s default internal-IPs-are-trusted model (e.g. TLS-based authentication).

fd0, 2017-07-03:
I've had a quick look at the packages a few weeks ago, but then abandoned the project due to the high complexity. Having a WPA/WPA2 implementation (at least for pre-shared keys) in Go would be very desirable, but looks like a huge undertaking.

kaofishy, 2017-07-03:
Right, I've spent a few hours looking into this and apparently the BCM43438 on the RPi3 supports hardware WPA/WPA2 authentication at the hardware level, but it seems no one has ever tried using it without wpa_supplicant. I guess the "best" way to go is to use breakglass and manually invoke wpa_supplicant.

stapelberg, 2017-07-04:
Thanks for the investigation! I didn’t realize the hardware supports this directly. Maybe we could get some sample code from Broadcom or the Raspberry Pi foundation? My assumption is that instructing the hardware to join an encrypted network should be way more feasible than implementing a supplicant ourselves.

kaofishy, 2017-07-04:
It's actually a rather complicated SoC containing a Cortex-M3 and some unknown microcontroller. Unlike Broadcom, Cypress seems much more generous with releasing data sheets. http://www.cypress.com/file/298076/download

kaofishy, 2017-07-31:
Whelp I guess this is why we shouldn't trust binary blobs
https://arstechnica.com/information-technology/2017/07/broadcom-chip-bug-opened-1-billion-phones-to-a-wi-fi-hopping-worm-attack/

quite, 2018-04-13:
There is also the project to replace wpa_supplicant. Still in C, to be sure. https://git.kernel.org/pub/scm/network/wireless/iwd.git/

zet4, 2018-05-29:
With the recent release of Raspberry Pi3 B+ which has an improved wifi chipset I would really love to see wifi support in gokrazy! 
As well as hopefully native version of hostapd so one could turn it into a travel router of sorts.

bzub, 2018-11-30:
I'm working on a containerized solution for wifi. It's not pure-go, but hopefully a nice compromise that exposes as little non-go to the host as possible.

I'd like to get feedback on whether to use wpa_supplicant or iwd?

@kaofishy can you elaborate on what advantages wpa_supplicant has with the Raspberry Pi hardware? I've been running iwd on a pi zero W without issue for a while but haven't done anything demanding. wpa_supplicant also has a familiar config file going for it. Personally I've just enjoyed how easy it is to configure iwd and there's well maintained [golang bindings](https://godoc.org/github.com/amenzhinsky/iwd). In either case the wifi container would probably need dbus running.

stapelberg, 2020-05-27:
For people who only subscribe to this issue and haven’t seen the other one:

In https://github.com/gokrazy/gokrazy/issues/57, I have implemented WiFi support for unencrypted networks. See that issue for details and rationale. I hope this is sufficient for a number of people (perhaps together with the upcoming TLS support, see https://github.com/gokrazy/gokrazy/pull/53), or at least provides a good stepping stone for further exploration.

I also looked into using https://tailscale.com/ on gokrazy, and aside from a few minor stumbling blocks, their use of iproute2 and iptables is what doesn’t work on gokrazy: https://github.com/tailscale/tailscale/issues/77 and https://github.com/tailscale/tailscale/issues/391

This is too bad, as securing a gokrazy Raspberry Pi using tailscale would be a pretty neat option for people. I suppose you can always configure your own WireGuard tunnel.

stapelberg, 2020-06-01:
FYI: gokrazy now supports TLS for the web interface and updates, see https://gokrazy.org/userguide/tls-for-untrusted-networks/.

This makes using unencrypted WiFi networks (see https://gokrazy.org/userguide/unencrypted-wifi/) more attractive.

JustinAzoff, 2022-01-23:
WPA PSK support might not be that difficult depending on which interface is used.  the nl80211 api looks pretty complicated, but (older? deprecated?) wext is very simple:

```
2499 hostap/src/drivers/driver_wext.c
 12255 hostap/src/drivers/driver_nl80211.c
```

If I'm reading [this](https://w1.fi/cgit/hostap/tree/src/drivers/driver_wext.c#n2107) right, turning on PSK and connecting to an SSID involves using SIOCSIWESSID,SIOCSIWAUTH, and  SIOCSIWENCODEEXT to set ssid,auth mode, psk.

the `driver_flags` wpa_cli output on my pi zero shows:

```
4WAY_HANDSHAKE_8021X
4WAY_HANDSHAKE_PSK
```

which is the hardware offloading I believe.

Downside is that using wext at this point in time might not be desirable, and it's potentially incompatible with the netlink bits you have now.

stapelberg, 2022-01-26:
Whether (or how long) wext will stick around is a question for the Linux wi-fi maintainers I suppose :)

I have no clue if the code you linked to is sufficient to join an encrypted network. Perhaps a proof-of-concept implementation would be useful to figure out the best next steps?

JustinAzoff, 2022-01-26:
Agreed on both counts.  I'll see if I can extract just the ioctl bits from the wext driver into a small .c program that lets you do `./wpa ssid password`.  If that works translating it to go (or using cgo?) would be simple enough.

JustinAzoff, 2022-01-27:
I did some testing using alpine 3.15 which has kernel 5.15.4.. Extracting the wext bits was easy, but it turns out that doesn't seem to work at all anymore on the pi.

my standalone c program fails to run `SIOCSIWENCODEEXT`.  All the ioctls for setting mode, ssid, cipher bits work, until it tries to set the PSK.

wpa_supplicant fails in a similar way when told to use `wext`

```
# wpa_supplicant  -D wext -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf 
Successfully initialized wpa_supplicant
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument```

So it seems like the `wext` approach might be a non-starter.  I will see if I can figure out what the extra 10K lines of code in driver_nl80211.c do, and what a minimal association program would look like.

stapelberg, 2022-01-28:
Huh. Can you check with Raspbian if wext works in principle on the hardware?

JustinAzoff, 2022-01-28:
Sure, that's easy enough to test. will just have to reimage the card.

JustinAzoff, 2022-02-07:
Haven't had a chance to try wext again, but I've been reviewing how driver_nl80211.c works, and I think the netlink api might not be too bad.  There's a ton of code in there for things like frequencies, channels, and AP mode that aren't immediately relevant, although I could see AP mode being useful to someone.

The important bits seem to be in `nl80211_connect_common` and connecting using a PSK looks easy enough.  For simple ssid+psk with offload it builds up a netlink message like this:

```
nla_put(msg, NL80211_ATTR_SSID, params->ssid_len, params->ssid);

//ver NL80211_WPA_VERSION_1 or 2
nla_put_u32(msg, NL80211_ATTR_WPA_VERSIONS, ver);

nla_put_u32(msg, NL80211_ATTR_AKM_SUITES, RSN_AUTH_KEY_MGMT_PSK_SHA256);

nla_put_flag(msg, NL80211_ATTR_WANT_1X_4WAY_HS);

//psk is the 32 byte psk, like wpa_passphrase returns.
nla_put(msg, NL80211_ATTR_PMK, 32, params->psk);
```

and then sends that across.  The wifi bits you added here  https://github.com/mdlayher/wifi/blob/84f0b9457fdd566bb03cb33b6dd0dadeacca614f/client_linux.go#L96 are the same thing, it just only sets the ssid attr.  Adding the extra attributes might be enough to get WPA with PSK to work.  I'll see if I can get this to work

Probably need to rethink the connect api, but 

```
Connect(ifi *Interface, ssid string, psk string)
```

will work for testing

stapelberg, 2022-02-07:
Nice! 👍

Thanks for looking into this :)

mpictor, 2022-02-20:
Cool! Was looking into this myself. I tried to `strace` wpa_supplicant while disconnecting/reconnecting, and extract the netlink messages.  

Sadly, [strace's output format has changed](https://docs.pyroute2.org/debug.html#strace-compatibility-note). It no longer produces output that can be decoded by pyroute2.

I tried tcpdump, but wasn't able to create the requisite network device - it seems that the raspbian kernel is missing one or more necessary options.

At this point, I decided it might be easier to read through the C code than to find the netlink messages. Doing some searches, just in case I'd missed a go wifi/wpa implementation, led me to this discussion.

----

FWIW I have managed to modify `golang-evdev` to rid it of cgo. With that, game controllers can be used with gokrazy - for example to control an rc car or self-propelled server. The code's pretty ugly at the moment; I need to clean it up and do a PR.

stapelberg, 2022-02-20:
> Sadly, [strace's output format has changed](https://docs.pyroute2.org/debug.html#strace-compatibility-note). It no longer produces output that can be decoded by pyroute2.

I have noticed this as well when working with https://github.com/google/nftables and reported it as a bug at https://github.com/strace/strace/issues/100. Just build an older strace version for now :)

JustinAzoff, 2022-02-20:
Ah, that is interesting about strace.  Where I last left off was:

 * Updated the `wifi` package code-gen and headers so it has all the new enums.  I can get a MR going for that project.
 * Updated the Connect method to send the additional netlink messages.

Unfortunately, my attempts to include what I thought were the minimal amount of netlink messages to get WPA going do not work.  I can add more things, starting with the BSSID, but would be nice to fully understand what wpa_supplicant is doing.

I did the strace capture of wpa_supplicant and tried to decode it using pyroute like https://jvns.ca/blog/2017/09/03/debugging-netlink-requests/ describes, but that is working.  I'm not sure if it's for the same reason you mention though, the strace I used must be the older version already.  From what I can tell the message it sends is only 176 bytes, so this can't be that complicated.  I can see where the raw data includes things like the BSSID of the AP, the SSID, and the 32 byte key that `wpa_passphrase` returns.

```
$ python decoder.py pyroute2.netlink.nl80211.nl80211cmd msg.associate 
[lots of HEX] {'attrs': [('NL80211_ATTR_IE', '[lots of HEX]')],
 'cmd': 4,
 'header': {'flags': 3,
            'length': 33,
            'pid': 2150432776,
            'sequence_number': 3,
            'type': 8},
 'reserved': 1,
 'version': 0}
........................................
Traceback (most recent call last):
  File "/tmp/env/lib/python3.9/site-packages/pr2modules/netlink/__init__.py", line 1411, in ft_decode
    self.decode_nlas(offset)
  File "/tmp/env/lib/python3.9/site-packages/pr2modules/netlink/__init__.py", line 1341, in decode_nlas
    (length, base_msg_type) = struct.unpack_from('HH', self.data,
struct.error: unpack_from requires a buffer of at least 64720 bytes for unpacking 4 bytes at offset 64716 (actual buffer size is 119)

Traceback (most recent call last):
  File "/tmp/env/lib/python3.9/site-packages/pr2modules/netlink/__init__.py", line 1411, in ft_decode
    self.decode_nlas(offset)
  File "/tmp/env/lib/python3.9/site-packages/pr2modules/netlink/__init__.py", line 1341, in decode_nlas
    (length, base_msg_type) = struct.unpack_from('HH', self.data,
struct.error: unpack_from requires a buffer of at least 64720 bytes for unpacking 4 bytes at offset 64716 (actual buffer size is 119)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "decoder.py", line 37, in <module>
    msg.decode()
  File "/tmp/env/lib/python3.9/site-packages/pr2modules/netlink/__init__.py", line 1011, in decode
    self.ft_decode(offset)
  File "/tmp/env/lib/python3.9/site-packages/pr2modules/netlink/__init__.py", line 1414, in ft_decode
    raise NetlinkNLADecodeError(e)
pr2modules.netlink.exceptions.NetlinkNLADecodeError: unpack_from requires a buffer of at least 64720 bytes for unpacking 4 bytes at offset 64716 (actual buffer size is 119)

```

Possibly pyroute2.netlink.nl80211.nl80211cmd is the wrong message type to be using here?

JustinAzoff, 2022-02-20:
Ok, I sent over the MR to update the `wifi` package here: https://github.com/mdlayher/wifi/pull/34

It doesn't have support for wpa yet, but does have all the new enum values required, once the netlink message is worked out

JustinAzoff, 2022-02-20:
I got it to work!

I ran_wpa supplicant in debugging mode, which output this:

```
  * WPA Versions 0x2
  * pairwise=0xfac04
  * group=0xfac04
  * akm=0xfac02
```

Those values were similar, but slightly different from the ones I was trying to use.  The values I was using would just return invalid argument or would be accepted but fail to work.

And those values are:

```
#define RSN_AUTH_KEY_MGMT_FT_PSK RSN_SELECTOR(0x00, 0x0f, 0xac, 4)
#define RSN_AUTH_KEY_MGMT_PSK_OVER_802_1X RSN_SELECTOR(0x00, 0x0f, 0xac, 2)
```
which makes sense, I was trying to use RSN_AUTH_KEY_MGMT_PSK_SHA256 and RSN_CIPHER_SUITE_CCMP_256 which I guess are wrong.

It also did not work with `nl80211.WpaVersion1 | nl80211.WpaVersion2`. 

so, anyway with Connect amended to be this:

```
func (c *client) Connect(ifi *Interface, ssid string, psk string) error {
        // Ask nl80211 to connect to the specified SSID.
        b, err := netlink.MarshalAttributes(
                []netlink.Attribute{
                        {
                                Type: nl80211.AttrIfindex,
                                Data: nlenc.Uint32Bytes(uint32(ifi.Index)),
                        },
                        {
                                Type: nl80211.AttrSsid,
                                Data: []byte(ssid),
                        },
                        {
                                Type: nl80211.AttrWpaVersions,
                                Data: nlenc.Uint32Bytes(nl80211.WpaVersion2),
                        },
                        {
                                Type: nl80211.AttrCipherSuiteGroup,
                                Data: nlenc.Uint32Bytes(uint32(0xfac04)),
                        },
                        {
                                Type: nl80211.AttrCipherSuitesPairwise,
                                Data: nlenc.Uint32Bytes(uint32(0xfac04)),
                        },
                        {
                                Type: nl80211.AttrAkmSuites,
                                Data: nlenc.Uint32Bytes(uint32(0xfac02)),
                        },
                        {
                                Type: nl80211.AttrWant1x4wayHs,
                                Data: nil,
                        },
                        {
                                Type: nl80211.AttrPmk,
                                Data: []byte(psk),
                        },
                })
...
```

It immediately connects to my home network.

Next steps I need to think about what the actual interface should be.  I think adding a new method ConnectWPA is the way to go, so the current api won't break.  I also need to find a `go` implementation of whatever it is that `wpa_passphrase` does to turn the ssid+key into the 32 byte psk, but having to hardcode the psk wouldn't be the worst problem.

JustinAzoff, 2022-02-20:
ok! well that was easy, turns out it's just

```pbkdf2.Key([]byte(psk), []byte(ssid), 4096, 32, sha1.New)```

I have a branch https://github.com/JustinAzoff/wifi/tree/wpa-psk that might work well enough for testing.  It adds a `c.ConnectWPAPSK` function that takes ssid + psk.

I'll see if I can get an image built and running on my pi using this branch.

JustinAzoff, 2022-02-20:
I'm having some trouble getting my 3b to come up properly, but it looks like a wifi firmware issue.

If I use the latest kernel package, I see this in the kernel log

```
6,276,4173563,-;brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
4,277,4182737,-;brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.bin failed with error -2
4,278,4194858,-;brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2
```

but it doesn't say "Falling back to sysfs fallback" and nothing ever shows up in /sys/class/firmware/.

If I use the old v0.0.0-20200526220649-e7d397b3cf44 kernel it does request those files via /sys, but after adding them I just get

```
3,299,9619524,-;brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

stapelberg, 2022-02-20:
You can use the extrafiles mechanism of the packer (https://github.com/gokrazy/tools/blob/45e6b6fc6c3fdb4391cf80239e588a28a0de8d6d/cmd/gokr-packer/packer.go#L355, works like all the other per-package config described at https://gokrazy.org/userguide/package-config/) to include the firmware files in your gokrazy root file system.

I’ll add proper documentation to the package-config guide about extrafiles soon.

JustinAzoff, 2022-02-21:
It works!

I copied all the firmware from another pi to extrafiles/github.com/gokrazy/wifi/lib/firmware/brcm/ and rebuilt the image, it came right up!

Side note, what really helped figure this out was a simple service that did

```
func main() {
        f, err := os.Open("/dev/kmsg")
        if err != nil {
                log.Fatal("Failed to open: %v", err)
        }
        _, err = io.Copy(os.Stderr, f)
        if err != nil {
                log.Fatal("Failed to read: %v", err)
        }
}
```
wasn't pretty, but made it super easy to view the kernel log without having to setup ssh/breakglass.

So,

https://github.com/justinazoff/wifi/tree/wpa-psk has the updates for basic WPA PSK support

https://github.com/justinazoff/wifi-wpa is the fork of gokrazy/wifi updated to call ConnectWPAPSK

The AttrCipherSuiteGroup/AttrCipherSuitesPairwise stuff that's currently hardcoded will likely have to be fixed, but I'd be interested in knowing if this doesn't already work for most people.

mpictor, 2022-02-22:
I was going to suggest using go-bindata rather than goembed, as the version of it I used (some time ago) didn't give linters fits and didn't create files with long lines or lots of escape sequences. Unfortunately, unless you wish to clone [a very old version](https://github.com/jteeuwen/go-bindata/commit/212d2a5cdcb78d5) you still get long lines with escapes.

------
The service restarts many times before loading all firmwares - not sure if this is expected. 
```cfg
  # ...
  2022/02/21 19:13:44 gokrazy: attempt 59, starting ["/user/wifi"]
  2022/02/21 19:13:44 no interfaces found
  2022/02/21 19:13:44 gokrazy: exit status 1
  2022/02/21 19:13:45 gokrazy: attempt 60, starting ["/user/wifi"]
  2022/02/21 19:13:45 feeding firmware (2074 bytes) to brcmfmac via sysfs/brcm!brcmfmac43455-sdio.txt
  2022/02/21 19:13:45 no interfaces found
  2022/02/21 19:13:45 gokrazy: exit status 1
  2022/02/21 19:13:46 gokrazy: attempt 61, starting ["/user/wifi"]
  2022/02/21 19:13:46 feeding firmware (14036 bytes) to brcmfmac via sysfs/brcm!brcmfmac43455-sdio.clm_blob
  2022/02/21 19:13:46 no interfaces found
  2022/02/21 19:13:46 gokrazy: exit status 1
  2022/02/21 19:13:47 gokrazy: attempt 62, starting ["/user/wifi"]
  2022/02/21 19:13:47 connecting to SSID "xxxxxx"...
```
The last line keeps repeating every 15s.

dmesg
```cfg
...
[    5.130791] random: crng init done
[    5.271399] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): No External EEPROM. Setting MAC Speed
[    5.285945] libphy: lan78xx-mdiobus: probed
[   32.004286] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt failed with error -2
[   32.019321] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt
[   33.167264] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   64.478904] cfg80211: failed to load regulatory.db
[   93.150985] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2
[   93.163415] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.txt
[   93.333638] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   94.264861] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[  516.049903] ieee80211 phy0: brcmf_cfg80211_connect: BRCMF_C_SET_SSID failed (-110)
```
This last line may have been written when I stopped the wifi service.

hardware is a pi 3B+

Any suggestions on debugging this?

Merovius, 2022-02-22:
> I was going to suggest using go-bindata rather than goembed

Why not [`//go:embed`](https://pkg.go.dev/embed)? It doesn't require `go generate` and doesn't have the overhead of having to read/parse/compile a large `.go` file.

stapelberg, 2022-02-22:
Yeah we should definitely switch to //go:embed. Our wifi package predates it, but it’s a nicer solution than goembed, go-bindata, etc.

stapelberg, 2022-02-22:
With the latest packer, wifi and kernel package, the `brcmfmac` driver is now built as a loadable kernel module. This makes firmware loading work again, as long as the firmware is included in the root file system (I don’t know why the sysfs firmware loading fallback is broken).

With the following per-package extra files workaround, I get working (unencrypted) WiFi on the Raspberry Pi 4 again:

```
mkdir -p extrafiles/github.com/gokrazy/wifi/lib/firmware/brcm/
cp ~/go/src/github.com/gokrazy/wifi/third_party/firmware-nonfree/* extrafiles/github.com/gokrazy/wifi/lib/firmware/brcm
```

Next up is looking into a way to make this step unnecessary, i.e. have packages influence the extra files in the root file system.

stapelberg, 2022-02-22:
> So,
> 
> https://github.com/justinazoff/wifi/tree/wpa-psk has the updates for basic WPA PSK support
> 
> https://github.com/justinazoff/wifi-wpa is the fork of gokrazy/wifi updated to call ConnectWPAPSK

Looks good on first glance! @JustinAzoff would you mind sending these as pull requests to their respective repositories please?

mpictor, 2022-02-23:
My 3B+ is now connecting. PEBCAK: I thought the packages passed as args to gokr-packer would be searched under `$GOPATH/src` first, but that wasn't the case - so it was pulling down the non-psk version.

If I unplug ethernet, it stops responding on the wireless interface - and comes back when I plug the cable back in. Not sure what would cause that, and dmesg had nothing interesting. And yes, I am connecting over the wifi IP :)

JustinAzoff, 2022-02-23:
@mpictor ah, that just sounds like a routing issue.  Even though you are connecting over the wlan0 ip, the kernel is sending the responses to you out of eth0 because eth0 and wlan0 are on the same subnet.  If you boot without the cable connected things should work properly.

anupcshan, 2022-02-24:
(unrelated to the wifi discussion, but related to loading firmware in gokrazy)

It might be possible to build the kernel with specific firmware blobs embedded in it - that avoids the need for loading modules/firmware on demand, or as part of application code. I did something like this for another device. [Relevant kernel config](https://github.com/anupcshan/gokrazy-odroidxu4-kernel/blob/32fe20104f420c2d0fb9113e9e6b5ba60b80edeb/cmd/gokr-build-kernel/config.txt#L17-L18) and corresponding [build script snippet](https://github.com/anupcshan/gokrazy-odroidxu4-kernel/blob/32fe20104f420c2d0fb9113e9e6b5ba60b80edeb/cmd/gokr-build-kernel/build.go#L50-L80).

stapelberg, 2022-02-24:
> @mpictor ah, that just sounds like a routing issue. Even though you are connecting over the wlan0 ip, the kernel is sending the responses to you out of eth0 because eth0 and wlan0 are on the same subnet. If you boot without the cable connected things should work properly.

That sounds like the most likely explanation to me, too. 

I think you could use IPv6 link-local addresses (the ones starting with `fe80::`) to reliably connect to the device in this situation. The downside is that while these work fine for SSH, browsers typically don’t accept them…

> It might be possible to build the kernel with specific firmware blobs embedded in it - that avoids the need for loading modules/firmware on demand, or as part of application code. I did something [...]

Thanks for the tip! However, embedding the wifi firmware into the kernel means that users can no longer choose whether they need it or not. Including it in the `github.com/gokrazy/wifi` package seems preferable to me for that reason :)

stapelberg, 2022-02-24:
> Next up is looking into a way to make this step unnecessary, i.e. have packages influence the extra files in the root file system.

This is now done! With the latest packer, no extra steps should be necessary to use the wifi package.

@JustinAzoff Are you up for sending your changes as a PR or would you prefer if someone else did?

stapelberg, 2022-02-28:
@mpictor FYI, I updated the github.com/gokrazy/wifi package to also include working firmware for the Raspberry Pi 3B and Raspberry Pi 3B+

JustinAzoff, 2022-02-28:
@stapelberg I can send over the PR's, I don't mind at all if someone else takes over though :-)

Other than moving some of those hardcoded values up to constants it should work as-is.

stapelberg, 2022-03-01:
With commit https://github.com/gokrazy/wifi/commit/f75d908efe145916124b67c2bc1dcb728a0273dd the wifi package now supports WPA-PSK! 🎉

I’ll update the documentation accordingly, but this should be ready for folks to try out now :)

stapelberg, 2022-03-01:
I’m closing this issue now that we have full WiFi support. Feel free to file separate issues in case some specific aspect needs improvement.

closed 2022-03-01 by stapelberg

stapelberg, 2022-03-05:
> > @mpictor ah, that just sounds like a routing issue. Even though you are connecting over the wlan0 ip, the kernel is sending the responses to you out of eth0 because eth0 and wlan0 are on the same subnet. If you boot without the cable connected things should work properly.

FYI, since commit https://github.com/gokrazy/gokrazy/commit/4c375187675fd3c3f2fb1381be87b7d046dd1f42, the gokrazy dhcp client sets route priorities and de-prioritizes routes of interfaces that are down. When you un-plug the ethernet cable now, it should just transparently switch over to wifi :)

StefanLobbenmeier, 2022-05-27:
Just read through this issue and looking for some clarification 

wifi setup at first sounds a bit like https://github.com/jasbur/RaspiWiFi to me, but reading through the implementation it seems that the configuration is only via the json file, not via an access point and Webserver

is this also something that is already possible with gokrazy?

stapelberg, 2022-05-27:
I haven’t tried doing WiFi in master mode on the Raspberry Pi. I don’t know if it can be done just with the netlink API (without an extra wpa_supplicant process).

So, no, this is currently not possible, but further investigation is welcome. I’d recommend filing a separate GitHub issue for this feature.