gokrazy/gokrazy #14: Installation on Mac OS X 10.11.6 fails

uho, 2017-03-05:
Following the proposed installation procedure, the installation fails:

0. System
     `$ uname -a`
     `Darwin computer.local 15.6.0 Darwin Kernel Version 15.6.0: Mon Jan  9 23:07:29 PST 2017; root:xnu-3248.60.11.2.1~1/RELEASE_X86_64 x86_64`
     `$ sw_vers -productVersion` 
     `10.11.6`

1. Checking for correct Go version 1.8:

     `$ go version` 
     `go version go1.8 darwin/amd64`

2. Installing gokrazy:

    ` $ go get -u github.com/gokrazy/tools/cmd/gokr-packer`
     `# github.com/gokrazy/gokrazy/internal/iface`
     `go/src/github.com/gokrazy/gokrazy/internal/iface/iface.go:108: undefined: syscall.SIOCADDRT`
     `go/src/github.com/gokrazy/gokrazy/internal/iface/iface.go:122: undefined: syscall.SIOCDELRT`

The correct behavior would be to install without errors.

stapelberg, 2017-03-05:
Thanks for the report. The code in question should only be compiled for the target (linux arm64), but is currently compiled for the host due to the way we pull in code. I’ll push a commit in a second which should fix this. Please reopen if you still encounter issues afterwards.

uho, 2017-03-06:
Fix changes behavior to

    `$ go get -u github.com/gokrazy/tools/cmd/gokr-packer`
    `# github.com/gokrazy/tools/cmd/gokr-packer`
    `go/src/github.com/gokrazy/tools/cmd/gokr-packer/parttable.go:81: undefined: unix.BLKGETSIZE64`
    `go/src/github.com/gokrazy/tools/cmd/gokr-packer/parttable.go:93: undefined: unix.BLKRRPART`

still not able to install without errors.

uho, 2017-03-06:
Fix changes behavior to

`$ go get -u github.com/gokrazy/tools/cmd/gokr-packer`
`# github.com/gokrazy/tools/cmd/gokr-packer`
`go/src/github.com/gokrazy/tools/cmd/gokr-packer/parttable.go:81: undefined: unix.BLKGETSIZE64`
`go/src/github.com/gokrazy/tools/cmd/gokr-packer/parttable.go:93: undefined: unix.BLKRRPART`

still not able to install without errors. Reopening this issue.

stapelberg, 2017-03-07:
Thanks. I’ll need to make that code conditional, then.

Given that I don’t have a macOS machine handy, any fix would be blind. I might be able to actually test gokrazy (and not just have it compile) on macOS next week. If you want to race me to it, contributions are very welcome :).

uho, 2017-03-07:
It looks like the constants

unix.BLKGETSIZE64
unix.BLKRRPART

are not available with darwin/amd64 as Darwin lacks these requests for IOCTL. 

A substitute for Darwin has to be identified in order to
determine the device size and re-read partition table.

Maybe [The discusscion How do I get the partition offset in OS X with C/C++? on Stackoverflow](https://stackoverflow.com/questions/16656218/how-do-i-get-the-partition-offset-in-os-x-with-c-c) gives some hints.

stapelberg, 2017-03-18:
The commits I just pushed should allow you to install `gokr-packer` on macOS.

## Prep work

Before you’ll be able to create an image, you’ll need to dump the system root certificates:
```
mkdir -p ~/.config/gokrazy
security find-certificate -a \
  -p /System/Library/Keychains/SystemCACertificates.keychain \
  > ~/.config/gokrazy/cacert.pem
```

## Test cross-compilation

Check whether `GOOS=linux GOARCH=arm64 go get -u github.com/gokrazy/hello` fails. If it fails due to a permission error (common if you have used the Go installer), you have 3 options:

1. Run gokrazy as root, thereby gaining write permission to `/usr/local/go`
2. Copy `/usr/local/go` to a temporary directory and set the `GOROOT` environment variable to that directory
3. Re-install Go from source into a directory where you have write permission.

## Creating an image

I’d recommend to follow https://github.com/gokrazy/tools#alternative-creating-an-sd-card-image and use whichever tool you feel comfortable with to write the image to an SD card.

## Creating an image (the harder(?) way)

In case you want to use `gokr-packer` to directly write to the SD card, a couple more steps are necessary. Notably, you’ll need to repeat all of these steps whenever you want to directly write to your SD card.

Locate your SD card in `diskutil list`. On my machine, it shows up as `/dev/disk2`.

In case the SD card contained any partitions with a file system, use `umount` to unmount the file systems (otherwise you’ll get a “device busy” error on the next command).

Overwrite the SD card with zeros to remove all partitions and file systems, preventing macOS from mounting the file systems after we overwrite the partition table:

```
dd if=/dev/zero of=/dev/rdisk2 bs=1m count=1024
```

Afterwards, the following worked for me:

```
gokr-packer -overwrite=/dev/disk2 github.com/gokrazy/hello
```

macOS prompts you halfway through the process about not being able to read any file systems (which aren’t written at that point). Dismiss the dialog by clicking its “ignore” button.

stapelberg, 2017-03-18:
I’m closing this issue. We can discuss/track individual remaining issues/improvements to the experience in separate issues, if desired.

closed 2017-03-18 by stapelberg

uho, 2017-03-20:
Verified this successfully.

Thanks for the effort.

A pleasant Mac program to write images to SD cards is Etcher (https://etcher.io/).