gokrazy/gokrazy #16: How to overwrite the SD card in Windows?

henson, 2017-03-06:
Your job is great, but i only have Windows programming environment, how to overwrite the SD card in Windows? I wondered. The only way is to install virtual machine?

stapelberg, 2017-03-07:
I don’t use Windows, so I don’t know. Perhaps someone could contribute some code to overwrite SD cards on Windows?

Drache93, 2017-03-17:
You'll have to pack to a .img:

```
gokr-packer \
  -overwrite=/tmp/full.img \
  -target_storage_bytes=2147483648 \
  github.com/gokrazy/hello
```

Then write to disk, I use Win32 Disk Imager for all my Pi needs:
https://sourceforge.net/projects/win32diskimager/

You'll need to match the storage size so you don't make an image too big to fit on your card. I assume all the tools to resize are still on the img. I hadn't noticed the above commands when i tried the VM route.

stapelberg, 2017-03-18:
Thanks for the additional details.

There’s no resizing tools within gokrazy currently.