gokrazy/gokrazy #17: key store

joeblew99, 2017-03-16:
generally you need a key chain of sorts to store a password to a DB.
In IOT world anyone can grab your device, so you need to encrypt sensitive data.
A key store allows this for use cases of:
- DB password.
- credit cards and other Sensitive info

I have not looked into how hard this is
there are go loibs out there for the basic crypto stuff.

is this something you feel is worthwhile putting effort into ?
i would like to try to get it working, but might need to hassle you a bit

stapelberg, 2017-03-17:
> In IOT world anyone can grab your device

gokrazy’s target audience are devices which run in your home or other trusted places. I’d hope that the people you trust enough to let into your home won’t steal your physical devices (their infected computers might try to attack your devices via the network, though).

If you’re worried about this, feel free to develop a solution, but I don’t think it makes sense to merge it into gokrazy, at least not right now. Since there is no technical difference between a Go package living in the gokrazy repository itself and one living in a third-party repository, maintaining your package elsewhere should be fine. If the community adopts your solution, we can revisit.

closed 2017-03-17 by stapelberg

joeblew99, 2017-03-17:
fair enough, and yes i see your point. thanks for the feedback.

i wil try out my own. Its perfectly possible to do it with 100% golang.
It can be used to hold any credentials for things you need, without putting it in your source code.