gokrazy/gokrazy #19: Feature Idea. QUIC transport

joeblew99, 2017-03-17:
With IOT devices, you want to be able to reach it, and you want to be able to collaborate with other users / Devices.
QUIC is a UDP based protocol developed by google for this.
Its 100% golang and not kernel dependent and thats part of why its so compelling for IOT.

It makes it easy to do peer to peer and it very very fast.
It also very easy to use it to standard golang Http handler.

Gokrazy can use it for talking to the RaspberryPI from anywhere, easily.
Apps built on top can use it in a myriad of ways.

Here some great URLS:
https://github.com/lucas-clemente/quic-go
- The core QUIC code.

https://github.com/marten-seemann/libp2p-quic-transport
- Using QUIC with the IPFS system. Really nice as it makes it very easy to send messages between different IOT devices or mobiles with a consensus and data store and pubsub. Its kind of go it all.
- IPFS allows any transport protocol to be used, and the addition of QUIC is very new, so dont expect this to be stable yet. But you can se where its going.

stapelberg, 2017-03-17:
If apps running on top of gokrazy want to use QUIC (or any other transport protocol), they’re free to do so. AFAICT, we don’t need to change anything about gokrazy itself. Am I misunderstanding something?

joeblew99, 2017-03-22:
your right. No need to be addressed at this level.
i am using it allot for IOT and so wanted to let you know.
Because QUIC can punch through NAT its a nice weapon.
Will close this now....

closed 2017-03-22 by joeblew99