gokrazy/gokrazy #22: Mac OS X fails with "resource busy"

shutej, 2017-03-25:
At least on Mac OS X 10.12.3, gokrazy

sudo -E $(which gokr-packer) -overwrite=/dev/disk2 github.com/gokrazy/hello
2017/03/25 13:53:04 packer.go:222: installing [github.com/gokrazy/hello]
2017/03/25 13:53:08 packer.go:123: partitioning /dev/disk2
2017/03/25 13:53:08 parttable.go:82: device holds 7861174272 bytes
2017/03/25 13:53:08 packer.go:131: waiting for /dev/disk2s1 to appear
2017/03/25 13:53:09 packer.go:282: open /dev/disk2s1: resource busy

My diskutil shows that the NOOBS partition is ejectable after partitioning, so it's probably being remounted by the OS immediately after it's written?

stapelberg, 2017-03-25:
Yes. Issue #14 is related, specifically https://github.com/gokrazy/gokrazy/issues/14#issuecomment-287560193

Not sure if we can do better than that, i.e. I don’t know of a way to tell the OS to not mount the partitions. Do you know?

shutej, 2017-03-26:
Thanks!  I think it may be easier for you to declare writing SD cards out of scope and point people to resin.io, then you're unlikely to have to take this hassle on as a bug every time an OS X user (like me) comes along.

shutej, 2017-03-26:
Or, at least change the default instructions to point only to automating SD card overwriting as an "advanced" feature, and make the default instructions delegate to another tool?

stapelberg, 2017-03-26:
For macOS, that’s correct, for Linux (my primary environment), the current instructions are more convenient.

Maybe we should add an operating system specific message about this to the packer.

palmamartin, 2022-01-03:
In order to fix the `resource busy` error run the following command `diskutil unmountDisk /dev/disk#`. In your case `diskutil unmountDisk /dev/disk2`.

stapelberg, 2022-01-09:
I wonder if we could make the packer run the unmount commands automatically on macOS

bykof, 2022-03-01:
> In order to fix the `resource busy` error run the following command `diskutil unmountDisk /dev/disk#`. In your case `diskutil unmountDisk /dev/disk2`.

if I unmount the disk and run gokrazy i get: 
`sync /dev/disk2: inappropriate ioctl for device`

stapelberg, 2022-03-01:
Thanks, this should be fixed with commit https://github.com/gokrazy/tools/commit/0721be47155a1a0597d1e4ebf4787835bb4dbf36