gokrazy/gokrazy #23: Problem with permissions on Debian Stretch

mboujong, 2017-04-16:
Are there problems with the program path to go and work directory? Does the procedure require certain rights?

bla@debian:~/go/bin$ sudo setcap CAP_SYS_ADMIN,CAP_DAC_OVERRIDE=ep $(go env GOPATH)/bin/gokr-packer

bla@debian:~/go/bin$ gokr-packer -overwrite=/dev/sdb github.com/gokrazy/hello
2017/04/16 21:45:55 packer.go:225: installing [github.com/gokrazy/hello]
go install runtime/internal/sys: mkdir /usr/local/go/pkg/linux_arm64/runtime: permission denied

stapelberg, 2017-04-16:
Thanks for the report!

You’ll need to install Go into a directory in which you have write permission.

This problem is tracked upstream at https://github.com/golang/go/issues/18981#issuecomment-287560719, so I’m closing this issue. Feel free to re-open if there’s anything we can do on our end to improve things.

closed 2017-04-16 by stapelberg

XenGi, 2017-06-27:
I followed the instructions in the [Archwiki](https://wiki.archlinux.org/index.php/Go#Enable%20cross%20compilation%20for%20other%20platforms), and now it works.
What I did:

```
$ cd ~/downlaods
$ wget https://storage.googleapis.com/golang/go1.8.3.src.tar.gz
$ tar xaf go1.8.3.src.tar.gz
$ cd go/src
$ GOROOT_BOOTSTRAP=/usr/lib/go GOOS=linux GOARCH=amd64 ./make.bash --no-clean
$ cd /usr/lib/go/src
$ sudo GOROOT_BOOTSTRAP="$HOME/downloads/go" GOOS=linux GOARCH=arm64 ./make.bash --no-clean
```

Maybe you can put this in a FAQ or "typical problems" section.

stapelberg, 2017-06-27:
The [quickstart](https://gokrazy.org/quickstart.html) already links to https://golang.org/dl/, which links to binaries and source installation instructions. When following the quickstart to the letter, everything should work.