gokrazy/gokrazy #24: go 1.8 does not work on raspbian

Hollywood64, 2017-04-19:
I tried go 1.8.0 and 1.8.1 and it does not work with the same error message
I am using raspbian on a raspberry pi 3 with
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
My second sd-card is not mounted and with no partition with 32GB.
gokr-packer shows the following messages
packer.go:225: installing [github.com/gokrazy/hello]
packer.go:123: partitioning /dev/sda
packer.go:285: invalid argument

stapelberg, 2017-04-20:
Thanks for giving gokrazy a shot!

What’s the result of `fdisk -l /dev/sda`?

Also, can you supply an strace log? Run `strace -f -o /tmp/strace.log -s2048 -v gokr-packer …`, then compress and attach /tmp/strace.log. Thanks.

Hollywood64, 2017-04-20:
Hi Michael,

the result of fdisk is:

    pi@raspberrypi:~ $ sudo fdisk -l /dev/sda

    Disk /dev/sda: 29 GiB, 31104958464 bytes, 60751872 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: dos
    Disk identifier: 0x05efbb9d

    Device     Boot Start      End  Sectors Size Id Type
    /dev/sda1        8192 60741764 60733573  29G  c W95 FAT32 (LBA)

I  attached  the end  of the strace.log, because it was compressed 
(7-zip ultra) 17MB, which was too large for sending per e-mail.

1388  clock_gettime(CLOCK_REALTIME,  <unfinished ...>
1383  select(0, NULL, NULL, NULL, {0, 20} <unfinished ...>
1388  <... clock_gettime resumed> {1492678030, 560556102}) = 0
1388  write(2, "2017/04/20 10:47:10 packer.go:123: partitioning 
/dev/sda\n", 57 <unfinished ...>
1383  <... select resumed> )            = 0 (Timeout)
1388  <... write resumed> )             = 57
1383  clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
1388  openat(AT_FDCWD, "/dev/sda", 
O_RDWR|O_CREAT|O_TRUNC|O_LARGEFILE|O_CLOEXEC, 0666 <unfinished ...>
1383  <... clock_gettime resumed> {1579, 411314044}) = 0
1383  clock_gettime(CLOCK_REALTIME, {1492678030, 560864433}) = 0
1383  select(0, NULL, NULL, NULL, {0, 20}) = 0 (Timeout)
1383  clock_gettime(CLOCK_MONOTONIC, {1579, 411656906}) = 0
1388  <... openat resumed> )            = 3
1383  clock_gettime(CLOCK_REALTIME,  <unfinished ...>
1388 *ioctl(3, BLKGETSIZE64, 0x10747cd8 <unfinished ...>*
1383  <... clock_gettime resumed> {1492678030, 561192764}) = 0
1388 *<... ioctl resumed> )             = -1 EINVAL (Invalid argument)*
1383  select(0, NULL, NULL, NULL, {0, 20} <unfinished ...>
1388  close(3 <unfinished ...>
1383  <... select resumed> )            = 0 (Timeout)
1383  clock_gettime(CLOCK_MONOTONIC, {1579, 412008778}) = 0
1383  clock_gettime(CLOCK_REALTIME, {1492678030, 561526564}) = 0
1383  select(0, NULL, NULL, NULL, {0, 20} <unfinished ...>
1388  <... close resumed> )             = 0
1388  clock_gettime(CLOCK_REALTIME,  <unfinished ...>
1383  <... select resumed> )            = 0 (Timeout)
1388  <... clock_gettime resumed> {1492678030, 561711406}) = 0
1383  clock_gettime(CLOCK_MONOTONIC, {1579, 412331745}) = 0
1383  clock_gettime(CLOCK_REALTIME, {1492678030, 561851665}) = 0
1388  write(2, "2017/04/20 10:47:10 packer.go:285: invalid argument\n", 
52 <unfinished ...>
1383  select(0, NULL, NULL, NULL, {0, 20} <unfinished ...>
1388  <... write resumed> )             = 52
1388  exit_group(1)                     = ?
1383  <... select resumed> )            = 0 (Timeout)
1388  +++ exited with 1 +++

I assume the marked lines show the error.

regards
Hollywood64

stapelberg, 2017-04-20:
Hm, it’s odd that the BLKGETSIZE64 ioctl doesn’t work on your machine. What’s the result of `strace -eraw=ioctl -f blockdev --getsize64 /dev/sda`?

Hollywood64, 2017-04-20:
Am 20.04.2017 um 12:12 schrieb Michael Stapelberg:
> |strace -eraw=ioctl -f blockdev --getsize64 /dev/sda|

Hi Michael,

i attached the result.

regards

Hollywood64, 2017-04-20:
Hi Michel,

i compared the results of the following commands:

1. sudo setcap CAP_SYS_ADMIN,CAP_DAC_OVERRIDE=ep /usr/bin/strace
     strace -eraw=ioctl -f blockdev --getsize64 /dev/sda 
 >/tmp/strace4.log 2>&1

2. sudo strace -eraw=ioctl -f blockdev --getsize64 /dev/sda 
 >/tmp/strace5.log 2>&1

Could it be a problem with access rights?

Does user pi get access rights with sudo setcap ...

|sudo setcap CAP_SYS_ADMIN,CAP_DAC_OVERRIDE=ep ~/go/bin/gokr-packer 
gokr-packer -overwrite=/dev/sda github.com/gokrazy/hello|

Should I put the user pi into some groups ?

I have tested 2 different USB sd-card readers.

pi@raspberrypi:~ $ dmesg | tail
[ 1664.159236] scsi host1: usb-storage 1-1.2:1.0
[ 1665.157939] scsi 1:0:0:0: Direct-Access     ChipBank SD/MM Reader     
4080 PQ: 0 ANSI: 2
[ 1665.159215] sd 1:0:0:0: Attached scsi generic sg0 type 0
[ 1665.160117] sd 1:0:0:0: [sda] 60751872 512-byte logical blocks: (31.1 
GB/29.0 GiB)
[ 1665.160394] sd 1:0:0:0: [sda] Write Protect is off
[ 1665.160413] sd 1:0:0:0: [sda] Mode Sense: 0b 00 00 08
[ 1665.160674] sd 1:0:0:0: [sda] No Caching mode page found
[ 1665.160691] sd 1:0:0:0: [sda] Assuming drive cache: write through
[ 1665.166617]  sda: sda1
[ 1665.169209] sd 1:0:0:0: [sda] Attached SCSI removable disk

the other is

[ 1719.984174] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[ 1719.984456] scsi host2: usb-storage 1-1.2:1.0
[ 1720.981136] scsi 2:0:0:0: Direct-Access     Lexar    microSD RDR      
0815 PQ: 0 ANSI: 6
[ 1720.982306] sd 2:0:0:0: Attached scsi generic sg0 type 0
[ 1721.188986] sd 2:0:0:0: [sda] 60751872 512-byte logical blocks: (31.1 
GB/29.0 GiB)
[ 1721.189836] sd 2:0:0:0: [sda] Write Protect is off
[ 1721.189857] sd 2:0:0:0: [sda] Mode Sense: 23 00 00 00
[ 1721.190524] sd 2:0:0:0: [sda] Write cache: disabled, read cache: 
enabled, doesn't support DPO or FUA
[ 1721.195122]  sda: sda1
[ 1721.197941] sd 2:0:0:0: [sda] Attached SCSI removable disk

Do you have an idea?

commands:

sudo setcap CAP_SYS_ADMIN,CAP_DAC_OVERRIDE=ep /usr/bin/strace
strace -eraw=ioctl -f blockdev --getsize64 /dev/sda >/tmp/strace4.log 2>&1

result:

execve(0x7eed41fc, [0x7eed5620], [/* 0 vars */]) = 0
brk(0)                                  = 0x9cd000
uname({sys="Linux", node="raspberrypi", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76f38000
access("/etc/ld.so.preload", R_OK)      = 0
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=42, ...}) = 0
mmap2(NULL, 42, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76f37000
[... 134 lines cut ...]
mprotect(0x76f20000, 8192, PROT_READ)   = 0
mprotect(0x76f26000, 20480, PROT_READ|PROT_WRITE) = 0
mprotect(0x76f26000, 20480, PROT_READ|PROT_EXEC) = 0
cacheflush(0x76f26000, 0x76f2b000, 0, 0x15, 0x7ebef360) = 0
mprotect(0x25000, 4096, PROT_READ)      = 0
mprotect(0x76f6a000, 4096, PROT_READ)   = 0
munmap(0x76f60000, 32327)               = 0
brk(0)                                  = 0x198d000
brk(0x19ae000)                          = 0x19ae000
open("/usr/lib/locale/locale-archive", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=3111696, ...}) = 0
mmap2(NULL, 2097152, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76be5000
close(3)                                = 0
open("/dev/sda", O_RDONLY|O_LARGEFILE)  = 3
ioctl(0x3, 0x80041272, 0x7ebef668)      = 0
fstat64(1, {st_mode=S_IFREG|0644, st_size=3729, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76f67000
close(3)                                = 0
write(1, "31104958464\n", 1231104958464
)           = 12
close(1)                                = 0
munmap(0x76f67000, 4096)                = 0
close(2)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++

stapelberg, 2017-04-20:
What does `go version` say on your Raspberry Pi? I think the issue might be a wrong constant value in golang.org/x/sys/unix: it currently defines `BLKGETSIZE64 = 0x80081272` in unix/zerrors_linux_arm.go, but blockdev uses 0x80041272 on your machine.

You could try modifying that constant in $(go env GOPATH)/src/golang.org/x/sys/unix/zerrors_linux_arm.go, then re-installing gokr-packer.

Hollywood64, 2017-04-20:
Hi Michael,

this is my current go is:

pi@raspberrypi:~ $ go version
go version go1.8 linux/arm

Thank you, there is no error now.

I noticed to set ulimit here: 
https://dave.cheney.net/2015/09/04/building-go-1-5-on-the-raspberry-pi

pi@raspberrypi:~/go $ ulimit -s 1024
pi@raspberrypi:~/go $ ulimit -s
1024
pi@raspberrypi:~/go $ cd ~
pi@raspberrypi:~ $ go get -u github.com/gokrazy/tools/cmd/gokr-packer
pi@raspberrypi:~ $ sudo setcap CAP_SYS_ADMIN,CAP_DAC_OVERRIDE=ep 
~/go/bin/gokr-packer
pi@raspberrypi:~ $ gokr-packer -overwrite=/dev/sdb github.com/gokrazy/hello
2017/04/20 16:32:24 packer.go:225: installing [github.com/gokrazy/hello]
2017/04/20 16:32:29 packer.go:291: -target_storage_bytes is required 
when using -overwrite with a file
pi@raspberrypi:~ $ gokr-packer -overwrite=/dev/sda github.com/gokrazy/hello
2017/04/20 16:32:40 packer.go:225: installing [github.com/gokrazy/hello]
2017/04/20 16:32:45 packer.go:123: partitioning /dev/sda
2017/04/20 16:32:45 parttable.go:82: device holds 31486640128 bytes
2017/04/20 16:32:46 packer.go:131: waiting for /dev/sda1 to appear
2017/04/20 16:32:47 write.go:92: writing boot file system
2017/04/20 16:32:54 write.go:196: writing root file system: 
map[/user/hello:/home/pi/go/bin/hello /etc/: /proc/: /sys/: 
/gokrazy/ntp:/home/pi/go/bin/ntp 
/gokrazy/init:/tmp/gokr-packer333190163/init /localtim:/etc/localtime 
/cacerts:/etc/ssl/certs/ca-certificates.crt 
/gokr-pw.txt:/home/pi/.config/gokrazy/http-password.txt /dev/: /tmp/: 
/perm/:]
If your applications need to store persistent data, create a file system 
using e.g.:

         mkfs.ext4 /dev/sda4

To boot gokrazy, plug the SD card into a Raspberry Pi 3 (no other model 
supported)

To interact with the device, gokrazy provides a web interface reachable at:

http://gokrazy:HalloWorld@gokrazy/

There will be not be any other output (no HDMI, no serial console, etc.)
pi@raspberrypi:~ $

When booting from the new sd-card then it does not work.

I see nothing on screen and the network LEDs are off.

What do you mean? Are there other entry points in raspbian linux wrong too ?

regards
Hollywood64

stapelberg, 2017-04-20:
To be clear, you did change the constant value to make it work, yes? If so, I’ll send a fix upstream.

Hollywood64, 2017-04-20:
yes i changed

  nano $(go env GOPATH)/src/golang.org/x/sys/unix/zerrors_linux_arm.go

the following line only

//      BLKGETSIZE64                     = 0x80081272
         BLKGETSIZE64                     = 0x80041272

regards

stapelberg, 2017-04-20:
Thanks. Sent https://go-review.googlesource.com/#/c/41215/

stapelberg, 2017-04-21:
This is fixed upstream with https://github.com/golang/sys/commit/ea9bcade75cb975a0b9738936568ab388b845617

closed 2017-04-21 by stapelberg