gokrazy/gokrazy #25: Auto-updating from gokrazy

Merovius, 2017-04-19:
As discussed offline, it would be great, if gokrazy would support full auto-updates (driven by gokrazy itself).
It would probably require shipping a copy of the go tool and the packer, regularly checking for updates via the github API, fetching a tarball of the content (as we won't have git/mercurial/svn on the rpi) if there is one, rebuilding the image and rebooting. Most of this can probably be built as a service, though some changes to the packer are required: a) The packer needs to add a released binary of the go tool (we probably don't want to do a full recompilation?) and b) it needs to add meta-information about the build, so that the packer running on gokrazy can rebuild it.

It's probably also worth it to consider user-packages for auto-updates, though the process is probably going to be different. Most third-party things probably shouldn't be built from HEAD, so either we need to rely on some kind of versioning scheme to decide what to update to, or rely on github releases and download binaries.
Updated versions should first be written to and started from /tmp, to make sure that they start correctly. Afterwards, the packer needs to build a new image containing the updated binary.

I think in the end we'd want auto-updates both for gokrazy itself and the user-installed services, to have fully self-updating appliances.

stapelberg, 2017-04-20:
> The packer needs to add a released binary of the go tool (we probably don't want to do a full recompilation?)

I think it’d be nice if the auto-update mechanism lived in a separate package, e.g. github.com/gokrazy/selfdestruct. We’d need to find a way to package the go tool (and gokr-packer) so that `go install` installs their binaries, then gokr-packer would just pick it up like any other package. Merely copying the host’s go tool and gokr-packer would not work in many cases: the host binaries are not compiled with `CGO_ENABLED=0`. Thoughts?

> Most third-party things probably shouldn't be built from HEAD, so either we need to rely on some kind of versioning scheme to decide what to update to, or rely on github releases and download binaries.

Playing the version game is something we can’t win right now, I think. gokrazy intentionally uses the go tool for fetching and building Go packages, and I don’t feel compelled to change that currently (this may change when/if the dep tool becomes part of the toolchain). If people have more complicated needs for determining the version, they need to either:

* put that logic into some custom infrastructure and produce go gettable packages
* do “external” (as opposed to on-device) updates

FWIW, here’s the script I run via cron to automatically update my installation:

```sh
#!/bin/sh

export GOPATH=/home/michael/go
export PATH=/home/michael/go/bin:/home/michael/go-latest/bin:$PATH

go get -u github.com/gokrazy/kernel
go get -u github.com/gokrazy/firmware
gokr-packer -serial_console=disabled -hostname=dr \
  -update=http://gokrazy:secret@dr/ \
  github.com/stapelberg/zkj-nas-tools/avr-x1100w \
  github.com/stapelberg/hmgo \
  github.com/stapelberg/zkj-nas-tools/dornröschen \
  github.com/gokrazy/breakglass \
  github.com/gokrazy/bakery/cmd/bootery
```

Merovius, 2017-04-20:
> I think it’d be nice if the auto-update mechanism lived in a separate package, e.g. github.com/gokrazy/selfdestruct

Though I disagree with the name; yes, that'd be nice. The packer still needs to add the information of *what* to fetch, though. I looked into trying to extract that from the binary; the go-tool adds the path that a binary was built from and we could use a heuristic to guess the import path from that, but I don't think it's entirely reliable. The packer knows the import paths and it seems wrong to me to first drop that information and then employing a heuristic to try to reconstruct it.

Indeed, that's the reason I opened this issue in the first place; otherwise I would've just went ahead and tried to do it. But I really believe that letting the packer do its part here is the right thing.

> We’d need to find a way to package the go tool (and gokr-packer) so that go install installs their binaries

I think Francesc Campoy at some point made a go-gettable version of the go-tool, though I can't find it right now. I asked him on twitter. Alternatively, we could do the download from gokrazy itself; i.e. make a service that fetches the correct binary from https://golang.org/dl/ into tmp and uses that to compile things. The service then gets started on boot and can check for updates.

I think most of the design decisions (re versions and how to fetch things) could be solved by experimentation and prototyping. Technically, even the thing I'm most interested in right now (namely packaging the list of import paths to install) could be done by just wrapping the packer; 
 But if we think it's gokrazy should support this (and I do), it would really benefit from some support from first-class tooling.

joeblew99, 2017-04-20:
@Merovius Is this issue assuming that compilation happens on the RaspPI ?

Merovius, 2017-04-20:
@joeblew99 Either that, or downloading binaries (don't particularly care which, TBH). My main want is, that the Pi ends up being a self-contained appliance that will just take care of its own updates.

I am dissatisfied with the "let a cronjob on a workstation update it" solution because a) I don't have a static workstation but only a laptop that moves around, b) I might want to move to a different machine and don't want to have to worry about migrating/fixing the cronjob when that happens and c) the whole reason to have an rpi in my home in the first place is to have an always-on computer that can automate things; needing to maintain that from a different machine kind of defeats the purpose, IMHO.

stapelberg, 2017-04-20:
> Though I disagree with the name

Do you have a better suggestion? I like how it conveys that things might go horribly wrong (since there’s no canarying involved), but I see how it can be too pessimistic.

> The packer still needs to add the information of what  to fetch, though

Agreed. Pull requests welcome.

> that fetches the correct binary from https://golang.org/dl/ into tmp and uses that to compile things

I think these binaries are not compiled with `CGO_ENABLED=0`. Also, there does not seem to be a pre-compiled binary for arm64 at all (only armv6). Possibly that can be fixed, but it’s another hurdle to be aware of.

> I am dissatisfied with the "let a cronjob on a workstation update it" solution

I understand your reasons. One thing to note, possibly as a temporary workaround: you could also set up a port forwarding and run your updater in the cloud (e.g. on an amazon or google VM, or even on travis, or…).

In general, I like the concept of a self-updating gokrazy installation, though.

Merovius, 2017-04-20:
>  I like how it conveys that things might go horribly wrong (since there’s no canarying involved), but I see how it can be too pessimistic.

I think upstream CI should already serve as a canary. For user-services I'd like to built some canarying in (as alluded to in my earlier comment).
There still might be things going wrong, which is true, but there always might be things going wrong. Choosing a name like this very clearly says that this is an unsupported usecase and not supposed to work, which is contrary to my wishes (but might be your intention). But if that's so, this issue should just be closed.

> I think these binaries are not compiled with `CGO_ENABLED=0`

Hm, true, at least on my machine the go tool even is dynamically linked. That's a bummer. There is an issue about creating statically linked binaries, but it doesn't make me optimistic.

I think the best way to provide this would be to provide an appropriately compiled go tool via CI, like the kernel and firmware repos. As a bonus, we could have the same canary-process that we currently use for those to also test the go binaries on gokrazy, if we deem that appropriate.

I will send a PR to have the packer add the build-data to the image for now.

stapelberg, 2017-04-20:
> For user-services I'd like to built some canarying in (as alluded to in my earlier comment).

Can you elaborate on how this canarying would work?

> Choosing a name like this very clearly says that this is an unsupported usecase and not supposed to work, which is contrary to my wishes (but might be your intention).

It being “not supposed to work” is too strong a statement to make. I’d compare it to the “gokrazy” name itself: we want to convey that you shouldn’t base your serious business™ on this idea. But see below:

> I think upstream CI should already serve as a canary.

Yeah, I think you’re right about this, at least with regards to the kernel/firmware. The user services remain, and — as per my question above — I’m not entirely clear yet how their canarying would work. I think if we find a good solution for that aspect, we can probably make in-place upgrades a first-class feature.

We might think about different channels (like the CoreOS beta and stable channels) to limit breakage to a small set of users. In other words, if my Raspberry Pi breaks, I’m not going to be very angry and probably in a good position to fix it before it breaks other people’s devices.

> I think the best way to provide this would be to provide an appropriately compiled go tool via CI, like the kernel and firmware repos. As a bonus, we could have the same canary-process that we currently use for those to also test the go binaries on gokrazy, if we deem that appropriate.

Agreed.

stapelberg, 2017-04-20:
That all sounds pretty good. Thanks!

chlunde, 2020-04-21:
Would it be possible to
1. Download the kernel to a new location (a/b kernel) *without* activating it
2. Set up a watchdog event to reset the PI in a few minutes
3. kexec the new kernel
4. On successful DHCP (or some other criteria), disable watchdog and flip the bytes to point to the new partition

The watchdog is neat but, it would be nice even without that

stapelberg, 2020-04-21:
This can work for the root file system (which uses the a/b scheme), but not for the kernel (which resides on the FAT boot partition which does not use an a/b scheme).

andig, 2021-01-04:
> As discussed offline, it would be great, if gokrazy would support full auto-updates (driven by gokrazy itself).

I'm wondering if we can trim the problem to a smaller size to make progress?
- We already have external updates from the grokrazy updater (including build+update).
- The next step would be updating the root fs from a built image. This could be done by a bundled client application/ service, doesn't need to be gokrazy. At this stage clients can self-update.
- Including the build process in the previous step allows going from source to updated appliance.
- Additional steps are a) updating kernel and b) some sort of update discovery if it needs be universally controlled from gokrazy side.

Maybe I'm misunderstanding the issue, but what would be missing to move up to step 2 (update from built image/ packaged service)?

Merovius, 2021-01-04:
I should probably clarify that I didn't actually do anything about this and I don't think it's super likely I will in the future - so if anyone wants to work on it, go ahead.

stapelberg, 2021-01-04:
> This can work for the root file system (which uses the a/b scheme), but not for the kernel (which resides on the FAT boot partition which does not use an a/b scheme).

Actually, re-reading the issue, I think I missed the point in this reply: one can kexec into a new kernel in an arbitrary location, so we could have /boot/vmlinuz.canary which would be booted into and then renamed into /boot/vmlinuz from the target system.

> * The next step would be updating the root fs from a built image. This could be done by a bundled client application/ service, doesn't need to be gokrazy. At this stage clients can self-update.

A third-party package which updates the root file system can readily be built. It would use the same API as the existing gokrazy tooling (HTTP requests). If you provided images via your CI, you could just pull them onto devices automatically.

You can update any combination of partitions: either root, or root and boot (or even boot only, but that seems the least interesting option).

andig, 2021-01-04:
> Actually, re-reading the issue, I think I missed the point in this reply: one can kexec into a new kernel in an arbitrary location, so we could have /boot/vmlinuz.canary which would be booted into and then renamed into /boot/vmlinuz from the target system.

Thats unfortunately beyond my *nix foo.

@stapelberg how could one upgrade just the root given a built image? Would you prefer the /update endpoint or using gokrazy/tools directly?

stapelberg, 2021-01-05:
> **Update** seems using `updater.Target.StreamTo()` might be a good approach.

Yes, see https://github.com/gokrazy/tools/blob/fdd90fc6817876e08b352fae84f2a2794524ccc0/cmd/gokr-updater/updater.go#L77 for the sequence of API calls for an update.

andig, 2021-01-05:
Found one more hurdle: the packer returns the `rootSize` in https://github.com/gokrazy/tools/blob/fdd90fc6817876e08b352fae84f2a2794524ccc0/cmd/gokr-packer/packer.go#L300. How would I determine root size for a pre-built image? A helper to read partitions (and their sizes) from file would be needed?

andig, 2021-01-05:
And then there is:

    package github.com/andig/evcc/cmd/dl
            cmd/dl/dl.go:13:2: use of internal package github.com/gokrazy/internal/httpclient not allowed
    package github.com/andig/evcc/cmd/dl
            cmd/dl/dl.go:14:2: use of internal package github.com/gokrazy/internal/updater not allowed

Would it hurt to un-private these packaged? I could provide the PR.

andig, 2021-01-05:
Here's what I got so far to achieve "level 2":

- [x] make gokrazy/internal usable: https://github.com/andig/gokrazy-internal
- [x] transfer built image to applicance
- [x] transparently handle gzipped files on client side
- [ ] add httpclient gzip compression using https://github.com/go-chi/chi/blob/master/middleware/compress.go (golang HTTP server should be able to handle zipped requests?)
- [x] provide upload progress https://github.com/gokrazy/internal/pull/8
- [ ] discover used amount of root partition according to https://github.com/gokrazy/gokrazy/issues/25#issuecomment-754556378

It's basically working although the update process is slow (2 minutes without user filesystem on rpi3/wifi) due to needing to transfer (and write) 500MB.

@stapelberg do you have any hint how to determine used root filesystem size from squashfs? Worst case I'd need to extract that from the build process and append it to the filename.

stapelberg, 2021-01-06:
> to have the actual _used_ part of the partition.

I suppose we could in theory figure out how big each file system is, but it seems cleaner to store the size as metadata.

That said, when you make the packer write separate files for boot/root file systems, you only get the actually used parts, no? Only writing a full image has unused parts at all. Does your updater need a full image (why?), or could it just deal with the individual boot/part file systems?

andig, 2021-01-08:
To summarize: level "2", i.e. updating from guest application, is done:

[image]

stapelberg, 2021-01-08:
Very nice! Thanks for sharing and organizing this effort.

Can you share a link to your code and any other documentation you might have?

We should probably reference it on our website in some form.

andig, 2021-01-08:
Its part of a bigger application: https://github.com/andig/evcc (EV charge controller). The updater is here https://github.com/andig/evcc/pull/580/files#diff-820034361b625ccaf3681004ef5474b488726f61385baf9071e128ce3b5d80e6, including streaming a gzipped rootfs file from Github assets. The UI is bootstrap with vue.js integrated via websockets.

stapelberg, 2021-01-10:
AFAICT, you’re only updating the root partition: https://github.com/andig/evcc/blob/b78e30af6b3623abdf38d618cec04046de56eeb1/server/updater/gokrazy.go#L83

You should also update the boot partition, otherwise the devices will be stuck on the same old kernel version forever.

andig, 2021-01-11:
> You should also update the boot partition, otherwise the devices will be stuck on the same old kernel version forever.

For time being thats ok for me. 

To update multiple partitions the challenge is that you'd either need to prepare and publish them separately (as I did with the rootfs) or build an image and add meta data. The meta is needed if you want to restrict the amount of data to download and push to the device. It is not easily detectable from the built image and simpler to save upon build.