gokrazy/gokrazy #27: running from OSX

gedw99, 2017-05-01:
just trying this out form OSX.

Setcaps is not available, and it might be why the update to the SD card was refused ?

using diskutil to make sure i hit the right disk :)

````
x-MacBook-Pro:bin apple$ diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *251.0 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage Ohne Titel              250.1 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3

/dev/disk1 (internal, virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS Ohne Titel             +249.8 GB   disk1
                                 Logical Volume on disk0s2
                                 3877FA65-5A90-469F-B4AD-0D12C262693B
                                 Unencrypted

/dev/disk2 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *15.9 GB    disk2
   1:             Windows_FAT_16 RECOVERY                1.3 GB     disk2s1
   2:                      Linux                         33.6 MB    disk2s5
   3:             Windows_FAT_32 boot                    69.2 MB    disk2s6
   4:                      Linux                         14.6 GB    disk2s7

x-MacBook-Pro:bin apple$ sudo setcap
Password:
sudo: setcap: command not found
x-MacBook-Pro:bin apple$ sudo setcap
sudo: setcap: command not found
x-MacBook-Pro:bin apple$ go get github.com/gokrazy/hello
x-MacBook-Pro:bin apple$ gokr-packer -overwrite=/dev/disk2 github.com/gokrazy/hello
2017/05/01 17:05:17 packer.go:225: installing [github.com/gokrazy/hello]
2017/05/01 17:05:18 gotool.go:60: getting incomplete packages [github.com/gokrazy/gokrazy/cmd/dhcp github.com/gokrazy/gokrazy/cmd/ntp]
2017/05/01 17:05:34 packer.go:123: partitioning /dev/disk2
2017/05/01 17:05:34 packer.go:285: open /dev/disk2: permission denied

````

stapelberg, 2017-05-01:
Thanks for giving gokrazy a shot.

I’m closing this issue in favor of issue #22, as the latter contains more details/pointers already.

closed 2017-05-01 by stapelberg

gedw99, 2017-05-01:
@stapelberg neither of those issue help..

Have a look and see what you think..
If i can sort this out we can write a golang cmd exec to automate it for others and stop the mayhem :)

bash:
```

diskutil list

# at this point i assume its /dev/disk2
# ! Make sure you dont use wrong disk :)

DISK=/dev/disk2
GOPROJ=github.com/gokrazy/hello

# unmount disk so we get full access
sudo diskutil unmount $DISK
sudo diskutil unmountDisk $DISK

## zero out disk 
#dd if=/dev/zero of=$DISK bs=1m count=1024

# deploy new OS with golang program
sudo gokr-packer -overwrite=$DISK $GOPROJ

````

output from bash:
````
x-MacBook-Pro:gokrazy apple$ ./t-updatesd.sh 
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *251.0 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage Ohne Titel              250.1 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3

/dev/disk1 (internal, virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS Ohne Titel             +249.8 GB   disk1
                                 Logical Volume on disk0s2
                                 3877FA65-5A90-469F-B4AD-0D12C262693B
                                 Unencrypted

/dev/disk2 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                                                   *15.9 GB    disk2

disk2 was already unmounted or it has a partitioning scheme so use "diskutil unmountDisk" instead
Unmount of all volumes on disk2 was successful
2017/05/01 17:23:27 packer.go:225: installing [github.com/gokrazy/hello]
2017/05/01 17:23:32 packer.go:123: partitioning /dev/disk2
2017/05/01 17:23:32 packer.go:285: open /dev/disk2: permission denied
x-MacBook-Pro:gokrazy apple$ ./t-updatesd.sh 
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *251.0 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage Ohne Titel              250.1 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3

/dev/disk1 (internal, virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS Ohne Titel             +249.8 GB   disk1
                                 Logical Volume on disk0s2
                                 3877FA65-5A90-469F-B4AD-0D12C262693B
                                 Unencrypted

/dev/disk2 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                                                   *15.9 GB    disk2

Password:
disk2 was already unmounted or it has a partitioning scheme so use "diskutil unmountDisk" instead
Unmount of all volumes on disk2 was successful
2017/05/01 17:24:36 packer.go:225: installing [github.com/gokrazy/hello]
warning: "github.com/gokrazy/gokrazy/cmd/..." matched no packages
2017/05/01 17:24:39 gotool.go:60: getting incomplete packages [github.com/gokrazy/hello]
warning: "github.com/gokrazy/gokrazy/cmd/..." matched no packages
warning: "github.com/gokrazy/gokrazy/cmd/..." matched no packages
/tmp/gokr-packer056488109/init.go:9:2: cannot find package "github.com/gokrazy/gokrazy" in any of:
	/usr/local/Cellar/go/1.8.1/libexec/src/github.com/gokrazy/gokrazy (from $GOROOT)
	/Users/apple/go/src/github.com/gokrazy/gokrazy (from $GOPATH)
2017/05/01 17:24:42 packer.go:246: exit status 1
x-MacBook-Pro:gokrazy apple$ go get github.com/gokrazy/gokrazy
# github.com/gokrazy/gokrazy/internal/iface
../../../../../../github.com/gokrazy/gokrazy/internal/iface/iface.go:108: undefined: syscall.SIOCADDRT
../../../../../../github.com/gokrazy/gokrazy/internal/iface/iface.go:122: undefined: syscall.SIOCDELRT

````

gedw99, 2017-05-01:
@stapelberg 
i wonder if we can make a docker image that will do this all and so eliminate these cross OS issues?

stapelberg, 2017-05-01:
There’s no need to throw docker into the mix. 

The easiest route for macOS users is to generate an SD card image using e.g. `-overwrite=/tmp/gokrazy.img` (as opposed to specifying a device name) and then writing that image to an SD card with your tool of choice.

gedw99, 2017-05-01:
ah i saw that somewhere.
But could not work out how to make the img in the first place.

stapelberg, 2017-05-01:
With regards to the output you provided in comment https://github.com/gokrazy/gokrazy/issues/27#issuecomment-298354691:

The tool fails because macOS mounts the partitions we create immediately, not allowing us to overwrite their contents. I *think* this is the same as issue #22, but with a different error message — if it’s not the same issue, it surely is related.

Installing the `github.com/gokrazy/gokrazy` package fails because that package is not what you think it is: it contains code (such as the init system and userland) to be run on the device, not on the host operating system. On the host (macOS), you should only install the packer; the go tool will install all of its dependencies as necessary. The gokrazy package itself will be cross-compiled by the packer. If you re-installed the packer as root, you should be able to use it as root, if you so wish. Note that it’s not required to be root, unless you need direct device access, which — as we’ve established — currently is not convenient on macOS.

> But could not work out how to make the img in the first place.

Just replace the `-overwrite` flag accordingly. Let me know if you can’t get it to work.

gedw99, 2017-05-01:
i think this is wrong:
https://github.com/gokrazy/tools#alternative-creating-an-sd-card-image

it shoudl say:
"sudo dd if=/tmp/full.img of=/dev/sdb bs=1M"
NOT
"sudo dd if=/tmp/full.fat of=/dev/sdb bs=1M"

??

gedw99, 2017-05-01:
also this gets the following error on OSX:
````
dd: bs: illegal numeric value
```

SO you need to replace the "bs=1M" shoudl be "bs=1m"

gedw99, 2017-05-01:
gokr-packer \
  -overwrite=/tmp/full.img \
  -target_storage_bytes=2147483648 \
  $GOPROJ

How to work out the size in bytes  i need to use ?
In Diskutil it says my sd card is 15.93 GB

stapelberg, 2017-05-01:
> How to work out the size in bytes i need to use ?
> In Diskutil it says my sd card is 15.93 GB

Sorry, not sure. You can just use 15G as a lower bound and waste 1G of your SD card. Note that if you’re okay with limiting your persistent storage partition to only 500M, you can just create a 2G image, which will be faster to write.

For subsequent updates, use the `-update` flag, or create images for the boot/root partition and overwrite them individually; it’ll be much faster.

> Its taking forever to write to the SD card.....

https://superuser.com/questions/631592/why-is-dev-rdisk-about-20-times-faster-than-dev-disk-in-mac-os-x

gedw99, 2017-05-02:
i saw that on the rasp Pi web site about rdisk....

There might be  a bug in the multipart approach:
````
gokr-packer -overwrite_boot=/tmp/boot.fat -overwrite_root=/tmp/root.fat $GOPROJ
sudo dd if=/tmp/boot.fat of=$DISK bs=1m
sudo dd if=/tmp/root.fat of=$DISK bs=1m
````

The root.fat never is created.

gedw99, 2017-05-02:
@stapelberg see above :)

stapelberg, 2017-05-02:
Filed https://github.com/gokrazy/tools/issues/5 for that issue, thanks.