gokrazy/gokrazy #28: PATH, GOROOT, GOPATH

YaguraStation, 2017-05-16:
I am sure there is a simpler and more elegant way of handling things than I am currently doing. And it probably should be referenced in the README.md somewhere, but what is best practice when using `gokr-packer` with `sudo `in regards to environment variables?

stapelberg, 2017-05-16:
The best practice is to not use sudo at all, but rather grant the required capabilities to the gokr-packer binary, as https://gokrazy.org/quickstart.html does via `sudo setcap CAP_SYS_ADMIN,CAP_DAC_OVERRIDE=ep $(go env GOPATH)/bin/gokr-packer`.

Why does that not work for you?

YaguraStation, 2017-05-16:
TLDR: That simply answers the question, sorry. 

It's a long - now uninteresting story - from my first experiment with `gokrazy`, setting an environment up for _root_, then reverting back to user and messed up `secure_path` and `env_keep` in the sudoers file... eventually libcap2-bin not working anymore. Probably based on some initial misunderstanding of a few errors, the role of setcap, and necessity of elevated rights. Works on a "clean" machine just fine now.

closed 2017-05-16 by YaguraStation

tv42, 2018-03-27:
Permanently giving some random binary `CAP_SYS_ADMIN` is a really bad idea. That allows it to e.g. extract secrets from from kernel memory and from disk. Transient privileges (for this run only) are much safer.

stapelberg, 2018-03-27:
How exactly can one grant transient privileges?

tv42, 2018-03-27:
You're already doing that with sudo, for the setcap run.

stapelberg, 2018-03-27:
Using sudo for the gokr-packer command results in compilation happening as root, which is not intended.

If you dislike granting CAP_SYS_ADMIN, don’t specify a privileged device to the -overwrite= flag and do your own copying.

tv42, 2018-03-27:
Sounds like a great reason to separate building from writing to SD card. *shrug* If this doesn't seem important to you, I'll just have to take note that you don't care and move along.