gokrazy/gokrazy #33: Enable GPIO

BakeRolls, 2017-08-21:
I'd like to use Gobot with gokrazy. I've added `CONFIG_GPIO_SYSFS=y` to the GPIO block in [build.go](https://github.com/gokrazy/kernel/blob/master/cmd/gokr-build-kernel/build.go#L207) to get `/sys/class/gpio/export` and `/sys/class/gpio/unexport`. However, gokrazy is not allowed to r/w.

I am neither certain on how to allow it to the user nor if it'd be a good idea.

stapelberg, 2017-08-21:
Currently, everything is running as root, so I don’t think we’re facing a permissions issue here. Not sure why the writes are denied — I’d recommend using https://github.com/gokrazy/breakglass with busybox to do some interactive debugging.

stapelberg, 2020-05-29:
I have submitted https://github.com/google/periph/commit/03ec8a96c847582929169b5db6974137d8fe9b8c and https://github.com/google/periph/commit/dcd940c1af2f50b331bbf0e6d1f6748d8a069fd8 a while ago, so maybe https://periph.io/ can be used on gokrazy for GPIO control now.

We should double check and write up a little post for the user guide: https://github.com/gokrazy/gokrazy/issues/51

maruel, 2020-05-29:
I should probably make a release once you confirmed your changes helped, as I haven't released in a while.

stapelberg, 2020-06-01:
Verified periph.io works with the Raspberry Pi 3 Model B Rev 1.2:

I consulted https://pinout.xyz/, then connected jumper wires to:
* pin 18 (BCM24)
* pin 20 (GND)
…and connected the jumper wires to a multimeter.

Running a slightly modified version of https://periph.io/device/led/:
1. Added log messages so the user knows what’s going on.
2. Changed the toggle interval from 500ms to 5s for easier measurement with a multimeter (as compared to an LED).
3. Changed GPIO to match my setup.

```go
package main

import (
	"log"
	"time"

	"periph.io/x/periph/conn/gpio"
	"periph.io/x/periph/host"
	"periph.io/x/periph/host/rpi"
)

func doGPIO() error {
	log.Printf("Loading periph.io drivers")
	// Load periph.io drivers:
	if _, err := host.Init(); err != nil {
		return err
	}
	log.Printf("Toggling GPIO forever")
	t := time.NewTicker(5 * time.Second)
	for l := gpio.Low; ; l = !l {
		log.Printf("setting GPIO to %v", l)
		// Lookup a pin by its location on the board:
		if err := rpi.P1_18.Out(l); err != nil {
			return err
		}
		<-t.C
	}
	return nil
}

func main() {
	if err := doGPIO(); err != nil {
		log.Fatal(err)
	}
}
```

With current `periph.io/x/periph v3.6.2+incompatible` (last release, I assume), this fails with a message about gpiomem and sysfs being unavailable.

With latest master `periph.io/x/periph v3.6.3-0.20200529014005-4d322a6db74f+incompatible`, the program works as expected: I can measure 0V and 3.3V with the multimeter, alternating every 5 seconds.

So, yes, please do a new release :)

maruel, 2020-06-03:
I pushed v3.6.3 to fix you. I'll try to not forget to write release notes tomorrow.
As soon as I'm done with panicparse v2, I'll get back to periph v4. I've learned valuable lessons about go.mod and versioning with panicparse so I'm glad I tried to do it there first.

closed 2020-06-15 by stapelberg