gokrazy/gokrazy #37: Use a randomly generated password by default

Chaz6, 2018-03-27:
Please make the image use a random password generated at write time instead of a common password known to potential attackers. Default passwords are the source of multiple CVE's and seems at odds with the primary goal of reducing attack surface.

stapelberg, 2018-03-27:
What makes you think we use a common password? We already generate a password the first time you use gokrazy.

closed 2018-03-27 by stapelberg

Chaz6, 2018-03-27:
A quote from the page at https://gokrazy.org/quickstart.html:-

```
To interact with the device, gokrazy provides a web interface reachable at:

	http://gokrazy:secretpw@gokrazy/
```

There is no indication that it generates a random password anywhere on the page currently.

YaguraStation, 2018-03-27:
I think for anybody who actually ran `gokr-packer` IRL, this was always very clear.

Chaz6, 2018-03-27:
Great thanks!

stapelberg, 2018-03-29:
Done with https://github.com/gokrazy/gokrazy.github.io/commit/4a6852ccda7d4271a2dc5fef438ad01fa1c96deb