gokrazy/gokrazy #39: Examples refer to private repo https://github.com/stapelberg/mediaserver

tv42, 2018-03-27:
https://github.com/gokrazy/gokrazy

```
gokr-packer \
  -overwrite_init=$(go env GOPATH)/src/github.com/stapelberg/mediaserver/cmd/init/init.go \
  github.com/gokrazy/hello
```

(Note that the package must result in a binary called “init”.)

Then, edit the github.com/stapelberg/mediaserver package to your liking. When done, pack an image with your own init package:

```
gokr-packer \
  -init_pkg=github.com/stapelberg/mediaserver \
  -overwrite=/dev/sdb \
  github.com/gokrazy/hello
```

That repo is private/non-existent.

stapelberg, 2018-03-27:
Yeah, it’s a placeholder for the repo name of the application you’d like to create on gokrazy.

closed 2018-03-27 by stapelberg

tv42, 2018-03-27:
That's not at all clear from the readme.

commit a470a80a (2018-03-29): README: expand on the “changing init behavior” example

fixes #39