gokrazy/gokrazy #40: Support 32-bit ARM

probonopd, 2018-04-06:
I know that it is currently not implemented, but please do support 32-bit ARM architectures. It would allow us to run this on target hardware such as the Orange Pi Zero, which costs less than a fifth of a Raspberry Pi 3 and is more than enough for many workloads.

Thanks for your consideration.

http://www.orangepi.org/orangepizero/

probonopd, 2018-04-06:
Assuming that I have a kernel that has all the required modules built in, how much do you estimate the remaining work to be? You make it very clear that you don't want to do it, but do you see any major showstoppers for someone else to do it? And if someone else did it, would you merge it?

stapelberg, 2018-04-06:
> Assuming that I have a kernel that has all the required modules built in, how much do you estimate the remaining work to be?

https://github.com/gokrazy/firmware/issues/15 contains a checklist, with most items not being kernel-related.

> You make it very clear that you don't want to do it, but do you see any major showstoppers for someone else to do it?

If the above list looks feasible, I don’t think there are any showstoppers. That’s not a guarantee that they don’t exist, though, just that I don’t know of any upfront.

> And if someone else did it, would you merge it?

No. I’ll only immediately merge what I can support.

If a separate effort proves that it has sufficient long-term interest from the person who does the porting and maintenance, and from users, I’d consider merging it and collaborating. I’m thinking on the order of months here, though.