gokrazy/gokrazy #44: containers or sandboxes for non-go code

bzub, 2018-11-30:
I've been working on a way to run OCI containers via runc in gokrazy in a way that the non-go/cgo binaries can be updated with the usual gokr tools. Unfortunately that probably means packing binaries into go byte slices. My specific use case is to enable wifi via IWD. Related: #13

I don't expect this to be officially supported as-is, however i wanted to see if there would be any interest in officially supporting the use of containers or a sandboxing tool like bubblewrap? To get runc or any code that uses libcontainer running the only cgo exception would be https://github.com/opencontainers/runc/tree/master/libcontainer/nsenter

If that small exception was made for libcontainer, gokrazy users would be able to pull container images, configure sandboxing features and run anything they like with some added safety. Perhaps it could be seen as a more sophisticated breakglass.

stapelberg, 2018-11-30:
Have you seen https://gokrazy.org/prototyping.html (just checking, it seems relevant)? :)

> I've been working on a way to run OCI containers via runc in gokrazy in a way that the non-go/cgo binaries can be updated with the usual gokr tools. Unfortunately that probably means packing binaries into go byte slices. My specific use case is to enable wifi via IWD. Related: #13

Cool! Where would the binaries come from, though, i.e. what keeps them up to date?

> I don't expect this to be officially supported as-is, however i wanted to see if there would be any interest in officially supporting the use of containers or a sandboxing tool like bubblewrap? To [...]

It seems like a small enough cgo exception indeed. What do you mean by “officially supporting”? I’m fine with creating a repository under the gokrazy org, so that interested users can easily discover it. Will you commit to maintaining it? Personally, I like to know there’s 0 C (whether for or, more importantly, _inside_ any containers), so I don’t want to burden myself with maintaining it :)

> Perhaps it could be seen as a more sophisticated breakglass.

Yeah, I can very well imagine using (and recommending) this for breakglass’s purposes, i.e. interactive introspection/development/debugging sessions.

Does that make sense?

bzub, 2018-11-30:
> Cool! Where would the binaries come from, though, i.e. what keeps them up to date?

I'm using a build script that pulls the latest Arch Linux packages and makes a container image. So that would have to be run periodically to update a git repo. https://gist.github.com/bzub/9323bd58894609a10058229d1217dc60

> It seems like a small enough cgo exception indeed. What do you mean by “officially supporting”? I’m fine with creating a repository under the gokrazy org, so that interested users can easily [...]

I would maintain it. Not looking to put any extra burden on anyone. :) The only change to the existing gokrazy org repos should be the exception for importing the libcontainer package which uses cgo.

> Yeah, I can very well imagine using (and recommending) this for breakglass’s purposes, i.e. interactive introspection/development/debugging sessions.

> Does that make sense?

Yah I think we're on the same page. Once the above mentioned libcontainer exception is added to gokrazy tools I should be pretty close to a functional iwd/dbus container + code for use with gokrazy that can also act as an example for users that want that kind of prototyping workflow. If I get time I'll submit a PR for the exception.

stapelberg, 2022-04-26:
I just published https://gokrazy.org/packages/docker-containers/, which outlines how to use the gokrazy podman package to run containers.

I’m not 100% sure where to take it from here, i.e. what most people will want to use this for, or what would be most useful to provide in addition. But perhaps this is already a good start, and I’m curious to see what y’all build with it :)