gokrazy/gokrazy #46: [Question] Using PiCam with gokrazy

kortschak, 2018-12-31:
Apologies if this is not the correct forum; I don't see a mailing list.

I am interested in using gokrazy with a PiCam module. I don't see any cam support userland here, so I am suspecting that I would need to use raspivid and its C runtime dependencies as described in the [prototyping](https://gokrazy.org/prototyping.html) page. Is this correct?

stapelberg, 2018-12-31:
AFAICT, the `raspivid` tool uses the VideoCore decoding support via the `/dev/vchq` device node, but the driver providing that device node has not yet made it into the upstream kernel properly (it’s currently in staging: https://github.com/torvalds/linux/tree/v4.20/drivers/staging/vc04_services/bcm2835-camera).

Hence, the first step is to get that kernel driver somewhat working (or wait until it does).

Then, you’re likely correct, as the existing Go libraries for interfacing the Raspberry Pi camera all rely either on `raspivid` itself, or on the C API it uses:

* https://github.com/dklassen/picamera (stub using the [mmal](https://github.com/raspberrypi/userland/tree/master/interface/mmal) C API)
* https://github.com/technomancers/piCamera (calling raspivid)
* https://github.com/dhowden/raspicam (calling raspivid)

If you want quick results, attaching a USB webcam seems like the easier path.

cweagans, 2021-02-02:
Is there some mechanism that one would be able to use to provide a custom kernel that had this stuff built in? and/or would it make sense to use what the rpi foundation is providing so that all of the hardware bits work OOTB? If this were documented somewhere, I think it would be enough to let people start experimenting with the pi zero, right?

stapelberg, 2021-02-02:
If you have further questions, please open a new issue. This is not related to camera support at all AFAICT.

> Is there some mechanism that one would be able to use to provide a custom kernel that had this stuff built in?

You can set the `-kernel_package` flag of the `gokr-packer` to pick up whichever kernel you like, yes.

> and/or would it make sense to use what the rpi foundation is providing so that all of the hardware bits work OOTB?

It might make sense for you, but for gokrazy itself, the idea is to use upstream linux and pull in new versions automatically, so that security updates are available ASAP. The Raspberry Pi foundation’s kernel isn’t updated as quickly, so it’s not attractive to use their kernel.

> If this were documented somewhere, I think it would be enough to let people start experimenting with the pi zero, right?

Quoting from https://gokrazy.org/quickstart/:

> Many people assume the Raspberry Pi Zero W could be used as well. Unfortunately, it uses the older BCM2835 SoC, which is not arm64, so gokrazy won’t run on it.

See also https://www.reddit.com/r/golang/comments/5xgf8u/gokrazy_a_purego_userland_for_your_raspberry_pi_3/dei7b2m/ for why we only focus on a few device models.

andig, 2021-02-02:
>> Many people assume the Raspberry Pi Zero W could be used as well. Unfortunately, it uses the older BCM2835 SoC, which is not arm64, so gokrazy won’t run on it.

> See also https://www.reddit.com/r/golang/comments/5xgf8u/gokrazy_a_purego_userland_for_your_raspberry_pi_3/dei7b2m/ for why we only focus on a few device models.

As I read this- even if not supported by gokrazy- it would be possible to build and run e.g. on an arm7 as supported by Go if a custom kernel is supplied.

The actual build process is in https://github.com/gokrazy/kernel/blob/master/cmd/gokr-build-kernel/build.go. It seems it should be possible to tweak this for building Raspbian kernels but I'm lacking the linux skills to do so.

cweagans, 2021-02-02:
> If you have further questions, please open a new issue. This is not related to camera support at all AFAICT.

Apologies, I see now that I skipped including a few steps of my thought process in my comment. I was asking in the context of being able to use the aforementioned kernel support for the camera. If there's some way to use a custom kernel, then there's a way to get camera support. As a nice side effect, if there's a way to use a custom kernel, then there is _also_ a way to experiment with the pi zero.

> See also reddit.com/r/golang/comments/5xgf8u/gokrazy_a_purego_userland_for_your_raspberry_pi_3/dei7b2m for why we only focus on a few device models.

(OT, apologies)

Actually, _I'm_ thinking about taking this on separately. I'm certainly not suggesting that _you_ take this on, though I would be grateful for your occasional help and thoughts on the topic if you're willing to provide that kind of input. I am also an open source maintainer and I'm all too familiar with time constraints. Good on you for setting boundaries like this :)

My interest in the Zero is that it's a very low power, low profile device that is easy to build devices around + there isn't necessarily an expectation of networking support. The full size Pi is nice for more robust applications, but way overkill for what I'm working on.

stapelberg, 2021-02-02:
Okay, if you want to maintain the raspberry pi foundation kernel, that’d be a good option for people I think.

We can list your repository on https://gokrazy.org/platforms/ under a separate section.

oliverpool, 2022-04-02:
There is now a repo which provides the raspberry pi foundation kernel for gokrazy, documented on the website: https://gokrazy.org/platforms/#community

The challenge of embedding `raspivid` remains. Maybe [compiling with static linking](https://gokrazy.org/prototyping/#go-software-not-written-for-gokrazy-grafana) could work ? (I don't have a PiCam to test)

psanford, 2023-07-15:
I was able to get this working by using the rpi kernel. I captured video using the v4l interface and a statically complied ffmpeg.

Hermitter, 2025-11-30:
I'm not sure if this is already solved, but does [gokrazy's rpi kernel](https://github.com/gokrazy/kernel.rpi) support pi cameras?

I tried loading some of the relevant looking kernel modules, but I wasn't able to have mine show as a video device under `/dev`