gokrazy/gokrazy #47: SPI not available

schubter, 2019-05-10:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [x] gokrazy/rpi3b+
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

i've added to the config.txt in gokrazy/kernel the following lines

dtparam=spi=on
dtoverlay=spi-bcm2835

also tried with dtoverlay=spi-bcm2708

no spi devices are available
/tmp/breakglass156884305 # ls -al /dev/sp*
ls: /dev/sp*: No such file or directory

also nothing in the kernel log
/tmp/breakglass156884305 # dmesg | grep spi

## Expected behavior

something along the lines like this in dmesg
[   4.813403] bcm2708_spi 20204000.spi: SPI Controller at 0x20204000 (irq 80)

and the corresponding devices in /dev

stapelberg, 2019-05-10:
It might well be that our kernel config just doesn’t contain the required modules.

schubter, 2019-05-10:
i've checked the config here: https://github.com/gokrazy/kernel/blob/master/cmd/gokr-build-kernel/build.go#L22

schubter, 2019-05-10:
i've now tried different config combinations with no avail.

CONFIG_SPI_SPIDEV=y
CONFIG_SPI_MASTER=y

any advice?

stapelberg, 2019-05-10:
Not off the top of my head. I might be able to get a chance to look at this tomorrow.

schubter, 2019-05-10:
thank you, and also thanks for the incredible quick response :)

stapelberg, 2019-05-13:
Try setting `CONFIG_SPI_BCM2835=y`

stapelberg, 2019-05-13:
Just realized that we already set that option. Nevermind.

stapelberg, 2019-05-13:
See spidev.c: https://github.com/torvalds/linux/blob/d7a02fa0a8f9ec1b81d57628ca9834563208ef33/drivers/spi/spidev.c#L47-L51

From that, I’m thinking you might just need to create the device node yourself. In gokrazy, we don’t have udev or similar and just rely on devtmpfs instead. It sounds like spidev device nodes are not created in devtmpfs.

I’m not too familiar with SPI, but perhaps these commands do the trick?
```
# mknod /dev/spidev0.0 c 153 0
# mknod /dev/spidev0.1 c 153 1
```

If that doesn’t do the trick, can you check whether `/sys/bus/spi/devices/` contains anything?

schubter, 2019-05-13:
thank you. i'll try it with mknod.

```
/tmp/breakglass843182288 # ls -al /proc/device-tree/soc/
---
drwxr-xr-x    2 0        0                0 May 12 23:38 spi@7e204000
drwxr-xr-x    2 0        0                0 May 12 23:38 spi@7e215080
drwxr-xr-x    2 0        0                0 May 12 23:38 spi@7e2150c0
...
```

```
cat spi@7e2*/status`
disabled
disabled
disabled
```

schubter, 2019-05-13:
i'v just created the device nodes but access seg faults

stapelberg, 2019-05-13:
With what message/backtrace specifically?

schubter, 2019-05-13:
```
panic: runtime error: invalid memory address or nil pointer dereference
  [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xbff30]
  
  goroutine 1 [running]:
  periph.io/x/periph/host/sysfs.(*SPI).CLK(0x0, 0x16e9e0, 0x0)
  	/Users/aschubert/.go/src/periph.io/x/periph/host/sysfs/spi.go:148 +0x20
  main.main()
  	/Users/aschubert/.go/src/github.com/schubter/box/cmd/box/box.go:153 +0x6a8
  2019/05/13 09:00:35 gokrazy: exit status 2
  2019/05/13 09:00:36 gokrazy: attempt 261, starting [/user/box]
  panic: runtime error: invalid memory address or nil pointer dereference
  [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xbff30]
``

maruel, 2019-05-13:
Ok then it definitely sounds like a bug in periph. I wanted to test it for years but never took the time to do so. @schubter can you take a deeper look?

schubter, 2019-05-13:
yeah i will, most likely later today. i wanted to try accessing the spi device without periph to see if the device node is actually working.
anything else you'd recommend to look

maruel, 2019-05-13:
Looking at https://github.com/google/periph/blob/v3.5.0/host/sysfs/spi.go#L148

I suspect you are calling CLK() on a nil pointer. Are you checking for errors?

schubter, 2019-05-13:
i do, currently i have this block in there: https://github.com/google/periph/blob/ac8d5328f2b147b7ca36d6c37415f944560d5ec5/conn/spi/spireg/example_test.go#L58 before doing anything else, which access CLK

maruel, 2019-05-13:
The bug is here: https://github.com/google/periph/blob/ac8d5328f2b147b7ca36d6c37415f944560d5ec5/conn/spi/spireg/example_test.go#L70

It's missing a break.

schubter, 2019-05-13:
i've just noticed the line in the other log: Failed to open: sysfs-spi: open /dev/spidev0.0: no such device or address

```
Driver Loaded:  bcm283x-gpio
  Driver Loaded:  rpi
  Driver Loaded:  sysfs-gpio
  Driver Loaded:  sysfs-i2c
  Driver Loaded:  sysfs-led
  Driver Loaded:  sysfs-spi
  Driver Loaded:  sysfs-thermal
  Driver Skipped:  allwinner-gpio-pl: A64 CPU not detected
  Driver Skipped:  allwinner-gpio: Allwinner CPU not detected
  Driver Skipped:  pine64: pine64 board not detected
  3 Model B v1.2, B+ from /proc/device-tree/system/linux,revision
  SPI ports available:
  - /dev/spidev0.0
    SPI0.0
Failed to open: sysfs-spi: open /dev/spidev0.0: no such device or address
```

schubter, 2019-05-13:
seems like the driver for bcm2835 is still missing
```
/sys/bus/spi/drivers:
total 0
drwxr-xr-x    9 0        0                0 May 13 15:25 .
drwxr-xr-x    4 0        0                0 May 13 15:25 ..
drwxr-xr-x    2 0        0                0 May 13 15:26 cros-ec-spi
drwxr-xr-x    2 0        0                0 May 13 15:26 ds3234
drwxr-xr-x    2 0        0                0 May 13 15:26 m25p80
drwxr-xr-x    2 0        0                0 May 13 15:26 mmc_spi
drwxr-xr-x    2 0        0                0 May 13 15:26 rt5677
drwxr-xr-x    2 0        0                0 May 13 15:26 spidev
drwxr-xr-x    2 0        0                0 May 13 15:26 wm8753
```

stapelberg, 2019-05-13:
The driver for bcm2835 should be built-in. See also `zcat /proc/config.gz | grep SPI_`

maruel, 2019-05-13:
In my last comment, I meant `continue`, not `break`.

schubter, 2019-05-13:
```
/tmp/breakglass843182288 # zcat /proc/config.gz | grep SPI_
# CONFIG_MTD_SPI_NAND is not set
CONFIG_MTD_SPI_NOR=y
CONFIG_MTD_SPI_NOR_USE_4K_SECTORS=y
# CONFIG_SPI_CADENCE_QUADSPI is not set
CONFIG_SPI_HISI_SFC=y
# CONFIG_SPI_MTK_QUADSPI is not set
CONFIG_SCSI_SPI_ATTRS=y
# CONFIG_SPI_DEBUG is not set
CONFIG_SPI_MASTER=y
CONFIG_SPI_MEM=y
# CONFIG_SPI_ALTERA is not set
CONFIG_SPI_ARMADA_3700=y
# CONFIG_SPI_AXI_SPI_ENGINE is not set
CONFIG_SPI_BCM2835=y
CONFIG_SPI_BCM2835AUX=y
CONFIG_SPI_BCM_QSPI=y
# CONFIG_SPI_BITBANG is not set
# CONFIG_SPI_CADENCE is not set
# CONFIG_SPI_DESIGNWARE is not set
# CONFIG_SPI_FSL_LPSPI is not set
# CONFIG_SPI_FSL_QUADSPI is not set
CONFIG_SPI_NXP_FLEXSPI=y
# CONFIG_SPI_GPIO is not set
# CONFIG_SPI_IMX is not set
# CONFIG_SPI_FSL_SPI is not set
# CONFIG_SPI_FSL_DSPI is not set
CONFIG_SPI_MESON_SPICC=m
CONFIG_SPI_MESON_SPIFC=y
# CONFIG_SPI_MT65XX is not set
# CONFIG_SPI_OC_TINY is not set
# CONFIG_SPI_OMAP24XX is not set
CONFIG_SPI_ORION=y
CONFIG_SPI_PL022=y
# CONFIG_SPI_PXA2XX is not set
CONFIG_SPI_ROCKCHIP=y
# CONFIG_SPI_RSPI is not set
# CONFIG_SPI_QCOM_QSPI is not set
CONFIG_SPI_QUP=y
# CONFIG_SPI_QCOM_GENI is not set
CONFIG_SPI_S3C64XX=y
# CONFIG_SPI_SC18IS602 is not set
# CONFIG_SPI_SH_MSIOF is not set
# CONFIG_SPI_SH_HSPI is not set
# CONFIG_SPI_SIFIVE is not set
# CONFIG_SPI_SPRD is not set
# CONFIG_SPI_SPRD_ADI is not set
# CONFIG_SPI_SUN4I is not set
# CONFIG_SPI_SUN6I is not set
# CONFIG_SPI_MXIC is not set
CONFIG_SPI_TEGRA114=y
CONFIG_SPI_TEGRA20_SFLASH=y
CONFIG_SPI_TEGRA20_SLINK=y
CONFIG_SPI_THUNDERX=y
# CONFIG_SPI_UNIPHIER is not set
# CONFIG_SPI_XCOMM is not set
# CONFIG_SPI_XILINX is not set
# CONFIG_SPI_XLP is not set
# CONFIG_SPI_ZYNQMP_GQSPI is not set
CONFIG_SPI_SPIDEV=y
# CONFIG_SPI_LOOPBACK_TEST is not set
# CONFIG_SPI_TLE62X0 is not set
# CONFIG_SPI_SLAVE is not set
# CONFIG_CXD2880_SPI_DRV is not set
```

indeed it is, anything else that needs to be enabled?

schubter, 2019-05-14:
good news, after hacking around in the kernel a bit more i've now got the spi devices working
```
  Driver Loaded:  bcm283x-dma
  Driver Loaded:  bcm283x-gpio
  Driver Loaded:  rpi
  Driver Loaded:  sysfs-gpio
  Driver Loaded:  sysfs-i2c
  Driver Loaded:  sysfs-led
  Driver Loaded:  sysfs-spi
  Driver Loaded:  sysfs-thermal
  Driver Skipped:  allwinner-gpio-pl: A64 CPU not detected
  Driver Skipped:  allwinner-gpio: Allwinner CPU not detected
  Driver Skipped:  pine64: pine64 board not detected
  3 Model B v1.2, B+ from /proc/device-tree/system/linux,revision
  SPI ports available:
  - /dev/spidev0.0
    0
    SPI0.0
    CLK : SPI0_CLK(GPIO11)  MOSI: SPI0_MOSI(GPIO10)  MISO: SPI0_MISO(GPIO9)  CS  : SPI0_CS0(GPIO8)- /dev/spidev1.0
    1
    SPI1.0
    CLK : SPI1_CLK(GPIO21)  MOSI: SPI1_MOSI(GPIO20)  MISO: SPI1_MISO(GPIO19)  CS  : SPI1_CS0(GPIO18)- /dev/spidev2.0
    2
    SPI2.0
    CLK : SPI2_CLK(GPIO42)  MOSI: SPI2_MOSI(GPIO41)  MISO: SPI2_MISO(GPIO40)  CS  : SPI2_CS0(GPIO43)
```

i need to run some more tests and disable some configs again and see what affects what and should be able to provide a patch at the end of the week.

schubter, 2019-05-15:
@maruel is it ok to contact you, about some things i've noticed in periph and had to change to get the things running, to discuss a few things quickly or do you want me to open an issue in the periph repository?

maruel, 2019-05-15:
Sure, either over gopher slack or via email.
For periph specific fixes, then opening an issue there is probably a good idea.

chlunde, 2020-04-17:
Hi, you are discussing using `dtparam=spi=on`. As far as I can see this has no effect on the upstream kernel. In raspbian it seems toggle the spi flag under `__override__` in the `dtsi` file:

Reference: `dtparam` for spi in raspbian kernel (not upstream): https://github.com/raspberrypi/linux/blame/rpi-4.19.y/arch/arm/boot/dts/bcm2708-rpi.dtsi#L83

Example PR adding another dtparam: https://github.com/raspberrypi/linux/commit/f16e91dad2af9d57aef477cc1f522040353849f5

But in gokrazy, dumping `bcm2710-rpi-3-b-plus.dtb` from https://github.com/gokrazy/kernel shows it does not contain any overrides:
```bash
fdtdump bcm2710-rpi-3-b-plus.dtb | grep __overrides__
# no hits..
```
So I think this means `dtparam` is not supported.  Is it possible to emulate the effect of `dtparam` from user space through sysfs or similar? I tried to poke around but did not find a way.

```bash
/tmp/breakglass514323355 # echo -n 1 > /sys/firmware/devicetree/base/soc/spi@7e2150c0/status 
sh: can't create /sys/firmware/devicetree/base/soc/spi@7e2150c0/status: Permission denied
```

schubter, 2020-04-17:
This issue can be closed once the PR on the kernel repo is merged. I have fixed it there and it’s working for me perfectly fine since then. 

https://github.com/gokrazy/kernel/pull/181

schubter, 2020-04-19:
resolved with https://github.com/gokrazy/kernel/commit/f686f39050a21745ce357e8e4ff66320ce77f74a

closed 2020-04-19 by schubter

x86ed, 2023-10-06:
@schubter @chlunde do you have example go code and kernel config for using the SPI interface in gokrazy?

ralfuslongus, 2025-07-15:
I am sorry, but i think the issue is not solved.
I use  gokrazy on rpi3b with this LCD Display:
[https://www.waveshare.com/1.3inch-lcd-hat.htm](url)
I like to use it by SPI
When connect by ssh i get:
model: Raspberry Pi 3 Model B Rev 1.2
build: 2025-07-15T16:59:25+02:00

`dmesg|grep -i spi` prints nothing
`ls /dev/sp*` prints No such file or directory

if i use
`mknod /dev/spidev0.0 c 153 0`
`mknod /dev/spidev0.1 c 153 1`
like suggested by stapelberg i get the 2 devices, but if i use them with periph.io/x/conn/v3/spi/spigeg i get the same error as schubter before the fix: `sysfs-spi: open /dev/spidev0.0: no such device or address`

What do i miss? I think the fix should be included in the current version i use, doesn't it?
I would love to use Gokrazy the first time, but if i can't use SPI i can't use it.
Thank you
(should i reopen the issue?)