gokrazy/gokrazy #48: platform: add support for the rpi4b

stapelberg, 2019-06-24:
The Raspberry Pi 4 was just released: https://www.raspberrypi.org/blog/raspberry-pi-4-on-sale-now-from-35/

I have ordered one to make sure gokrazy works on it.

stapelberg, 2019-06-30:
We’re currently blocked on https://github.com/raspberrypi/linux/issues/3032 (arm64 kernel support for the Pi 4 being in early stages).

Not sure which patches are required to make this work.

stapelberg, 2019-07-04:
To summarize progress on the referenced issue: when manually placing the dtb file, armstub and raspberrypi/linux kernel image on the SD card, and limiting the memory to 1G, I can get some output on the serial console.

I can’t get an upstream kernel to print anything on the serial console, though.

I suppose it’ll be a while before the necessary changes are upstreamed.

vianpl, 2019-07-08:
> To summarize progress on the referenced issue: when manually placing the dtb file, armstub and raspberrypi/linux kernel image on the SD card, and limiting the memory to 1G, I can get some output on [...]

Can you try adding gpu_freq=500 to config.txt? AFAIK the VPU clock and UART are derived from the same PLL and I suspect the firmware may be altering the frequencies behind your back.

stapelberg, 2019-12-01:
Side note: dividuum described how to do firmware updates at https://news.ycombinator.com/item?id=21674550, which is important for better thermals and power usage: https://www.raspberrypi.org/blog/thermal-testing-raspberry-pi-4/

stapelberg, 2019-12-01:
https://github.com/lategoodbye/rpi-zero/issues/43 has a good overview of the current status. Looks like Linux 5.6 can be the earliest we could possibly see good-enough support.

stapelberg, 2020-06-19:
The EEPROM updates are now in place, and I have been using a Pi 4 to develop and to run my home automation for weeks without any trouble.

Hence, the Pi 4 is now officially a target platform of https://gokrazy.org/!

closed 2020-06-19 by stapelberg