gokrazy/gokrazy #49: Bluetooth support

joneskoo, 2019-08-25:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [X] gokrazy/rpi3b+
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I installed github.com/joneskoo/ruuvi-prometheus  which depends on Bluetooth hardware. It failed to start.

After breaking glass, and looking at the kernel, it looks like Bluetooth is intentionally disabled.

I could not find an option to disable serial console and enable Bluetooth instead. I know they can't be enabled at the same time on the hardware UART, possibly not at all.

On Alpine Linux, I know I need to use bluez btattach to initialize the Bluetooth interface on serial port. Online, I saw pointers that it could be enabled without bluez using dts overlay: https://github.com/RoEdAl/alarm-bluetooth-raspberrypi

## Expected behavior

Raspberry Pi 3B+ Bluetooth supported (as option)

joneskoo, 2019-08-27:
I assume it will be a requirement to keep the serial port console? So
Bluetooth implicitly needs to be behind some flag?

If I find time to hack it to enable it as a proof of concept, would you
consider including such an option? I’m not sure if/when I’ll find such time
but will not do it if we know it won’t be included as option.
-- 
Joonas Kuorilehto

stapelberg, 2019-08-27:
> After breaking glass, and looking at the kernel, it looks like Bluetooth is intentionally disabled.

No, it just never was intentionally enabled, and we started from a minimal kernel config :)

I don’t use bluetooth much, so I’m not sure what needs to be done to make this work.

Maybe you could try a few things and share how it’s going? Sounds like you have a few good pointers already (dts changes, freeing up the serial port).

stapelberg, 2019-08-27:
> I assume it will be a requirement to keep the serial port console?

Yes.

> So Bluetooth implicitly needs to be behind some flag?

Possibly. Our serial console setup is a bit different from the Raspberry Pi’s default setup, so I can’t say which changes are required.

> If I find time to hack it to enable it as a proof of concept, would you consider including such an option?

I will definitely consider it. Bluetooth support would be great to have.

joneskoo, 2022-03-05:
Sharing here we discussed this on Twitter: https://twitter.com/joneskoo_yx/status/1500152012536795136?s=21

What I tried so far:

```diff
diff --git a/cmd/gokr-rebuild-kernel/kernel.go b/cmd/gokr-rebuild-kernel/kernel.go
index 6834ed5..688d465 100644
--- a/cmd/gokr-rebuild-kernel/kernel.go
+++ b/cmd/gokr-rebuild-kernel/kernel.go
@@ -43,11 +43,11 @@ var dockerFileTmpl = template.Must(template.New("dockerfile").
 var patchFiles = []string{
        "0001-Revert-add-index-to-the-ethernet-alias.patch",
        // serial
-       "0101-expose-UART0-ttyAMA0-on-GPIO-14-15-disable-UART1-tty.patch",
-       "0102-expose-UART0-ttyAMA0-on-GPIO-14-15-disable-UART1-tty.patch",
-       "0103-expose-UART0-ttyAMA0-on-GPIO-14-15-disable-UART1-tty.patch",
-       "0104-bluetooth.patch",
-       "0105-bluetooth-cm4.patch",
+       //"0101-expose-UART0-ttyAMA0-on-GPIO-14-15-disable-UART1-tty.patch",
+       //"0102-expose-UART0-ttyAMA0-on-GPIO-14-15-disable-UART1-tty.patch",
+       //"0103-expose-UART0-ttyAMA0-on-GPIO-14-15-disable-UART1-tty.patch",
+       //"0104-bluetooth.patch",
+       //"0105-bluetooth-cm4.patch",
        // spi
        "0201-enable-spidev.patch",
        // logo
```

Then I downloaded alpine and chrooted to it via breakglass, roughly:

```
% curl -O https://dl-cdn.alpinelinux.org/alpine/v3.15/releases/aarch64/alpine-minirootfs-3.15.0-aarch64.tar.gz
% tar cf alpine.tar alpine-minirootfs-3.15.0-aarch64.tar
% go run github.com/gokrazy/breakglass/cmd/breakglass -debug_tarball_pattern alpine.tar gokrazy
# tar xf alpine.tar
# mount -o bind /dev dev
# chroot .
/ # mount -o proc proc /proc
/ # mount -o sysfs sys /sys
/ # echo nameserver 8.8.8.8 > /etc/resolv.conf
/ # apk add bluez
fetch https://dl-cdn.alpinelinux.org/alpine/v3.15/main/aarch64/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/v3.15/community/aarch64/APKINDEX.tar.gz
(1/15) Installing dbus-libs (1.12.20-r4)
(2/15) Installing expat (2.4.6-r0)
(3/15) Installing dbus (1.12.20-r4)
Executing dbus-1.12.20-r4.pre-install
Executing dbus-1.12.20-r4.post-install
(4/15) Installing libffi (3.4.2-r1)
(5/15) Installing libintl (0.21-r0)
(6/15) Installing libblkid (2.37.4-r0)
(7/15) Installing libmount (2.37.4-r0)
(8/15) Installing pcre (8.45-r1)
(9/15) Installing glib (2.70.1-r0)
(10/15) Installing json-c (0.15-r1)
(11/15) Installing ncurses-terminfo-base (6.3_p20211120-r0)
(12/15) Installing ncurses-libs (6.3_p20211120-r0)
(13/15) Installing readline (8.1.1-r0)
(14/15) Installing eudev-libs (3.2.11_pre1-r0)
(15/15) Installing bluez (5.63-r0)
Executing busybox-1.34.1-r3.trigger
Executing dbus-1.12.20-r4.trigger
OK: 15 MiB in 29 packages
/ # apk add bluez-deprecated
(1/1) Installing bluez-deprecated (5.63-r0)
Executing busybox-1.34.1-r3.trigger
OK: 17 MiB in 30 packages
/ # hciconfig
Can't open HCI socket.: Address family not supported by protocol
/ # btattach -B /dev/ttyS0 -P bcm -S 115200 -N
Attaching Primary controller to /dev/ttyS0
Failed to flush serial port: I/O error
No controller attached
/ # btattach -B /dev/ttyS1 -P bcm -S 115200 -N
Attaching Primary controller to /dev/ttyS1
Failed set serial line discipline: Invalid argument
No controller attached
```

I'm not sure if I read this right. I came across this on search. http://events17.linuxfoundation.org/sites/events/files/slides/serdev-elce-2017-2.pdf  . If I understand right Linux might have something called serdev that could attach these without userland? I don't think bluez will be acceptable in gokrazy but if we can have an command line / alternate kernel to enable Bluetooth instead of serial or something, that's great. 

If it's not possible the same actually describes what btattach does.

`btattach -B /dev/ttyAMA0 -P bcm -S 115200 -N` is supposed to register the broadcom bluetooth HCI via the serial port so it shows as Bluetooth device (hci0). It needs to keep running to keep the serial discipline attached or hci0 goes away.

I wouldn't use bluez - I use http://gitlab.com/jtaimisto/bluewalker in my application which uses HCI socket directly from Go (for Bluetooth LE).

My goal is to get hci0 showing up and working with my https://github.com/joneskoo/ruuvi-prometheus .

Would appreciate if you can build for me a kernel with the Bluetooth serial device /dev/ttyAMA0 showing up. Thanks for bearing with me and my noise on Twitter.

stapelberg, 2022-03-05:
In addition to reverting, I think you’ll need to set the `-serial_console=ttyS1,115200` flag when running `gokr-packer`.

With your diff for the patches, and the following additional diff, I get a bunch of bluetooth loadable kernel modules:

```diff
--- i/cmd/gokr-build-kernel/build.go
+++ w/cmd/gokr-build-kernel/build.go
@@ -131,6 +131,40 @@ CONFIG_CFG80211=y
 CONFIG_BRCMFMAC=m
 CONFIG_NLMON=y
 
+# For Bluetooth:
+CONFIG_BT=m
+CONFIG_BT_RFCOMM=m
+CONFIG_BT_RFCOMM_TTY=y
+CONFIG_BT_BNEP=m
+# CONFIG_BT_BNEP_MC_FILTER is not set
+# CONFIG_BT_BNEP_PROTO_FILTER is not set
+# CONFIG_BT_CMTP is not set
+CONFIG_BT_HIDP=m
+
+#
+# Bluetooth device drivers
+#
+CONFIG_BT_HCIBTUSB=m
+CONFIG_BT_HCIBTSDIO=m
+CONFIG_BT_HCIUART=m
+CONFIG_BT_HCIUART_H4=y
+CONFIG_BT_HCIUART_BCSP=y
+CONFIG_BT_HCIUART_ATH3K=y
+CONFIG_BT_HCIUART_LL=y
+CONFIG_BT_HCIUART_BCM=y
+CONFIG_BT_HCIBCM203X=m
+CONFIG_BT_HCIBPA10X=m
+CONFIG_BT_HCIBFUSB=m
+CONFIG_BT_HCIDTL1=m
+CONFIG_BT_HCIBT3C=m
+CONFIG_BT_HCIBLUECARD=m
+CONFIG_BT_HCIBTUART=m
+CONFIG_BT_HCIVHCI=m
+CONFIG_BT_MRVL=m
+CONFIG_BT_MRVL_SDIO=m
+CONFIG_BT_ATH3K=m
+CONFIG_BT_WILINK=m
+
 # For using github.com/vishvananda/netlink
 CONFIG_NETFILTER_NETLINK_QUEUE=y
 CONFIG_XFRM_USER=y

```

`modprobe hci_uart` results in the following `dmesg` output:

```
[   68.590993] hci_uart: Unknown symbol hci_free_dev (err -2)
[   68.596621] hci_uart: Unknown symbol btbcm_finalize (err -2)
[   68.602393] hci_uart: Unknown symbol btbcm_initialize (err -2)
[   68.608319] hci_uart: Unknown symbol hci_alloc_dev_priv (err -2)
[   68.614505] hci_uart: Unknown symbol btbcm_read_pcm_int_params (err -2)
[   68.621271] hci_uart: Unknown symbol baswap (err -2)
[   68.626343] hci_uart: Unknown symbol bt_info (err -2)
[   68.631507] hci_uart: Unknown symbol bt_err (err -2)
[   68.636607] hci_uart: Unknown symbol hci_unregister_dev (err -2)
[   68.642721] hci_uart: Unknown symbol hci_recv_frame (err -2)
[   68.648500] hci_uart: Unknown symbol btbcm_write_pcm_int_params (err -2)
[   68.655300] hci_uart: Unknown symbol btbcm_set_bdaddr (err -2)
[   68.661226] hci_uart: Unknown symbol btbcm_check_bdaddr (err -2)
[   68.667336] hci_uart: Unknown symbol hci_register_dev (err -2)
[   68.673256] hci_uart: Unknown symbol bt_warn (err -2)
[   68.678393] hci_uart: Unknown symbol __hci_cmd_sync (err -2)
[   68.684153] hci_uart: Unknown symbol hci_recv_diag (err -2)
[   69.037786] Bluetooth: Core ver 2.22
[   69.041506] NET: Registered PF_BLUETOOTH protocol family
[   69.046895] Bluetooth: HCI device and connection manager initialized
[   69.053363] Bluetooth: HCI socket layer initialized
[   69.058336] Bluetooth: L2CAP socket layer initialized
[   69.063483] Bluetooth: SCO socket layer initialized
[   69.083587] Bluetooth: HCI UART driver ver 2.3
[   69.088151] Bluetooth: HCI UART protocol H4 registered
[   69.093362] Bluetooth: HCI UART protocol BCSP registered
[   69.098794] Bluetooth: HCI UART protocol LL registered
[   69.104005] Bluetooth: HCI UART protocol ATH3K registered
[   69.109780] Bluetooth: HCI UART protocol Broadcom registered
[   69.119904] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[   69.127529] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[   69.263075] uart-pl011 3f201000.serial: no DMA platform data
[   69.513470] Bluetooth: hci0: BCM: chip id 107
[   69.518274] Bluetooth: hci0: BCM: features 0x2f
[   69.524446] Bluetooth: hci0: BCM4345C0
[   69.528294] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[   69.534508] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[   69.541263] Bluetooth: hci0: BCM: 'brcm/BCM4345C0.hcd'
[   69.546506] Bluetooth: hci0: BCM: 'brcm/BCM.hcd'
[   69.568142] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.574931] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.581723] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.588519] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.595321] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.602134] Bluetooth: hci0: Failed to read codec capabilities (-56)
```

stapelberg, 2022-03-05:
Extracting `BCM4345C0.hcd` from https://packages.debian.org/sid/bluez-firmware and throwing it into `/home/michael/go/src/github.com/gokrazy/wifi/_gokrazy/extrafiles/lib/firmware/brcm` makes the module go one step further:

```
[   59.705506] Bluetooth: hci0: BCM: chip id 107
[   59.710301] Bluetooth: hci0: BCM: features 0x2f
[   59.716464] Bluetooth: hci0: BCM4345C0
[   59.720294] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[   59.734634] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.hcd' Patch
[   60.527125] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+
[   60.533015] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0342
[   60.559895] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   60.566778] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   60.573676] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   60.580542] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   60.587439] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   60.594309] Bluetooth: hci0: Failed to read codec capabilities (-56)

```

I’m not sure if a /dev/hci* should be present at that point. Perhaps with using the kernel’s `devtmpfs` we just don’t get device nodes for anything late in boot, and we’d need udev or similar instead? Not 100% sure about that.

stapelberg, 2022-03-05:
Yep, looks like Bluetooth works! (Despite `/dev/hci*` not being present.)

I can successfully run a program that uses the `github.com/paypal/gatt` package, which in turn uses `	fd, err := socket.Socket(socket.AF_BLUETOOTH, syscall.SOCK_RAW, socket.BTPROTO_HCI)`

joneskoo, 2022-03-06:
Thanks! That's really promising and indeed the socket did open and I can see AMA0 in dmesg. But didn't really work. I suspect there's something wrong with how the serial is configured. I'm not sure if this is because 5.16 kernel. I should compare with Alpine Linux edge which has 5.16 too.

Apparently the "Failed to read codec capabilities" is harmless and fixed in 5.17 but for me, when trying to use the HCI my breakglass SSH hangs (could be just WiFi breaking which would be possible since they're on the same chip?). The fbdev is fine and indicates 90% idle CPU so I guess it's just WiFi.

I also hit the "Unknown symbol" errors you mentioned and tried to fix that by adding kmod package to the kernel builder so dependencies list gets generated. That didn't appear to help, however together with using alpine userland to load the modules I didn't get the symbols thing any more.

I have no good ideas how to debug this further. I'm guessing it has something to do with the serial configuration that's causing a problem when trying to actually use it.

stapelberg, 2022-03-06:
Can you post your current `dmesg`? Also, try using ethernet instead of wifi so that your SSH doesn’t break :)

joneskoo, 2022-03-06:
Oh and hci* are not supposed to be in dev, they're Bluetooth interfaces similar to network interfaces and don't have a device file. If you want to test functionality, maybe you want to try https://gitlab.com/jtaimisto/bluewalker which has a CLI and can scan for Bluetooth LE devices which with all likelihood you have all around you (smart watches, mice, phones etc.)

```
[    6.082489] random: crng init done
[    6.210877] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): No External EEPROM. Setting MAC Speed
[   26.015483] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   26.024558] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.bin failed with error -2
[   26.233703] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   26.252042] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[   45.329470] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   67.449455] hci_uart: Unknown symbol hci_free_dev (err -2)
[   67.455098] hci_uart: Unknown symbol btbcm_finalize (err -2)
[   67.460874] hci_uart: Unknown symbol btbcm_initialize (err -2)
[   67.466807] hci_uart: Unknown symbol hci_alloc_dev_priv (err -2)
[   67.472984] hci_uart: Unknown symbol btbcm_read_pcm_int_params (err -2)
[   67.479753] hci_uart: Unknown symbol baswap (err -2)
[   67.484815] hci_uart: Unknown symbol bt_info (err -2)
[   67.489950] hci_uart: Unknown symbol bt_err (err -2)
[   67.494998] hci_uart: Unknown symbol hci_unregister_dev (err -2)
[   67.501100] hci_uart: Unknown symbol hci_recv_frame (err -2)
[   67.506869] hci_uart: Unknown symbol btbcm_write_pcm_int_params (err -2)
[   67.513669] hci_uart: Unknown symbol btbcm_set_bdaddr (err -2)
[   67.519593] hci_uart: Unknown symbol btbcm_check_bdaddr (err -2)
[   67.525693] hci_uart: Unknown symbol hci_register_dev (err -2)
[   67.531613] hci_uart: Unknown symbol bt_warn (err -2)
[   67.536749] hci_uart: Unknown symbol __hci_cmd_sync (err -2)
[   67.542510] hci_uart: Unknown symbol hci_recv_diag (err -2)
[   67.566351] hci_uart: Unknown symbol hci_free_dev (err -2)
[   67.571962] hci_uart: Unknown symbol btbcm_finalize (err -2)
[   67.577731] hci_uart: Unknown symbol btbcm_initialize (err -2)
[   67.583657] hci_uart: Unknown symbol hci_alloc_dev_priv (err -2)
[   67.589828] hci_uart: Unknown symbol btbcm_read_pcm_int_params (err -2)
[   67.596594] hci_uart: Unknown symbol baswap (err -2)
[   67.601653] hci_uart: Unknown symbol bt_info (err -2)
[   67.606785] hci_uart: Unknown symbol bt_err (err -2)
[   67.611833] hci_uart: Unknown symbol hci_unregister_dev (err -2)
[   67.617936] hci_uart: Unknown symbol hci_recv_frame (err -2)
[   67.623703] hci_uart: Unknown symbol btbcm_write_pcm_int_params (err -2)
[   67.630504] hci_uart: Unknown symbol btbcm_set_bdaddr (err -2)
[   67.636426] hci_uart: Unknown symbol btbcm_check_bdaddr (err -2)
[   67.642526] hci_uart: Unknown symbol hci_register_dev (err -2)
[   67.648459] hci_uart: Unknown symbol bt_warn (err -2)
[   67.653598] hci_uart: Unknown symbol __hci_cmd_sync (err -2)
[   67.659355] hci_uart: Unknown symbol hci_recv_diag (err -2)
[   68.015984] Bluetooth: Core ver 2.22
[   68.019744] NET: Registered PF_BLUETOOTH protocol family
[   68.025137] Bluetooth: HCI device and connection manager initialized
[   68.031605] Bluetooth: HCI socket layer initialized
[   68.036564] Bluetooth: L2CAP socket layer initialized
[   68.041705] Bluetooth: SCO socket layer initialized
[   68.057387] Bluetooth: HCI UART driver ver 2.3
[   68.061982] Bluetooth: HCI UART protocol H4 registered
[   68.067216] Bluetooth: HCI UART protocol BCSP registered
[   68.072670] Bluetooth: HCI UART protocol LL registered
[   68.077888] Bluetooth: HCI UART protocol ATH3K registered
[   68.083719] Bluetooth: HCI UART protocol Broadcom registered
[   68.093987] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[   68.101872] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[   68.232841] uart-pl011 3f201000.serial: no DMA platform data
[   68.474531] Bluetooth: hci0: BCM: chip id 107
[   68.479280] Bluetooth: hci0: BCM: features 0x2f
[   68.485369] Bluetooth: hci0: BCM4345C0
[   68.489181] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[   68.499455] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.hcd' Patch
[   69.296119] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+
[   69.302012] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0342
[   69.327424] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.334322] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.341116] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.347882] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.354680] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   69.361466] Bluetooth: hci0: Failed to read codec capabilities (-56)
```

This is what I have now

stapelberg, 2022-03-06:
Can you include the start of the `dmesg`, too, so that we can see the kernel command line?

For me, bluewalker works on a Raspberry Pi 3B+ (Raspberry Pi 3 Model B Plus Rev 1.3) and on a Raspberry Pi 4, even when connected via WiFi at the same time:
```
/tmp/breakglass1640076671 # /tmp/gok-run/bluewalker -device=hci0
.................................
Found 33 devices:
Device da:db:71:a2:xx:zz,random (static) (RSSI:-74 dBm; last seen Mar  6 18:09:31):
Events:Connectable undirected
Advertising Data Structures:
	Flags: [00000110](LE General Discoverable,BR/EDR not supported)
	Service Data: UUID: 0x180a, Data: 0x3504c37403
	16 Bit Service Class UUID: Data: 0x63fd
[…]
```

joneskoo, 2022-03-06:
I'd really like to have a copy of that known working kernel if you don't mind :)

```
/tmp/breakglass722962976 # dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.16.12 (gokrazy@docker) (aarch64-linux-gnu-gcc (Debian 6.3.0-18) 6.3.0 20170516, GNU ld (GNU Binutils for Debian) 2.28) #1 SMP PREEMPT Wed Mar 1 20:57:29 UTC 2017
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] NUMA: NODE_DATA [mem 0x3720eb40-0x37210fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 20 pages/cpu s41688 r8192 d32040 u81920
[    0.000000] pcpu-alloc: s41688 r8192 d32040 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: video=HDMI-A-1:1920x1200M@60 dma.dmachans=0x7ff5 bcm2709.boardrev=0xa020d3 bcm2709.serial=0xbd07a252 bcm2709.uart_clock=48000000 bcm2709.disk_led_gpio=29 bcm2709.disk_led_active_low=0 smsc95xx.macaddr=B8:27:EB:07:A2:52 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=60c24cc1-f3f9-427a-8199-0581fd1c0001/PARTNROFF=1 init=/gokrazy/init rootwait panic=10 oops=panic
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 842984K/970752K available (18560K kernel code, 3790K rwdata, 11224K rodata, 7744K init, 537K bss, 62232K reserved, 65536K cma-reserved)
```

`gokr-packer -update=yes -tls=self-signed -hostname gokrazy.m7.yx.fi -serial_console=disabled github.com/gokrazy/fbstatus github.com/gokrazy/wifi github.com/gokrazy/mkfs github.com/gokrazy/breakglass github.com/gokrazy/serial-busybox github.com/joneskoo/ruuvi-prometheus`

It's a bit tricky to try with Ethernet, I'd have to either hack breakglass to listen on public IPs too or do a bit of shuffling due to lack of ports. (My cabling setup is a bit work in progress so I only have WiFi and raw Internet indoors now.)

stapelberg, 2022-03-06:
Sure:
1. Extract https://t.zekjur.net/2022-03-06-bluetooth-kernel.tar.bz2 into your gokrazy/kernel directory
2. Add `-serial_console=ttyS1,115200` to your `gokr-packer` invocation.
3. Log in via breakglass
4. Run `modprobe hci_uart` 
5. Run `bluewalker -device=hci0`

joneskoo, 2022-03-06:
Thanks. Indeed it works but if I run my application which literally uses the bluewalker code, the WiFi stops working (huge packet loss). I'm going to play with it more but I did get working frames received with bluewalker so we are very close. Not a victory yet, but really appreciate how much you helped.

For what it's worth my kernel wasn't any worse looks like.

I'll keep fighting this, next I will check what happens if I copy the known working firmwares for both WiFi and Bluetooth from Alpine. Another theory is the serial baudrate which we did not set for ttyAMA0. Will need to find where it comes from. 115200 is what works for me in Alpine for sure. I've run this for some years on Alpine with WiFi and Bluetooth on 3B+ and Zero W so it can work. There I use the btattach but no opportunity for that here as kernel connects the serial somehow internally.

I'll let you know if I locate the culprit.

stapelberg, 2022-03-07:
I wonder if your WiFi network is possibly running on a channel that has more overlap with bluetooth than mine? Mine is on channel 11.

joneskoo, 2022-03-07:
I tried replacing the wifi package firmwares from known working alpine but that doesn't seem to have any effect.

1. I am able to perfectly run my application from breakglass and it received the expected measurements. There was no obvious issue at this point to breakglass ssh session (over WiFi).
2. When I opened the gograzy web page in my browser, I can no longer ping the gokrazy instance. After closing the browser and waiting for a while, I was able to ssh by breakglass again.
3. I can also `curl https://gokrazy:***@gokrazy.m7.yx.fi/ -k -v` without triggering this problem. So either some resources on the gokrazy web dashboard triggers the problem or it's related to amount of data transferred. I haven't seen similar issue on my Alpine instance which as said I've used extensively, and I can also transfer 500 kB/s+ no problem over ssh from the Pi Zero W on 2.4 GHz WiFi.

Bluetooth uses 2400 to 2483.5 MHz frequency range with narrow channels using adaptive frequency hopping. This range overlaps with WiFi channels 1-13 so seems unlikely to me this would explain such catastrophic "doesn't ping at all for tens of seconds" failure more, especially given it works perfectly on Alpine. (I can do backups of the instance over SSH over WiFi with no issues at all).

To confirm if data transfer causes the problem on gokrazy; I did some testing via breakglass. I connected the gokrazy directly to the Internet so I can ping it via wired interface (but no breakglass because of listen address).

1. If I execute `find /` (to print a lot of text), there is no problem after reboot (over WiFi).
2. I start `bluewalker -device hci0 &` and then immediately do "find /" via breakglass ; this triggers the problem. The find output starts to become delayed and ping starts to fail. 
3. On the wired interface there is no interference.

[image]

stapelberg, 2022-03-07:
> * When I opened the gograzy web page in my browser, I can no longer ping the gokrazy instance. After closing the browser and waiting for a while, I was able to ssh by breakglass again.

Could it be that the device crashes and then reboots?

Can you connect a USB-to-serial adapter to see? Otherwise debugging is just poking in the dark :)

joneskoo, 2022-03-07:
It doesn't recover as shown by stable ping over wired. Also it recovers and uptime is not lost. I'll enable ssh access over public IP with breakglass.

joneskoo, 2022-03-07:
I replaced *.dtb from alpine (old version) and this fixed the problem. No more WiFi issues when I have bluewalker or my own application running. So we can somewhat confidently say that the remaining problem is with the dtbs.

Unfortunately the alpine dtb also doesn't enable the kernel serial device and I need to use btattach from alpine to attach the Bluetooth serial. So that still leaves two options for what is causing the problem: 1) the dtb has some hardware/pin configuration that prevents correct function of bluetooth co-existence 2) there's a difference between kernel attached serial bluetooth and btattach.

In any case, progress! And proof that it's not a problem with my WiFi environment. I'll boot alpine edge which has same kernel version and see if they use serdev kernel attached Bluetooth serial or do they still need btattach, and then test with those dtb instead.

joneskoo, 2022-03-07:
Ah, Alpine doesn't build kernel for 3.16 for rpi. At least in Alpine Linux 3.15 I have to use btattach. In any case I did more testing with Alpine 3.15 and on 2.4 GHz WiFi before initializing Bluetooth I get about 5 MB/s transfer rates and after touching Bluetooth once rates drop to 100 kB/s or so. That's perfectly fine for me (IoT-like appliance use case) and it didn't have the 60 seconds without responding to ping issue we see on Gokrazy.

In any case, I realize the WiFi/BT coexistence is only problem for 2.4 GHz and indeed when I changed to SSID on 5 GHz everything seems to be perfect. I normally run these low bandwidth devices on 2.4 GHz to keep radio free of legacy 802.11a devices to let 802.11ax clients get full performance.

I was hoping to get Raspberry Zero 2 W to replace my old Bluetooth receiver appliance (Zero W) with a gokrazy based Zero 2 W but that only has 2.4 GHz radio on it. To make that work I'd need the BT/WiFi combination to work. But at least now I'll be able to build a full proof of concept with 3B+.

Would we want to create a github.com/gokrazy/bluetooth for the Bluetooth similar to WiFi, that will load the kernel module and firmware? I can bootstrap the repo but I don't want to touch the Broadcom firmware.

- [ ] Create github.com/gokrazy/bluetooth that loads kernel module on start @joneskoo 
- [ ] Add firmware to github.com/gokrazy/bluetooth
- [ ] Bluetooth/WiFi coexistence issue with 2.4 GHz WiFi (fix or mark as known issue?)
- [ ] Provide kernel with Bluetooth support

Can you help with providing the kernel - either as option of github.com/gokrazy/kernel or as separate package?

joneskoo, 2022-03-08:
Interesting. Despite not loading firmware and getting errors, the Bluetooth interface still loads. Maybe it's good to provide the firmware but doesn't seem to be required for functionality?

```
[   27.616957] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   27.626199] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.bin failed with error -2
[   27.660390] Bluetooth: Core ver 2.22
[   27.664264] NET: Registered PF_BLUETOOTH protocol family
[   27.673392] Bluetooth: HCI device and connection manager initialized
[   27.683508] Bluetooth: HCI socket layer initialized
[   27.692285] Bluetooth: L2CAP socket layer initialized
[   27.701139] Bluetooth: SCO socket layer initialized
[   27.763286] Bluetooth: HCI UART driver ver 2.3
[   27.772260] Bluetooth: HCI UART protocol H4 registered
[   27.781583] Bluetooth: HCI UART protocol BCSP registered
[   27.790954] Bluetooth: HCI UART protocol LL registered
[   27.799603] Bluetooth: HCI UART protocol ATH3K registered
[   27.808865] Bluetooth: HCI UART protocol Broadcom registered
[   27.819026] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[   27.829650] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   27.830488] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[   27.852907] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[   27.995594] uart-pl011 3f201000.serial: no DMA platform data
[   28.250269] Bluetooth: hci0: BCM: chip id 107
[   28.262948] Bluetooth: hci0: BCM: features 0x2f
[   28.276895] Bluetooth: hci0: BCM4345C0
[   28.288495] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[   28.302549] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[   28.317174] Bluetooth: hci0: BCM: 'brcm/BCM4345C0.hcd'
[   28.328300] Bluetooth: hci0: BCM: 'brcm/BCM.hcd'
[   28.356024] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   28.370595] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   28.385067] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   28.399466] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   28.412693] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   28.425894] Bluetooth: hci0: Failed to read codec capabilities (-56)
[   31.714401] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

At least works for me now.

```
gokr-packer -update=yes -tls=self-signed -hostname gokrazy.m7.yx.fi -serial_console=disabled \
  github.com/gokrazy/mkfs \
  github.com/gokrazy/wifi \
  github.com/gokrazy/bluetooth \
  github.com/joneskoo/ruuvi-prometheus
```

With go.mod:

```
replace github.com/gokrazy/kernel => ./2022-03-06-bluetooth-kernel

replace github.com/gokrazy/bluetooth => github.com/joneskoo/gokrazy-bluetooth v0.0.0-20220308105909-da349f9d8353
```

[image]

stapelberg, 2022-03-09:
Thanks for sharing your findings!

It’s interesting that Bluetooth can work without the firmware file.

When searching the web, I can find plenty of reports of interference with Bluetooth and 2.4 GHz WiFi. People sometimes discover that their case incurs extra interference, or sometimes the power supply is too weak (I even read about someone swapping a 2A power supply for a 3A one and having fewer issues afterwards). Not sure if any of that applies to your setup.

Some people resort to using an external WiFi adapter or external Bluetooth adapter. Obviously that’s not attractive if the device works with another software configuration, but I’m not sure what we can do other than report these issues to the developers.

I can try enabling the Bluetooth modules in gokrazy/kernel, but that requires taking a careful look at the current serial patches and possibly breaking changes for existing users :-/. I also vaguely remember some serial interface changes between the Pi 3 and Pi 4, so this will be a longer investigation…

Having a gokrazy/bluetooth program similar to the gokrazy/wifi program sounds good to me. Your repo looks like a good start.

joneskoo, 2022-03-09:
Draft for the website doc in #116. I think rest is blocked until we have the kernel and gokrazy/bluetooth.

I have my appliance working so I'm all happy now, just the upstream will be nice to have eventually. Many thanks for all your help and support.

joneskoo, 2022-03-11:
@stapelberg Looks like we need more work for Zero 2 W.

```
4.1881691 ALSA device list:
4.1912491
10: uc4-hdmi
4.2559051 mmc1: SDHCI controller on 3f300000.mm [3/300000.mmcl using PIO
4.2688331 UFS: Mounted root (squashfs filesystem) readonly on device 179:2
4.2776111 deutmpfs: mounted
4.2897711 Freeing unused kernel memory: 7744K
4.294683] Run /gokrazy/init as init process
4.373997] random: fast init done
5.609176] EXT4-fs (mmcblk0p4): recovery complete
5.6164091 EXT4-fs (mmcblk0p4): mounted filesystem with ordered data mode. Opts: (null).
13.419888] random: crng init done
27.230088] Bluetooth: Core ver 2.22
27.2341701 NET: Registered P_BLUETOOTH protocol family
27.239811] Bluetooth: HI device and connection manager initialized
27.2465121 Bluetooth: HCI socket layer initialized
27.2516731 Bluetooth: L2CAP socket layer initialized
27.2570491 Bluetooth: SC socket layer initialized
27.3570741 Bluetooth: HCI UART driver ver 2.3
27.3618771 Bluetooth: HCI UART protocol H4 registered
27.367350] Bluetooth: HI UART protocol BCSP registered
27.373166] Bluetooth: HCI UART protocol LL registered
27.3787091 Bluetooth: HI UART protocol ATH3K registered
27.385000] Bluetooth: HCI UART protocol Broadcom registered
27.421100] hi uart_bcm serial0-0: supply ubat not found, using dummy regulator
27.4292221 hi art bcm serial0-0: supply uddio not found, using dummy regulator
27.566336] uart-pl011 3201000 .serial: no MA platform data
29.700288] Bluetooth: hci0: command Oxfc18 tx timeout
37.860296] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
37.867304] Bluetooth: hcio: Failed to set baudrate
39.876290] Bluetooth: hci0: command Ox0c03 tx timeout
48.100300] Bluetooth: hci0: BCM: Reset failed (-110)
```

I suspect conflict with other serial device.

stapelberg, 2022-03-17:
> I can try enabling the Bluetooth modules in gokrazy/kernel, but that requires taking a careful look at the current serial patches and possibly breaking changes for existing users :-/

I think I have a plan for how to change the serial situation with minimal disruption:

* [ ] make `gokrazy/gokrazy` create a `/dev/serial0` symlink
  * ttyAMA0 is /dev/serial0 if uart0 == serial0 (old kernel)
  * ttyS0 is /dev/serial0 if uart1 == serial0 (new kernel with dtb overlay) or uart1 not found (new kernel)
  * Raspberry Pi OS does this [via /etc/udev/rules.d/99-com.rules](https://unix.stackexchange.com/q/541684/181634)
  * software like [hmgo](https://github.com/stapelberg/hmgo) should be changed to use `/dev/serial0` in the long run
* [ ] make gokrazy create a `/dev/ttyAMA0` → `ttyS0` symlink if `ttyAMA0` is not found (it’s absent with the new kernel because it’s claimed by the kernel Bluetooth driver)
  * this means software like [hmgo](https://github.com/stapelberg/hmgo) does not need an update when we change the kernel
* [ ] switch gokr-packer from its default `-serial_console=ttyAMA0,115200` to `-serial_console=serial0,115200`
  * Raspberry Pi firmware rewrites serial0 to ttyAMA0,115200 with the old kernel
  * Raspberry Pi firmware rewrites serial0 to ttyS0,115200 with the new kernel
  * **Possible breakage:** Users who explicitly specify `-serial_console=ttyAMA0,115200` in their `gokr-packer` invocation will end up with non-working Bluetooth (as before) and no more serial console on pin 14/15 until they switch to `-serial_console=serial0,115200` in their gokr-packer invocation.
* [ ] only make gokr-packer set `enable_uart=0` in `config.txt` if `-serial_console` is `off` (new value that’s different from `disabled`, which we should rename to `for-application` or similar). We need `enable_uart=1` so that the device clock is set appropriately by the firmware to get reliable `ttyS0` output.
* [ ] switch `gokrazy/kernel` to drop the dtb patches and enable Bluetooth kernel options

Thus far, I have only tested this on the Raspberry Pi 4 B. Left to do before we can start: test on the Pi 3B, Pi 3B+ and Pi Zero 2 W as well, both with default `-serial_console` and with `-serial_console=disabled` and userspace serial usage.

stapelberg, 2022-03-19:
`gokrazy/kernel` is now updated to include Bluetooth support out of the box!

> @stapelberg Looks like we need more work for Zero 2 W.

You can try again now with the latest `gokrazy/kernel` (which includes commit https://github.com/gokrazy/kernel/commit/b3ced18560eb6afcc51f71fe4abc412b2fc03916)

stapelberg, 2022-03-19:
Created https://github.com/gokrazy/bluetooth, so feel free to send a PR for that, and then let me know when the website docs PR is ready for review please :)

joneskoo, 2022-03-19:
@stapelberg  I think there needs to be initial commit before I can make a pull request. Or I fail at Github.

stapelberg, 2022-03-19:
Commit added

joneskoo, 2022-03-19:
By the way I did not use -serial_console=off , did you expect that? Even with =disabled, Bluetooth works.

```
lrwxrwxrwx    1 0        0                0 Mar 19 21:15 /sys/class/bluetooth/hci0/device -> ../../../serial0-0
lrwxrwxrwx    1 0        0                0 Mar 19 21:14 /sys/bus/serial/devices/serial0 -> ../../../devices/platform/soc/3f201000.serial/serial0
lrwxrwxrwx    1 0        0                0 Mar 19 21:14 /sys/bus/serial/devices/serial0-0 -> ../../../devices/platform/soc/3f201000.serial/serial0/serial0-0
```

joneskoo, 2022-03-19:
> Created https://github.com/gokrazy/bluetooth, so feel free to send a PR for that, and then let me know when the website docs PR is ready for review please :)

The website change is ready for review #116 . Of course the example will really work only when gokrazy/bluetooth#1 is merged too.

stapelberg, 2022-03-20:
> By the way I did not use -serial_console=off , did you expect that? Even with =disabled, Bluetooth works. And I didn't even update gokr-packer. I literally only updated the kernel (and added firmware files).

Yeah, the serial console is entirely independent from Bluetooth now. I am using both serial and Bluetooth, for example.

Thanks for all the pull requests! The `gokrazy/bluetooth` repo is now live, and so is the documentation at https://gokrazy.org/userguide/bluetooth/

I’ll close this issue and if anything else remains, we can best track it in a separate issue.

closed 2022-03-20 by stapelberg