gokrazy/gokrazy #51: start a userguide

stapelberg, 2019-12-11:
To explain the following features:

* [ ] remote syslog

prologic, 2019-12-11:
I was going to suggest; why not just use the built-in Wiki? I'd even be happy to contribute; although I haven't used gokrazy all that much to be honest (yet)

stapelberg, 2019-12-11:
I’ll think about it. In general, I like having the documentation on the proper website, but perhaps the wiki is a more pragmatic solution.

prologic, 2019-12-11:
Yeah I tried that once myself with one of my projects with the whole Github Pages thing and found it too much work to maintain and get it looking nice. The Builtin Wiki is definately more convenient I think in this case plus you can contribute to it like any other repo.

stapelberg, 2020-05-29:
I’ve come to really like the editing experience with https://gohugo.io/, and have used it successfully for https://distr1.org/.

The distri website sources are located in https://github.com/distr1/distri/tree/master/website and instruct hugo to render to the https://github.com/distr1/distri/tree/master/docs subdirectory, which is then picked up by GitHub pages.

I think we can do the same thing here, i.e. convert https://github.com/gokrazy/gokrazy.github.io to a hugo instance which renders to the docs/ subdirectory and then serve that. Shouldn’t be too involved.

stapelberg, 2020-06-01:
Okay, the userguide is now started at https://gokrazy.org/userguide/, and contributions of the remaining topics (see first comment of this issue) are welcome! Please comment here when you start working on a topic so that others don’t accidentally duplicate your work.