gokrazy/gokrazy #61: ICMP not allowed?

gottwald, 2020-06-01:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [X] gokrazy/rpi3b+
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I'm trying to do a ICMP ping in a Go program. Code worked on a normal linux machine.

Via gokrazy I get the error `socket: permission denied` although the process is running as UID and GID 0.

I tried the following ping implementations which yield the same result:

github.com/sparrc/go-ping
github.com/stapelberg/zkj-nas-tools/ping (specifically the func `PingUnprivileged`)

## Expected behavior

A working ICMP ping just like on plain linux.

stapelberg, 2020-06-01:
Can reproduce:
```
[pid   692] socket(AF_INET, SOCK_DGRAM, IPPROTO_ICMP) = -1 EACCES (Permission denied)
[pid   692] write(2<pipe:[1167]>, "2020/06/01 14:09:52 rtt 0s, err socket: permission denied\n", 582020/06/01 14:09:52 rtt 0s, err socket: permission denied
) = 58
```

stapelberg, 2020-06-01:
From [man page icmp(7)](https://manpages.debian.org/testing/manpages/icmp.7.en.html):

```
ping_group_range (two integers; default: see below; since Linux
       2.6.39)
              Range of the group IDs (minimum and maximum group IDs,
              inclusive) that are allowed to create ICMP Echo sockets.  The
              default is "1 0", which means no group is allowed to create
              ICMP Echo sockets.
```

stapelberg, 2020-06-01:
After running `echo 0 0 > /proc/sys/net/ipv4/ping_group_range`, the program works as expected:

```go
	rtt, err := ping.PingUnprivileged(context.Background(), "10.0.0.76")
	log.Printf("rtt %v, err %v", rtt, err)
	// 2020/06/01 14:12:49 rtt 276.796µs, err <nil>
```

(ping is github.com/stapelberg/zkj-nas-tools/ping)

stapelberg, 2020-06-01:
The kernel source code recommends to set this at the distribution level: https://github.com/torvalds/linux/blob/3d77e6a8804abcc0504c904bd6e5cdf3a5cf8162/net/ipv4/af_inet.c#L1822

This would be at the gokrazy level for us.

I think for now, it’d be best if you checked the `/proc/sys/net/ipv4/ping_group_range` file programmatically and changed it as required, in your program.

If many programs expect to be able to send unprivileged pings, we can add it to gokrazy itself.

closed 2020-06-01 by stapelberg

gottwald, 2020-06-01:
Thx a lot! will do!

drummonds, 2024-09-22:
+1 Just to say I found this helpful as I was trying to connect to a host and adding a ping to see if it was visible and then needed this.