gokrazy/gokrazy #65: command line arguments for a program

adamcstephens, 2020-07-02:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [X] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

My application starts up using no command like arguments.

## Expected behavior

I'd like a way to configure the arguments for the application.

Maybe there's a way, and I've just not discovered it yet. I can convert to a config file, but that seems to require manual file management at this time? I'd prefer to be able to "just deploy" gokrazy and have my configuration work, without having to manually add files to the /perm filesystem.

stapelberg, 2020-07-02:
There’s currently no easy way to specify command line arguments.

---

The most self-contained way is to install a separate program which starts the program in question as a child process. This is explained in https://gokrazy.org/prototyping/, section “Go software not written for gokrazy: Grafana”.

---

Regarding managing `/perm`, in https://router7.org/, the updater backs up the contents of `/perm` before installing a new version, so I have daily backups.

stapelberg, 2020-07-02:
If we wanted to add more support for flags, I suppose the flagfile approach would be the most straight-forward, i.e. one flag per line (`\n`-separated), like so:
```
--foo=bar
--baz=qux
```

However, how do we get the flag files into the gokrazy image? The gokrazy packer currently only accepts Go packages, so we’d need to add a way to specify files.

---

However, when adding flags, we should probably also be able to specify environment variables, and when you have that, maybe they should be templated in some way. At the very end of the extreme, we’d reimplement parts of systemd.

It’s not obvious to me where to draw the line here, so we currently don’t have any facilities.

adamcstephens, 2020-07-02:
I may take a look at some options. If I can keep the implementation minimal would packer be an acceptable place for this functionality, or would you prefer I wrap packer?

stapelberg, 2020-07-02:
> I may take a look at some options. If I can keep the implementation minimal would packer be an acceptable place for this functionality, or would you prefer I wrap packer?

Modifying the packer as a first step sounds reasonable to me, but I haven’t made up my mind about whether/how much of this functionality to include. Discussing things over actual code is typically better, though :)

adamcstephens, 2020-07-06:
I looked into this over the weekend, but didn't produce anything shippable.

By refactoring [this slice creation](https://github.com/gokrazy/tools/blob/ea3341009e65c07c91e6157fa3be8955e6aa8e73/cmd/gokr-packer/buildinit.go#L45-L51), `Cmd.Env` and `Cmd.Args` could be populated by the packer. [Env isn't passed directly currently](https://github.com/gokrazy/gokrazy/blob/6739b590288a96258c8d8fe824b87071c34ad49c/supervise.go#L283-L291), but could be. 

The question then becomes, what is the user interface? Are you wanting the packer tool to remain a command line tool only, or would you consider a config file for input?

stapelberg, 2020-08-08:
Sorry for the late reply. I had thought about this off and on over the last few days.

---

I think I like the rule of thumb that any mechanism we support should be trivially supported by the Go standard library, and a generally well-known concepts. So, statically configured environment variables and command line parameters are fine.

This seems like a good place for me to draw the line.

---

> I looked into this over the weekend, but didn't produce anything shippable.

Thanks for giving it a shot!

> By refactoring [this slice creation](https://github.com/gokrazy/tools/blob/ea3341009e65c07c91e6157fa3be8955e6aa8e73/cmd/gokr-packer/buildinit.go#L45-L51), `Cmd.Env` and `Cmd.Args` could be populated [...]

Yep, sounds reasonable.

> The question then becomes, what is the user interface? Are you wanting the packer tool to remain a command line tool only, or would you consider a config file for input?

Yeah, that is a good question.

A concept that we alreay promote is to use one directory per gokrazy instance.

We also have support for reading a per-device HTTP password based on directories.

I think a consistent option and good choice would be to use plain text files (flagfiles and environment files, i.e. one flag per line, or one key=value pair per line) within a directory per import path, looked for in the current directory.

E.g., to make the prometheus SNMP exporter work on gokrazy (or router7 specifically):

```
% mkdir -p github.com/prometheus/snmp_exporter
% echo "--config=/perm/snmp.yml" > github.com/prometheus/snmp_exporter/flagfile.txt
```

We should take care to make it very clear to users which config files are going into the image.

So, as soon as the packer uses any extra inputs, it should log their location and last modification timestamp.

andig, 2021-01-03:
> However, how do we get the flag files into the gokrazy image? The gokrazy packer currently only accepts Go packages, so we’d need to add a way to specify files.

That would actually be my feature request on its own, too. Working on OSX its not easy to to create the ext4 filesystem. It would be nice to bundle authorised keys for breakglass in `/user` instead of ext4.

stapelberg, 2021-01-04:
> Working on OSX its not easy to to create the ext4 filesystem

You can use any file system that the Linux kernel supports and is convenient with your host operating system. Maybe FAT is a better choice for you?

stapelberg, 2021-01-04:
I just pushed commit https://github.com/gokrazy/tools/commit/fdd90fc6817876e08b352fae84f2a2794524ccc0 (see the commit message for details).

Please give this a shot and report back if it works.

Still left to do before we can close this issue:
* [ ] report modification time of the flags files so that users can catch errors quickly.
* [ ] add the same mechanism for environment variables, too

andig, 2021-01-11:
For build args see discussion in https://github.com/gokrazy/tools/pull/21

stapelberg, 2021-01-18:
I think this is all done!

closed 2021-01-18 by stapelberg