gokrazy/gokrazy #66: Consider exposing Linux kernel version in web UI

mdlayher, 2020-07-13:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

Web UI displays useful information about the currently running system, but no Linux kernel version. There is the "version" string but this only seems to indicate when the system was last updated.

## Expected behavior

Ability to view Linux kernel version without resorting to breakglass:
```
/tmp/breakglass048386764 # uname -a
Linux monitnerr-1 5.7.8 #1 SMP PREEMPT Wed Mar 1 20:57:29 UTC 2017 aarch64 GNU/Linux
```

stapelberg, 2020-07-13:
Definitely a good idea. Pull requests welcome :)

mdlayher, 2020-07-13:
Will do later today!

PR #67 (2020-07-14): Add kernel information to web UI header

I figured it might be handy to also include the architecture. This provides the following output on my Pi 4:

```
host     monitnerr-1
kernel   Linux 5.7.8 (aarch64)
model    Raspberry Pi 4 Model B Rev 1.1
EEPROM   5a07872332
(SHA256) 3a46dda0da
```

Fixes #66.
closed 2020-07-14 by stapelberg