gokrazy/gokrazy #68: Firmware/BIOS information

CodeZombieCH, 2020-09-03:
I'm working on a PR to show firmware/BIOS information (in particular the version number) of my PC Engines router7 in the header of the web UI. With the recent addition of kernel information, I feel like adding an addition row with firmware information grows the header beyond a reasonable height. I came up with three options to display the firmware information in the header. Please let me know which one you would accept as a PR :-) 

## Option 1:
[image]
Compact version, where firmware information is added to the model information

## Option 2:
[image]
Current single column approach with detailed firmware

## Option 3:
[image]
Multiple columns with detailed firmware information with very small height

stapelberg, 2020-09-03:
Thanks for the detailed suggestion!

I prefer option 3: it’s the most compact, and given the gokrazy web interface’s target audience of developers, I think most viewers will be on a screen that’s wider than high.

CodeZombieCH, 2020-09-03:
Thanks for the feedback. I forgot to mention that option 3 is my preferred option as well. I will update my bootstrap skills and create a PR within the next days.