gokrazy/gokrazy #70: Power Consumption stats

hazcod, 2020-09-29:
Hi, are there any power consumption numbers available when using gokrazy?
I understand most of the factors are within the kernel, but still interesting (and possible nice PR) for gokrazy to share.

stapelberg, 2020-09-29:
Yeah, would be nice, but I don’t have any numbers handy.

Help welcome

mdlayher, 2020-09-29:
This data is exposed in the Prometheus node exporter ever since I made a couple of sensors tweaks in the kernel repo a while back. It should be easy enough to read the relevant hwmon files from sysfs and expose that. My APU (not using gokrazy) also seems to expose power draw so it should be portable across all gokrazy platforms. Temperature is something else that's useful to see.

hazcod, 2020-09-29:
Ahh, thanks for confirming, might be nice to add an example to the README?
(I am asking since I don't have any Pis laying around)

stapelberg, 2021-11-28:
Took some power usage measurements, including the power supply (official Raspberry Pi 4 power supply for the Pi 4, and a random USB power supply for the others).

| platform | power usage (idle)
|------------|------------------|
| rpi3b | 1.5W |
| rpi3b+ |  2.5W | 
| rpi4 | 2.8W | 
| rpiz2w | 0.8W |
| apu2c4 | 4.0W |

I’ll add these to the website soon

closed 2022-03-23 by stapelberg