gokrazy/gokrazy #73: [CI] Enable GitHub Actions crons periodically

stapelberg, 2020-12-30:
Turns out GitHub Actions disables cron jobs after 60 days of inactivity in a repository, which has happened at least once for the Raspberry Pi 4 EEPROM repository.

We can manually re-enable the jobs, but should automate this.

See also https://discourse.julialang.org/t/compathelper-workflows-disabled-after-60-days-of-inactivity/48870/3

stapelberg, 2021-01-04:
Deployed a little program which re-enables all gokrazy CI workflows daily.

closed 2021-01-04 by stapelberg

prologic, 2021-01-04:
> Deployed a little program which re-enables all gokrazy CI workflows daily.

Source? 😁

stapelberg, 2021-01-04:
It’s just 2 API calls:

```
func enableAllWorkflows(client *github.Client, owner, repo string) error {
	ctx := context.Background()
	workflows, _, err := client.Actions.ListWorkflows(ctx, owner, repo, &github.ListOptions{})
	if err != nil {
		return err
	}
	for _, wf := range workflows.Workflows {
		log.Printf("owner=%s repo=%s workflow=%d state=%q", owner, repo, wf.GetID(), wf.GetState())
		if wf.GetState() == "active" {
			continue
		}
		if _, err := client.Actions.EnableWorkflowByID(ctx, owner, repo, wf.GetID()); err != nil {
			return err
		}
	}
	return nil
}
```

It’s almost more work to get the plumbing in place (cronjob, GitHub access token, which repositories to work on), so I figured I wouldn’t publish a proper repository for this.