gokrazy/gokrazy #74: Feature: Allow custom port for gokrazy ui

andig, 2021-01-03:
## Platform

n/a

## Observed behavior

gokrazy uses http:80 for it's own purposes. I'd like to use that for the bundled application.

## Expected behavior

Specify custom port for created package, to be reused by updater.

Would be happy to work on a PR.

stapelberg, 2021-01-04:
I think this is a reasonable feature request, just not entirely sure yet about the configuration surface.

Maybe another file (`port`) in `~/.config/gokrazy`?

andig, 2021-01-04:
I've noticed `~/.config` in the quickstart but at that time wasn't sure if that's on the host (it is). I don't have `~/.config` on my mac after building an image- when is it generated for the default password?

For the port I'd say that it is image-specific if it needs be moved or not. Would it hurt to allow a `gokrazy` folder in side cwd that- if exists- takes precedence over `~/.config/gokrazy`?

Simple file with just `80,443` as content? Injected into build using build vars?

Merovius, 2021-01-04:
I think you really want [os.UserConfigDir](https://golang.org/pkg/os/#UserConfigDir), whatever that may end up being.

stapelberg, 2021-01-04:
> For the port I'd say that it is image-specific if it needs be moved or not

gokrazy’s configuration is centered around the concept of a hostname.

> I think you really want [os.UserConfigDir](https://golang.org/pkg/os/#UserConfigDir), whatever that may end up being.

We do use os.UserConfigDir, see https://github.com/gokrazy/internal/blob/master/config/config.go

> Simple file with just `80,443` as content?

I would suggest one port number per file. Perhaps we could name them `port-http` and `port-https`.

> Injected into build using build vars?

I don’t think any build vars are necessary? What did you have in mind?

fawick, 2021-01-04:
>  I don’t think any build vars are necessary? What did you have in mind?

The possibility to state additional Go build tags and or linker flags for setting symbol values (version info) would be nice.

andig, 2021-01-04:
> gokrazy’s configuration is centered around the concept of a hostname.

Got it.

> The possibility to state additional Go build tags and or linker flags for setting symbol values (version info) would be nice.

Instead of modifying the host program it would be nice to use the right build tags during packaging (`build-tags` file?).

andig, 2021-01-06:
There are two todos here:

- [ ] implement runtime: https://github.com/gokrazy/gokrazy/pull/77
- [ ] implement host tooling: help wanted

I'd suggest to move additional requirements like https://github.com/gokrazy/gokrazy/issues/74#issuecomment-753894459 to a new issue.

stapelberg, 2021-01-06:
Now done!

closed 2021-01-06 by stapelberg