gokrazy/gokrazy #75: Reboot loop on failed root fs update

andig, 2021-01-05:
## Platform

I’m using:

* [x] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

As part of https://github.com/gokrazy/gokrazy/issues/25 I've played with writing updates from custom update client. Using the sequence of:

	if err := target.StreamTo("root", rootReader); err != nil {
		log.Fatalf("updating root file system: %v", err)
	}

	if err := target.Switch(); err != nil {
		log.Fatalf("switching to non-active partition: %v", err)
	}

	if err := target.Reboot(); err != nil {
		log.Fatalf("reboot: %v", err)
	}

with an intentionally broken root fs (i.e. just the first 5MB), the device entered an endless reboot loop.

## Expected behaviour

Invalid root caught and device switched back to other A/B partition to make device accessible again.

stapelberg, 2021-01-05:
The expected behavior is what https://github.com/gokrazy/gokrazy/issues/25#issuecomment-617401988 describes.

andig, 2021-01-06:
> The expected behavior is what #25 (comment) describes.

I read #25 as relating to the kernel. I understood that the root fs already uses A/B partitioning and hence supports recovery from failed update?

stapelberg, 2021-01-06:
The root filesystem uses A/B partitioning, but has no way to test-boot into a new partition.

Currently, the partitioning only helps with transferring updates: overwriting the non-active partition doesn’t harm the active partition, so interrupted updates are not a problem.

stapelberg, 2021-01-30:
Current plan:

* [ ] enable kexec for arm64 for the Raspberry Pi (welcome side-effect: faster updates) to make possible a temporary boot into a new kernel and/or root file system
* [ ] add a testboot handler (kexec into inactive root partition) to be called instead of switch
* [ ] (maybe later) make gokr-packer generate a boot partition with the old (/vmlinuz) and new kernel (/vmlinuz.canary), testboot handler should pick the canary kernel when present
* [ ] add a component to the system which completes the update by doing/calling switch locally.

amurray-thegoodpenguin, 2021-03-11:
Regarding this: "kexec only works with nr_cpus=1: raspberrypi/linux#843 (see also https://www.thegoodpenguin.co.uk/blog/booting-linux-from-linux-with-kexec/)"

It should be possible to kexec on a RPI with multiple CPUs  - however you need to have some form of trusted firmware on the device - this can be achieved via ATF (https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/plat/rpi3.rst) - and other mechanisms. However you need really need to have control of the full software on the device.

stapelberg, 2021-03-12:
Thanks for the note. It looks like ATF (ARM trusted firmware) adds PSCI (Power State Coordination Interface) support which allows the kernel to park the extra CPUs. however, the specific repository also states:

> IMPORTANT NOTE: […] This port shouldn't be considered more than a prototype to play with and implement elements like PSCI to support the Linux kernel.

So I don’t think this is quite ready yet for prime time :)

https://www.thegoodpenguin.co.uk/blog/booting-linux-from-linux-with-kexec/ has more details and links to another PSCI workaround.

amurray-thegoodpenguin, 2021-03-12:
Yes I wrote that blog post :)

ATF is very very widely used and battle tested - I suspect the RPI port is marked as 'prototype-use-only' because the RPI lacks a hardware capability to support secure DRAM. The idea is that you have a trusted execution that uses DRAM that the non-secure world (e.g. Linux) cannot access.  However unless you're doing secure boot or using a TA or similar - then that's not a problem for you.

stapelberg, 2021-03-12:
Thanks for clarifying. I might play around with it when I get a chance :)