gokrazy/gokrazy #76: Broaden public API for runtime use

andig, 2021-01-05:
Refs https://github.com/gokrazy/gokrazy/issues/25

As part of trying to update a gokrazy device from running client I've found that I needed to use `updater.NewTarget` and hence `httpclient`. Both are currently part of the internal api and need be exposed (https://github.com/gokrazy/gokrazy/issues/25#issuecomment-754558193). I feel they would be a good fit for `gokrazy/tools` where the updater lives.

I have not looked into `internal/httpclient` yet. I assume it is tied to the host and `~/.config`? What we need here is something that reads device configuration from the runtime filesystem. That should be part of the public `gokrazy/gokrazy` api. 

If there are additional use cases for updating device from outside and without `gokrazy/tools` that would need exposing parts of `internal/httpclient` but should not be part of this issue.

andig, 2021-01-06:
To summarise I would see the following points regarding public api:
- [ ] expose `updater.NewTarget` for guest and host use (migrate to `gokrazy/tools`?)
- [ ] expose `httpclient` for guest (taking https://github.com/gokrazy/gokrazy/issues/74 runtime aspect into account once implemented)
- [ ] expose `httpclient` for host (taking https://github.com/gokrazy/gokrazy/issues/74 host aspect into account once implemented)

`httpclient` could be skipped if a higher-order API based on hostname only were available but it seems a rather limited approach.

stapelberg, 2021-01-06:
> I feel they would be a good fit for `gokrazy/tools` where the updater lives.

I think it might be better to put the updater into its own `gokrazy/updater` repository to more clearly separate this functionality from other functionality in the tools repository, which is more geared towards interactive usage. This allows for a more minimal dependency footprint, for example.

> What we need here is something that reads device configuration from the runtime filesystem.

We could alternatively just make the update API listen on a localhost-only HTTP port and skip the entire TLS dance. It will only slow down the update and there isn’t anything to protect when sending data to localhost :).

A password would still be necessary, though, otherwise a compromised application could take over whole devices trivially.

andig, 2021-01-06:
> A password would still be necessary, though, otherwise a compromised application could take over whole devices trivially.

Imho it can always do so as it would find the password in the filesystem?

stapelberg, 2021-01-06:
> Imho it can always do so as it would find the password in the filesystem?

We should make sure the password file is readable only by root. That way, security-conscious users can employ privilege separation to ensure application-level compromises don’t result in access to the password.

andig, 2021-01-07:
As discussed we'll not add runtime or host api for port or TLS configuration. Seems we're finished with this one ;)

closed 2021-01-07 by andig