gokrazy/gokrazy #84: Randomd suggestion

trafalmuffti, 2021-03-15:
When randomd can't read from random.seed, it should instead loop 256 times in writing to the entropy pool. Linux counts all interrupts as increasing entropy in my experience.

I'm also not sure about this, but the entropy struct according to the Linux specification uses c ints, which I assume is 32 bits in Linux.

You might be passing a struct with 0 bits of entropy as read by the kernel.

I personally use a bash script that runs 

`nice cat /bin/* > /dev/random/`

stapelberg, 2021-03-15:
> When randomd can't read from random.seed, it should instead loop 256 times in writing to the entropy pool. Linux counts all interrupts as increasing entropy in my experience.

Is there any Linux kernel documentation that explains how the entropy pool should be seeded? Or whether it should be seeded at all?

Did you run into an actual issue due to low entropy?

trafalmuffti, 2021-03-15:
I don't understand what you're trying to do with randomd anyway. 

To be most accurate, you would need to also read from entropy_avail and include it in random.perm in a backwards compatible way since it is possible a seed from a previous boot was of insufficient entropy.

I haven't run into an issue with low entropy with your library.

FiloSottile, 2024-01-27:
With a recent enough kernel, the random subsystem can be trusted to do the right thing thanks to the changes by Jason Donenfeld. The only thing userspace needs to do is save a seed if able. In fact, doing things like causing interrupts or writing the time is counterproductive because it will confuse the kernel about how much entropy it really has available at boot.

There are a couple issues with the current cmd/randomd:

- the `time.Now().String()` fallback should be removed
- overwriting with zeroes is counterproductive, an old seed is better than all zeros
- saving the seed should be done immediately after loading it, but after causing a pool remix

In practice, the correct behavior is to just do what Jason's seedrng does. https://git.zx2c4.com/seedrng/about/

Ignoring the blake2b implementation that you can take from x/crypto, it's 200 lines of C that should be easy to port to Go.

stapelberg, 2024-01-27:
Thanks for the clarification! I’m happy to review pull requests for these enhancements if someone wants to help :)