gokrazy/gokrazy #85: Add flag to gokr-packer to initialize perm partition

darkskiez, 2021-04-12:
When I want to create the perm partition in a filesystem image I have to mount the loopback image and format it, this is a bit time consuming and requires me to sudo etc.

As the offset and size is already known to gokr-packer it would be nice to suggest the mkfs.ext4 command to format it, or even run it under a flag.

mkfs.ext4 has an -E offset=<offset> <size> command specifically for preparing disk images.

stapelberg, 2021-04-14:
When writing directly to an SD card, the packer already prints an mkfs command that uses the UUID.

I agree that when writing file system images, it would be convenient if the packer printed an mkfs command that uses the `-E` flag (thanks for the hint!).

stapelberg, 2021-04-21:
https://github.com/gokrazy/tools/commit/e39dd5e7c9d05e3b9114d2db9a4b5b79c8459769 is related (but I screwed up the copy&paste)

stapelberg, 2022-03-01:
Just saw that it wasn’t mentioned in this issue yet: https://github.com/gokrazy/mkfs can automatically create the perm partition. We should mention it in the userguide.