gokrazy/gokrazy #86: web Interface link not working.

jasserouni, 2021-05-21:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [+] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior
I succeded to build gokrazy app in sdcard, but the generated web interface link is not woking. When I open it it shows "invalid username/password"
## Expected behavior
running web Interface link to control raspberry pi

stapelberg, 2021-05-21:
Can you check that the password you are using in your browser matches what is stored on the SD card in file etc/gokr-pw.txt? Username should be gokrazy.

jasserouni, 2021-05-21:
Hi it's the same I already checked and the auto-generated link is written correctly http://gokrazy:<password>@gokrazy/

stapelberg, 2021-05-21:
What’s the result of `curl -v link` on a terminal?

jasserouni, 2021-05-21:
*   Trying "ip address"...
* TCP_NODELAY set
* Connected to gokrazy (ip address) port 80 (#0)
* Server auth using Basic with user 'gokrazy'
> GET / HTTP/1.1
> Host: gokrazy
> Authorization: Basic Z29rcmF6eTpsOVJKRDMyemZZNEhPeW1LcDQzOQ==
> User-Agent: curl/7.58.0
> Accept: */*
> 
< HTTP/1.1 401 Unauthorized
< Content-Type: text/plain; charset=utf-8
< X-Content-Type-Options: nosniff
< Date: Fri, 21 May 2021 11:43:16 GMT
< Content-Length: 26
< 
invalid username/password
* Connection #0 to host gokrazy left intact

stapelberg, 2021-05-21:
Can you take a screenshot of the message that gokr-packer prints please? I wonder if there are any special characters. No need to redact the password, you can auto-generate a new one by deleting ~/.config/gokrazy/http-password.txt

jasserouni, 2021-05-21:
here is the screenshot.
PS: I ve got the same problem when I tried to build only hello program.
[image]

stapelberg, 2021-05-21:
Looks like the password matches what was in your HTTP request.

Are you 100% sure that the device you’re talking to over the network is the correct one? I.e. does “ping gokrazy” stop when you remove the cable from the Pi, and resume when you plug it back in? Do you have serial or HDMI connected to the Pi? What’s visible there?

jasserouni, 2021-05-21:
Actually I m not sure, I tried ping gokrazy and it still working even after unplugging the power cable and doesn't stop. 
I connected the rpi via hdmi now and it shows the bootloader has some failures in regulatory.db lines but still working until now.

stapelberg, 2021-05-23:
> Actually I m not sure, I tried ping gokrazy and it still working even after unplugging the power cable and doesn't stop.

That suggests the hostname “gokrazy” is taken by something else in your local network. Can you check your router’s status page to see if it contains any details regarding which IP address is used by which MAC address?

> I connected the rpi via hdmi now and it shows the bootloader has some failures in regulatory.db lines but still working until now.

Can you specify the `-serial_console=disabled` flag when running `gokr-packer`, and take a picture of the output you see on HDMI afterwards once the Pi has booted up?

jasserouni, 2021-05-25:
> That suggests the hostname “gokrazy” is taken by something else in your local network. Can you check your router’s status page to see if it contains any details regarding which IP address is used by which MAC address?
> I have no access to my router interface, but I tried with three different networks and the problem still exists.
> 
> Can you specify the `-serial_console=disabled` flag when running `gokr-packer`, and take a picture of the output you see on HDMI afterwards once the Pi has booted up?
> This one is done too but nothing really changed...

stapelberg, 2021-05-25:
> I have no access to my router interface, but I tried with three different networks and the problem still exists.

Can you temporarily disconnect all devices from your router except for your computer and the Pi?

> This one is done too but nothing really changed...

Can you take and share the picture? It might give us a hint as to what’s going on.

jasserouni, 2021-05-25:
here is the output after rebooting, I disconnected all rest devices. 
[image]
[image]
[image]

stapelberg, 2021-05-25:
Thanks. I see 2 messages from listeners.go:177, indicating the device is listening on 127.0.0.1:80 and [::1]:80 only.

I would have expected a third message from listeners.go:177 with an IP address from your local network.

This indicates that the Pi can’t obtain a DHCP lease from your router.

Do you have a different router available to which you could connect a computer and your Pi? I wonder if this problem is specific to your router or not.

jasserouni, 2021-05-25:
Humm now looks logic to me, sadly I don't have any other network option to test on it. But I will try to find another way to test it on new network.
I will keep the issue updated. Thanks.

stapelberg, 2021-05-27:
Just to be sure: you are on the most recent kernel, firmware and eeprom version, yes?

In your go.mod, I would expect the following lines:
```
	github.com/gokrazy/firmware v0.0.0-20210526042401-74abf0e9bc49 // indirect
	github.com/gokrazy/kernel v0.0.0-20210527044747-d34bc871b027 // indirect
	github.com/gokrazy/rpi-eeprom v0.0.0-20210430033855-45c7d5dc0a11 // indirect
```

If no, run:
```
go get github.com/gokrazy/firmware@latest
go get github.com/gokrazy/rpi-eeprom@latest
go get github.com/gokrazy/kernel@latest
```

The expected boot messages look like this:

[image]

Note the “Link is Up” message and the link-local IPv6 address (fe80:…) that the device is listening on.

If you don’t see these, it would suggest your Raspberry Pi can’t establish a physical ethernet link? Is your cable or switch port faulty perhaps?

jasserouni, 2021-05-27:
Thank you for your reply..
I updated the kernel and rebooted everything and now I have the same output as your picture but web interface still not working :/

jasserouni, 2021-06-01:
Fixed by setting specific hostname while creating the image using -hostname flag

closed 2021-06-01 by jasserouni