gokrazy/gokrazy #89: Question: Can we use gokrazy in commercial Iot implementations?

plkumar, 2021-07-10:
Can we use gokrazy in commercial Iot implementations?

stapelberg, 2021-07-10:
It’s an open source project, so nobody is stopping you from using it in any way you see fit :)

That said, there is no guaranteed support, no guarantee that any features you might want to add will be merged, and no warranty whatsoever for any problems that might arise.

Can you share more details about what your project is?

closed 2022-03-01 by stapelberg

sandrolino, 2024-12-17:
Are you aware of any companies using gokrazy in production?
I'm working for a startup doing flexibility aggregation for EVs and large roof-top solar systems (it's different from EVCC) and would love to use it. Having experience from other starutps(companies would make the evaluation easier (besides own testing of course).

stapelberg, 2024-12-18:
I am not aware of any company using gokrazy in production.