gokrazy/gokrazy #111: Wifi setup fails due to `"no interfaces found"`

christian-heusel, 2022-03-03:
## Platform

I’m using:

* [x] gokrazy/rpi3b (Raspberry Pi 3 Model B Rev 1.2)
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

On the software side everything should be up to date, i can also provide version numbers, just let me know which ones matter.

## Observed behavior

When setting up the newly implemented Wifi setup the program gives the following output:
```
1970/01/01 01:00:14 gokrazy: exit status 1
1970/01/01 01:00:15 gokrazy: attempt 2, starting ["/user/wifi"]
1970/01/01 01:00:15 no interfaces found
1970/01/01 01:00:15 gokrazy: exit status 1
1970/01/01 01:00:16 gokrazy: attempt 3, starting ["/user/wifi"]
1970/01/01 01:00:16 no interfaces found
1970/01/01 01:00:16 gokrazy: exit status 1
2022/03/03 17:56:30 gokrazy: attempt 4, starting ["/user/wifi"]
2022/03/03 17:56:30 no interfaces found
2022/03/03 17:56:30 gokrazy: exit status 1
2022/03/03 17:56:31 gokrazy: attempt 5, starting ["/user/wifi"]
2022/03/03 17:56:31 no interfaces found
2022/03/03 17:56:31 gokrazy: exit status 1
2022/03/03 17:56:32 gokrazy: attempt 6, starting ["/user/wifi"]
2022/03/03 17:56:32 no interfaces found
2022/03/03 17:56:32 gokrazy: exit status 1
2022/03/03 17:56:33 gokrazy: attempt 7, starting ["/user/wifi"]
2022/03/03 17:56:33 no interfaces found
2022/03/03 17:56:33 gokrazy: exit status 1
[...]
```

This issue was already described during the dev phase of the feature in [Issue #13](https://github.com/gokrazy/gokrazy/issues/13#issuecomment-1047345954), but should have been fixed by a firmware update.

## Expected behavior

The Wifi interface is found and correctly initialized with the given credentials/SSID.

stapelberg, 2022-03-03:
Can you provide the full `gokr-packer` output please? Are you sure you’re using the latest version?

Can you provide the `dmesg` output via https://github.com/gokrazy/breakglass please?

christian-heusel, 2022-03-03:
<details>
  <summary>Output of <code>gokr-packer</code> (Click to expand)</summary>

```
$ make update
go get -u github.com/gokrazy/tools/cmd/gokr-packer github.com/gokrazy/gokrazy github.com/gokrazy/kernel github.com/gokrazy/firmware github.com/gokrazy/rpi-eeprom
go get: installing executables with 'go get' in module mode is deprecated.
	To adjust and download dependencies of the current module, use 'go get -d'.
	To install using requirements of the current module, use 'go install'.
	To install ignoring the current module, use 'go install' with a version,
	like 'go install example.com/cmd@latest'.
	For more information, see https://golang.org/doc/go-get-install-deprecation
	or run 'go help get' or 'go help install'.
go install github.com/gokrazy/tools/cmd/gokr-packer@latest
gokr-packer \
	-update=http://gokrazy:***@johnny.fritz.box/ \
	-hostname=johnny \
	-serial_console=ttyS0,115200n8 \
	github.com/gokrazy/hello github.com/gokrazy/serial-busybox github.com/gokrazy/mkfs github.com/gokrazy/wifi github.com/gokrazy/breakglass github.com/prometheus/node_exporter
gokrazy packer v0.0.0-20220301215540-c1b172b2f083 on GOARCH=amd64 GOOS=linux

Updating gokrazy installation on http://johnny

Build target: CGO_ENABLED=0 GOARCH=arm64 GOOS=linux
Build timestamp: 2022-03-03T19:40:55+01:00
Loading system CA certificates from /etc/ssl/certs/ca-certificates.crt
Building 6 Go packages:

  github.com/gokrazy/hello

  github.com/gokrazy/serial-busybox

  github.com/gokrazy/mkfs

  github.com/gokrazy/wifi
    will include extra files in the root file system
      from extrafiles/github.com/gokrazy/wifi
      last modified: 2022-03-03T03:33:12+01:00 (16h7m43s ago)

  github.com/gokrazy/breakglass
    will be started with command-line flags
      from flags/github.com/gokrazy/breakglass/flags.txt
      last modified: 2022-03-03T04:38:52+01:00 (15h2m3s ago)
    will include extra files in the root file system
      from extrafiles/github.com/gokrazy/breakglass
      last modified: 2022-03-03T18:54:10+01:00 (46m45s ago)

  github.com/prometheus/node_exporter

[done] in 1.03s                        
2022/03/03 19:40:56 Adding loadable kernel modules from modulesDir=/home/chris/go/pkg/mod/github.com/gokrazy/kernel@v0.0.0-20220303073407-62a05d3ea17c/lib/modules
[done] in 3.09s               

Feature summary:
  use GPT: true
  use PARTUUID: true
  use GPT PARTUUID: true

Creating boot file system
[creating boot file system]
Kernel directory: /home/chris/go/pkg/mod/github.com/gokrazy/kernel@v0.0.0-20220303073407-62a05d3ea17c
EEPROM update summary:
  pieeprom.upd (sig 03b81e39c6)
  recovery.bin
  vl805.bin (sig 6246230ecd)
MBR summary:
  LBAs: vmlinuz=51666 cmdline.txt=136098
  PARTUUID: 3ef2d59d
[done] in 0.08s, 63 MiB                           

[... 335 lines cut ...]
[    1.144739] mmc1: new ultra high speed DDR50 SDHC card at address aaaa
[    1.145455] mmcblk1: mmc1:aaaa SC32G 29.7 GiB 
[    1.151559]  mmcblk1: p1 p2 p3 p4
[    1.157648] VFS: Mounted root (squashfs filesystem) readonly on device 179:3.
[    1.158184] devtmpfs: mounted
[    1.163177] Freeing unused kernel memory: 7744K
[    1.186834] Run /gokrazy/init as init process
[    1.186848]   with arguments:
[    1.186854]     /gokrazy/init
[    1.186859]   with environment:
[    1.186862]     HOME=/
[    1.186866]     TERM=linux
[    1.230895] hub 1-1:1.0: USB hub found
[    1.231182] hub 1-1:1.0: 4 ports detected
[    1.476226] EXT4-fs (mmcblk1p4): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[    3.230900] random: crng init done
[    5.271023] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    5.331165] bcmgenet fd580000.ethernet eth0: Link is Down
[    7.358873] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[    7.358938] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   10.720582] vc4-drm gpu: [drm] Cannot find any crtc or sizes
```
</details>

I now took the SDCard to a raspberry 4 and the problem persists 🤔

stapelberg, 2022-03-03:
Ah, I see the problem: the automatic kernel update from our CI did not include the kernel modules.

stapelberg, 2022-03-03:
Update the kernel package and try again now :)

closed 2022-03-03 by stapelberg

stapelberg, 2022-03-03:
Ah, I can reproduce the issue when I don’t use the `replace` directive in my `go.mod` file!

Turns out it was due to a symlink. This should be fixed with commit https://github.com/gokrazy/wifi/commit/8ebd82a86b74a6ed69a15277d3a4888249105b69, so please update your `github.com/gokrazy/wifi` package and try again :)

This combination works for me:
[image]

christian-heusel, 2022-03-03:
Now it works for me aswell! 🥳