gokrazy/gokrazy #117: Gokrazy supervisor keeps restarting process if manually started

joneskoo, 2022-03-09:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [x] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

As in tailscale/tailscale#4117. If a service implements gokrazy exit code 125 on GOKRAZY_FIRST_START=1, and the user starts the process anyway from console, it keeps being restarted even if it keeps failing in a loop.

## Expected behavior

Do not supervise and keep restarting process if it has reported code 125 on first start. That is, remember that process doesn't want to be supervised if it returns 125 and only start once.

stapelberg, 2022-03-09:
Alternatively, we could also reset GOKRAZY_FIRST_START to 1 when the user restarts a service, right?

joneskoo, 2022-03-09:
Good idea, no reason to not set the first start on every new start, including manual. Simple and compatible extension of the interface.

stapelberg, 2022-03-09:
Cool. Do you want to send a pull request or would you prefer if I did the change?

joneskoo, 2022-03-09:
I don't mind at all if you implement it. I've been working/procrastinating on appliance refresh during my vacation but the rest of the vacation is more busy.

stapelberg, 2022-03-13:
Turns out we can’t implement it like that after all, because that breaks `breakglass`.

stapelberg, 2022-03-20:
I suppose the best way would be to remove the “restart” button in favor of a “run once” and “supervise (run in a loop)” button.

`gok run` also would ideally run just once by default (currently runs in a loop).

closed 2022-03-21 by stapelberg