gokrazy/gokrazy #119: Running gokrazy on unsupported devices / alternative kernels (32-bits arm - Raspberry Pi B for instance)

oliverpool, 2022-03-21:
_Following #46, but as a new issue, since this is not related to camera support at all._ 

**This issue is not to be addressed by the maintainers of this repo**, since they [clearly stated](https://www.reddit.com/r/golang/comments/5xgf8u/gokrazy_a_purego_userland_for_your_raspberry_pi_3/dei7b2m/), that they are not interested in supporting alternative kernels ("supporting" in the sense of having to do all the work). 

However @stapelberg indicated in https://github.com/gokrazy/gokrazy/issues/46#issuecomment-771793364
> Okay, if you want to maintain the raspberry pi foundation kernel, that’d be a good option for people I think.
>
> We can list your repository on https://gokrazy.org/platforms/ under a separate section.

So the goal of this issue is to allow interested people to coordinate their efforts, to support alternative kernels inside gokrazy and document their findings, for them to be eventually officially documented (even without official support).

---

@cweagans you mentioned that you were thinking about "taking this on". Were you able to find some time to start something with your raspberry pi zero?

I would also really like to be able to use the gokrazy workflow with an old Raspberry Pi B (single go executable on top of the kernel), but will have to learn a lot about the linux kernel in the process.

cweagans, 2022-03-21:
I started poking at it. I don't have anything of substance working and much of what I did was just trying to understand the internals of the gokrazy ecosystem.

oliverpool, 2022-03-22:
I created a custom package for the `kernel` and the `firmware` and naively put the official raspberry pi files (dtb files and `kernel.img` renamed as `vmlinuz`).

It managed to boot, but didn't reach the init first.

I then realized that my SD card was too small (512 MB) - I will take a look if it is possible for gokrazy to fail with an error message (it currently succeeds without any suspicious log).

Taking a large enough SD card, I ended with the following error message:
```
No filesystem could mount root
```
Effectively the list of available filesystem driver did not include `SquashFS`.

I looked around, but wasn't able to find a pre-compiled pi kernel with SquashFS included so I went the cross-compile way...
https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compiling-the-kernel
...which isn't smooth on archlinux (the `crossbuild-essential-armhf` package is not available there).

So I set-up a docker container with the latest ubuntu (the latest official raspberry kernel requires glibc 2.34, which is only available starting with ubuntu impish).

```
FROM ubuntu:impish-20220316

WORKDIR /root/raspberrypi

RUN apt-get update && \
    apt-get install -y git bc bison flex libssl-dev make libc6-dev libncurses5-dev \
                crossbuild-essential-armhf

CMD ["/bin/bash"]
```

Running this interactively, I was able to add `SquashFS` to the config (thanks https://www.nayab.xyz/rpi3b-elinux/embedded-linux-rpi3-100-elinux-on-sdcard) and cross-compile the kernel!

Copying the resulting `arch/arm/zImage` as `vmlinuz` and the `arch/arm/boot/dts/bcm*-rpi-*.dt` files to my kernel package, I was able to run gokrazy and it looks good :tada: 

```
gokrazy build timestamp 2022-03-22T16:17:50+01:00
[1970/01/01 01:00:05 gokrazy.go:76: found hardware watchdog "Broadcom BCM2835 Watchdog...
[ 5.1586341] EXT4-fs (mmcblk0p4): mounted filesystem with ordered data mode. Opts:...
hostname "gokrazy"
gokrazy device model Raspberry Pi Model B Rev 2
```

I have to do some more tests, if everything is fine (connect it to the network for instance), but I am very excited! (so I couldn't wait to share it here ^^)

stapelberg, 2022-03-22:
Cool, thanks for sharing!

> So I set-up a docker container with the latest ubuntu (the latest official raspberry kernel requires glibc 2.34, which is only available starting with ubuntu impish).

The `gokr-rebuild-kernel` command does a very similar setup, so perhaps you can just re-use most of it.

The easy step is replacing the upstream kernel URL with the kernel URL you’re interested in: https://github.com/gokrazy/kernel/blob/c77af2125f44f8b0c0612d8bfbf1db25e8cb7c0a/cmd/gokr-build-kernel/build.go#L17

The harder step is wrangling the kernel configs until you have something that works (the config we currently use targets the upstream kernel, which has different drivers enabled for some hardware).

oliverpool, 2022-03-23:
I have created 2 repositories to distribute the firmware and the 32-bits kernel:
- https://github.com/oliverpool/kernel-rpi-os-32
- https://github.com/oliverpool/firmware-rpi (similar to https://github.com/gokrazy/firmware, but retrieve the files from https://archive.raspberrypi.org/debian/ instead of github)

```
GOARCH=arm ./gokr-packer \
    -kernel_package=github.com/oliverpool/kernel-rpi-os-32/dist \
    -firmware_package=github.com/oliverpool/firmware-rpi/dist \
    github.com/gokrazy/hello
```

@cweagans what device do you own? Could you give it a try?

oliverpool, 2022-03-24:
I improved the readme of both repo, to highlight the fact that they are in sync with the https://archive.raspberrypi.org/debian/dists/bullseye/ version (much behind your `kernel` and `firmware` repo, but - hopefully - tested by other people ;)

I also added some github action to check for new version (and eventually compile and comit) each day.
The kernel even gets a tag (`v1.0.0-5.10.103` for instance), so that it looks better in `go.mod` ;)

Before I add some documentation, I think it would be nice to move the repos under the `gokrazy` organization: what do you think about it?

stapelberg, 2022-03-24:
> Before I add some documentation, I think it would be nice to move the repos under the `gokrazy` organization: what do you think about it?

As soon as the repos are under the `gokrazy` org, people will assume some degree of official support.

Hence, I think it’d be better if the repository staid with the person who’s maintaining it (you, currently) :)

cweagans, 2022-03-24:
@oliverpool I will take a look tonight. I have several devices I can test on :)

@stapelberg would you be opposed to a `gokrazy-community` organization or something along those lines?

stapelberg, 2022-03-25:
Sure. I created a gokrazy-community org (and invited @oliverpool). We can put the repositories under that org and label it clearly as community-supported only :)

oliverpool, 2022-04-02:
Since this is working for me, and now officially documented on https://gokrazy.org/platforms/#community, I will close this issue.

@cweagans Were you able to give it a try ? What do you think of the documentation ? 

Feel free to ping me in a new issue if you need help :)

closed 2022-04-02 by oliverpool