gokrazy/gokrazy #129: Running gokrazy in qemu's raspi machines

damdo, 2022-04-27:
For ease of development and portability needs I'm planning to develop on gokrazy from within qemu machines.

For this purpose I've put together (inspired by qemubootery) a qemu config which is able to run gokrazy on an emulated x86_64 machine:
```
 qemu-system-x86_64 \
   -name gokrazy-amd64 \
   -m 4096 \
   -smp $(nproc) \
   -usb \
   -nographic \
   -serial mon:stdio \
   -boot order=d \
   -drive file=drive.img,format=raw \
   -netdev user,id=mynet0,hostfwd=tcp::8080-:80,hostfwd=tcp::2222-:22 \
   -device e1000,netdev=mynet0
```
The above works fine but given that the target hardware for my development is the raspberry pi, I'd like to run a qemu machine that mimics the raspi as close as possible.

It looks like `qemu-system-aarch64` supports the emulation of a raspberry pi 3 `raspi3`.
I'm wondering if anyone has been able to run gokrazy in such an environment. So far I haven't.
Thanks!

stapelberg, 2022-04-28:
Ah, I wasn’t aware of qemu’s `-M raspi3` flag.

When you say you haven’t been able to run gokrazy, can you be more specific about which problem you encountered? Given that it works in qemu for amd64, it should work just as well in qemu for arm64.

Does qemu emulate the Raspberry Pi’s boot loader as well, or is locating the kernel the issue you’re facing?

damdo, 2022-05-03:
So I dug a bit deeper on this, here is what I found.

The main reason running why gokrazy on `qemu-system-aarch64 -M raspi3b` was working for me at first, as you also alluded to, was related to it not emulating the physical Raspberry Pi bootloader.

Which meant the kernel had to be side loaded, rather than it being detected and loaded by the bootloader directly from the first partition of the drive.img produced by gokr-packer. It also needed the `-append` directive for the extra kernel options and the `-dtb` directive for referencing the correct raspi's device tree blob.
Once this was properly set I was able to get gokrazy running on it but with a couple of important caveats:
- the `raspi3b` machine emulation is supported since `qemu v2.12`, but the support for the `raspi3b` [usb controller, which is the sole component that allows networking on the raspi](https://stackoverflow.com/a/64420363), was only introduced in [`qemu v5.1`](https://wiki.qemu.org/ChangeLog/5.1). Which means one won't be able to use networking on the raspi3b before that specific qemu version.
- for the reason that the `raspi3b` networking is provided through the emulation of its usb controller, the network bandwidth is very limited and visibly struggles to also run the gokrazy's web UI, let alone performing a gokrazy update over-the-air (which in my experience always timed out).
- it looks like with `qemu v6.0.0+` the gokrazy kernel is loaded but the `/gokrazy/init` process crashes early. I've been able to isolate it to `gokrazy.Boot(...)` but I haven't looked further. Not yet sure why this happens.

So here is the configuration with which I've been able to get gokrazy working with `qemu-system-aarch64 -M raspi3b`, but **beware** that it might be slow to a point to not be usable:
```bash
# note: ./bcm2710-rpi-3-b-plus.dtb and ./vmlinuz
# will need to be extracted from the `drive.img` produced by gokr-packer.
# an example on how to do so can be found here:
# https://github.com/damdo/gokrazy-on-qemu/blob/f30e31fb538fcf3ece43ac1dbf6c3f7c296e1ede/extract_kernel.sh
# it requires v5.1 <= qemu  <= v5.x to work
    qemu-system-aarch64 \
      	-name gokrazy-arm64-raspi3b \
        -m 1024 \
        -no-reboot \
        -M raspi3b \
        -append "console=tty1 console=ttyAMA0,115200 dwc_otg.fiq_fsm_enable=0 root=/dev/mmcblk0p2 init=/gokrazy/init rootwait panic=10 oops=panic" \
        -dtb ./bcm2710-rpi-3-b-plus.dtb \
        -nographic \
        -serial mon:stdio \
        -drive file=drive.img,format=raw \
        -kernel ./vmlinuz \
        -netdev user,id=net0,hostfwd=tcp::8080-:80,hostfwd=tcp::2222-:22 \
        -device usb-net,netdev=net0
```

A new hope might come from the development work that has been happening around emulating the `raspi4b`, which might even support a Gigabit ethernet controller:
- https://gitlab.com/qemu-project/qemu/-/issues/461
- https://gitlab.com/philmd/qemu/-/commits/raspi4_wip
- https://github.com/MrMino/qemu-raspi4

--

That said once I found out about the (at least for my use case) non-usability of the `raspi3b` emulator for gokrazy, I tried making things work with an emulated machine that would be as close as possible with `raspi3b`: the `arm64` `virt` machine.

This machine is designed to be a flexible arm64 emulator and provides plenty of memory while rocking an emulated cortex-a72.
To make it run with gokrazy it required some hacking similar to what I described above around how kernel extraction but then the configuration looked quite similar:
```
# note: ./bcm2710-rpi-3-b-plus.dtb and ./vmlinuz
# will need to be extracted from the `drive.img` produced by gokr-packer.
# an example on how to do so can be found here:
# https://github.com/damdo/gokrazy-on-qemu/blob/f30e31fb538fcf3ece43ac1dbf6c3f7c296e1ede/extract_kernel.sh
    qemu-system-aarch64 \
      -name gokrazy-arm64 \
      -m 3G \
      -smp $(nproc) \
      -M virt,highmem=off \
      -cpu cortex-a72 \
      -nographic \
      -netdev user,id=net0,hostfwd=tcp::8080-:80,hostfwd=tcp::2222-:22 \
      -device e1000,netdev=net0 \
      -drive file=drive.img,format=raw \
      -kernel vmlinuz \
      -append "console=tty1 console=ttyAMA0,115200 root=PARTUUID=60c24cc1-f3f9-427a-8199-2e18c40c0001/PARTNROFF=1 init=/gokrazy/init rootwait panic=10 oops=panic"
```
Although this still was requiring some extra steps to extract the kernel.

So I thought I could leverage the qemu UEFI bootloader to achieve a smoother experience by loading the kernel directly from within the `drive.img`, similarly to what the experience with `qemu-system-x86_64` and gokrazy, already provides.

To do so, though, this required some changes to the gokrazy's packer to include some extra files in the first partition under `/EFI/BOOT/` to introduce the required EFI and grub files and configs, which can be found here: https://github.com/damdo/tools/commit/e43da43f5996bf4595d10b3b1705aaeac1322df0
@stapelberg it would be nice to have them included in gokrazy/tools :)
Which then allowed this config to work and auto load the kernel:
```
    qemu-system-aarch64 \
      -name gokrazy-arm64 \
      -m 3G \
      -smp $(nproc) \
      -M virt,highmem=off \
      -cpu cortex-a72 \
      -nographic \
      -netdev user,id=net0,hostfwd=tcp::8080-:80,hostfwd=tcp::2222-:22 \
      -device e1000,netdev=net0 \
      -drive file=drive.img,format=raw \
      -bios /usr/share/qemu-efi-aarch64/QEMU_EFI.fd
```
I put together a repo with an environment to reproduce all the above and to more in general experiment with gokrazy on qemu. It can be found at: https://github.com/damdo/gokrazy-on-qemu

stapelberg, 2022-05-03:
Thanks for sharing!

I don’t mind adding arm64 UEFI support to gokrazy, but do we really need grub to make that work? On amd64 we use systemd-boot only.

By the way, you can set -target_storage_bytes to 2 GB directly instead of resizing the image afterwards.

damdo, 2022-05-04:
> do we really need grub to make that work? On amd64 we use systemd-boot only.

Previously the .efi file that I was using for my tests, which coming from [here](http://download.opensuse.org/ports/aarch64/factory/repo/oss/EFI/BOOT/bootaa64.efi.mirrorlist), didn't support `cmdline.txt` so I had to load grub and define a grub.cfg to load the kernel.
Now I tried using the systemd-boot .efi (`systemd-bootaa64.efi`) and put it at `/EFI/BOOT/bootaa64.efi`. With this the UEFI bootloader is able to pick up the `cmdline.txt` directives and load the kernel correctly (and without grub!) :)

> I don’t mind adding arm64 UEFI support to gokrazy

Great! I'll open a PR then 👌 

> By the way, you can set -target_storage_bytes to 2 GB directly instead of resizing the image afterwards.

👍

damdo, 2022-05-04:
Also given that the main purpose behind opening this issue (running gokrazy on qemu's raspi machines), has been achieved (even though the performance are not "great" yet), I think we can close the issue.

closed 2022-05-07 by stapelberg