gokrazy/gokrazy #130: Updating with bad TLS certificates makes device unreachable

joneskoo, 2022-05-01:
## Platform

I’m using:

* [x] gokrazy/rpi02w
* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I was updating TLS certificate but accidently only updated the key file and not certificate so the key didn't match the certificate.

gokr-packer -update=yes completed successfully but gokrazy didn't come up.

Power cycling didn't recover it so it doesn't recover from backup partition either.

## Expected behavior

Validate TLS certificate matches key before updating.

(Is it possible to detect unbootable state and fall back to last known good partition by power cycling?)

joneskoo, 2022-05-01:
Is gokrazy intended to fall back to known-working partition? It would be really powerful if updates would only mark partition active after first successful boot on it and would have been a better fix, regardless of my fix to gokr-packer in the PR.

stapelberg, 2022-05-01:
> Is gokrazy intended to fall back to known-working partition? It would be really powerful if updates would only mark partition active after first successful boot on it and would have been a better fix, regardless of my fix to gokr-packer in the PR.

Yes, but that’s not quite done yet: https://github.com/gokrazy/gokrazy/issues/75

closed 2022-05-01 by stapelberg