gokrazy/gokrazy #144: DHCP client started for non-existant eth0 on Pi Zero series

Doridian, 2022-09-06:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [x] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64
* [x] gokrazy-community/rpi-zero 

## Observed behavior

The DHCP service is always set to autostart unless a `dontstart.txt` is explicitly provided for it.
The Pi Zero series do not have a wired ethernet interface, therefor the process will constantly crash and restart.

## Expected behavior

The DHCP service could see there is, in fact, no eth0, and exit cleanly. (Maybe wait a little bit in case the user actually plugged in a wired USB ethernet adapter)

stapelberg, 2022-09-07:
> The DHCP service could see there is, in fact, no eth0, and exit cleanly. (Maybe wait a little bit in case the user actually plugged in a wired USB ethernet adapter)

How would we distinguish the situation where there is no eth0 (ever) from one where the user plugs in a USB ethernet adapter after a few days of runtime, or from the situation where the USB ethernet adapter is behind an unstable USB hub and disappears/re-appears from the bus sometimes?

I don’t think we can make that distinction, so I think the current state (a restarting process) is actually desired.

Doridian, 2022-09-07:
I've always considered `gokrazy` as an appliance-type system, where you plug everything in before you boot it.

If supporting (un-)plugging hardware is desired, well, you can drag that argument on forever (what if they plug in two USB ethernet adapters behind an unreliable hub and they fight over who is eth0, etc etc).
The ideal solution would probably be something like a minimal hotplug or udev that quite simply detects whenever a network interface appears and latches onto it possibly. (That way you could also avoid the wifi process starting dhcp by itself)

Right now, I just don't like how `dhcp` floods the logs by default with messages. Maybe, another more immediate suggestion could be for `dhcp` to just wait until the interface it wants exists as opposed to restarting continuously?

stapelberg, 2022-09-07:
> I've always considered `gokrazy` as an appliance-type system, where you plug everything in before you boot it.

Yes, it’s meant for appliances, but even in an appliance use-case, I find it convenient to be able to re-configure my network connectivity as I see fit :) It certainly helps during development and initial installation, too.

If I have to choose between log flooding and graceful recovery, I’ll chose graceful recovery.

> Maybe, another more immediate suggestion could be for `dhcp` to just wait until the interface it wants exists as opposed to restarting continuously?

Yes, that’s a good suggestion. If the specified interface is not found, we can make `dhcp` wait for it to appear. Can you send a PR for that please?

PR #187 (2023-04-06): dhcp: wait indefinitely for interface to appear
On systems without an ethernet interface (like the Raspberry Pi Zero) the `dhcp` program goes into a crash loop because it can't find eth0.

This PR adds an initial loop to wait for the given interface name to appear before continuing with the rest of the program.

**NOTE**: Unfortunately I can't fully test this code because I don't have any ethernet dongle; on a plain Raspberry Pi Zero 2 W it works fine: the `dhcp` process is not crash looping anymore.

Fixes #144