gokrazy/gokrazy #147: “instance-centric” config re-design

stapelberg, 2022-09-25:
I have been thinking a bit about how to improve the historically grown configuration surface of gokrazy.

The result is a design document that you can find here:

https://docs.google.com/document/d/1AUDmc8lygLR59O29Oc4ckv-Buk2R9zrvPDO9JbPOts8/edit?usp=sharing

If you have any feedback or questions on these changes, please comment them here and I will modify the design as appropriate (if needed).

christian-heusel, 2022-09-26:
Looks like a great design and overall improvement! 🥳 👍🏻  
I have a few thoughts and suggestions:
- Regarding the upgrade path:
  - The proposal includes something like `gok config` to print the (instance aware) current config in the new format, maybe with some blank fields that require modification, that sounds great!
  -  something like `gok config --default` or similar to just get a blank config with all the availiable options would also be great
- Can we include a possibility for comments in the config file even though JSON does not support them? 😛  
- Would it make sense to also have the `config.json` on the instance in order to document its state?

Can I help anywhere with the implementation / documentation / testing of this feature? 😃

stapelberg, 2022-09-27:
Hey Christian, thanks for taking a look and sharing your feedback!

> * The proposal includes something like `gok config` to print the (instance aware) current config in the new format, maybe with some blank fields that require modification, that sounds great!

It would need to be a flag in the existing `gokr-packer` command so that folks can easily change their existing setup. That gives us the freedom to not carry over all these flags to the `gok` command.

> * something like `gok config --default` or similar to just get a blank config with all the availiable options would also be great

That would be the suggested `gok new` subcommand :)

> * Can we include a possibility for comments in the config file even though JSON does not support them? 

I agree that would be nice (and ignoring trailing commas), but it seems like the Go standard library’s json package doesn’t support that, and I would like to stick to the standard library.

As the design mentions, we could add a hook that `gok` calls, where you can translate from whichever format you want (json with comments, TOML, CUE, …) into json. 

> Can I help anywhere with the implementation / documentation / testing of this feature?

I’ll update this issue once I have the prototype ready for testing :)

stapelberg, 2022-10-04:
Here’s a branch that contains the first experimental commit in this direction:

https://github.com/gokrazy/tools/commits/config-redesign

Use `go install ./cmd/...` to install the new `gokr-packer` and `gok` versions. `gokr-packer` should keep working as before.

You can try out the new `gok` command like so:

1. Add `-write_instance_config=scan2drive` flag to your existing `gokr-packer` invocation. That will result in a `$HOME/gokrazy/scan2drive/config.json` file.
2. Run `gok update -i scan2drive` to do an update using the `gok` CLI. 

Integrating per-package config and `~/.config/gokrazy` into the JSON struct is not done yet.

Also, aside from `Packages` and `Hostname`, most other flags are in `InternalCompatibilityFlags` for now. I’ll go through them one by one to see what the desired name and structure is before I commit to the JSON config representation.

christian-heusel, 2022-10-05:
I am up and running again after creating the instance file and deploying via `gok update -i ...`, great work 😄 👍🏻

stapelberg, 2022-10-18:
Thanks for trying it out :) I have since also switched to using the config-redesign branch for all my devices, so at least the `gokr-packer` interface seems stable.

I just pushed commit https://github.com/gokrazy/tools/commit/e1181f52527e50c7e02ba2fe1dc44d2dc3012f9d, which makes it possible to express per-package configuration in config.json instead of the file system. Feel free to take that for a spin, too, and report back :)

I’ll also add a list of milestones I want to achieve before merging the changes of the config-redesign branch.

andig, 2022-10-20:
@stapelberg I'm following the new developments with keen interest. I'd like to provide images for https://evcc.io. From CI side, the biggest pain has been creating the filesystem which is alleviated by github.com/gokrazy/mkfs. Going forward, I'm wondering what your proposed approach is for CI builds. It seems to make sense to use `config.json` for package build and runtime config instead of the fixed `flags`/`buildflags` directories which just clutter the repo. However, with the redesign being instance-centric, I'm wondering where to put a `config.json` in the repo. Would something like 

    gok run --instance_dir packaging/gokrazy

work (with `packaging/gokrazy/config.json` containing the config)? Or would that create side effects for hostname etc?

stapelberg, 2022-10-21:
Hey @andig, thanks for your feedback!

Conceptually, only the *representation* of the configuration changes with this effort — what previously was in multiple files is now in one file, and locating this file is now done via the instance directory and instance name concepts that is shared across the different subcommands like `gok overwrite` or `gok run`.

In other words: I don’t think this changes anything with regards to what you can (or can’t!) do in CI with gokrazy.

As you suggested, you could switch from files to config.json and point your `gok` tool to a subdirectory of your repository’s git checkout dir.

For configuration that is dynamic (like the ldflags example), you can generate (or modify) config.json programmatically from within your CI pipeline. (You can also do that today with the file-based structure.)

In general, I’m happy to talk about how to structure a CI pipeline or which improvements we’d need to make that easier and/or more capable :) Maybe you could open a separate issue and list the pain points you currently see?

BTW, regarding the version injection via ldflags specifically: are you aware of the buildinfo that Go stamps into binaries by default in more recent versions? I’m using that in `gokr-packer` (see https://github.com/gokrazy/tools/blob/c87c806562c3493bd11aa5af46ce5175a70ca95e/cmd/gokr-packer/packer.go#L872), and since Go 1.18 you can even get VCS information, which I’m using in `gok --version`: https://github.com/gokrazy/tools/blob/e1181f52527e50c7e02ba2fe1dc44d2dc3012f9d/cmd/gok/cmd/root.go#L34

andig, 2022-10-21:
> BTW, regarding the version injection via ldflags specifically: are you aware of the buildinfo that Go stamps into binaries by default in more recent versions? 

OT: I'm aware of that discussion but unfortunately it doesn't contain git tags yet. There's an open Go issue- if of interest I'll dig for it.

andig, 2022-11-08:
> In other words: I don’t think this changes anything with regards to what you can (or can’t!) do in CI with gokrazy.

Definitely, only a question of simplicity using it imho. Opened https://github.com/gokrazy/gokrazy/issues/153.

stapelberg, 2022-11-20:
I just pushed a new version of the `config-redesign` branches:

* https://github.com/gokrazy/internal/commits/config-redesign
* https://github.com/gokrazy/tools/commits/config-redesign

(To test: clone both, check out the correct branch in both, in `tools`, add a replace directive pointing to your `internal` checkout, then `go install ./cmd/...`)

Notably, `gok add` is now implemented, which enables the following workflow, for local or remote packages:

```shell
# Create 'rw' instance
gok -i rw new

# Include the Go program living in ~/regelwerk
gok -i rw add ~/regelwerk

# Write to an SD card:
gok -i rw overwrite -full=/dev/sdx
```

Still missing: integration tests, a documentation reference for the config.json format, and updating the website for the new commands. It might take a few weeks until I get to the remaining parts, as I’ll be traveling soon.

But if you’re feeling adventurous, feel free to give this a shot and report back if it worked for you :)

andig, 2022-11-21:
Seems something's odd:

```
go install github.com/gokrazy/tools/cmd/gok@config-redesign
go: downloading github.com/gokrazy/tools v0.0.0-20221120213237-b4e250710f94
go: finding module for package github.com/gokrazy/internal/instanceflag
../../go/pkg/mod/github.com/gokrazy/tools@v0.0.0-20221120213237-b4e250710f94/cmd/gok/cmd/add.go:15:2: module github.com/gokrazy/internal@latest found (v0.0.0-20221006065815-2f79bee2fc4c), but does not contain package github.com/gokrazy/internal/instanceflag
```

damdo, 2022-11-21:
A bit late to the party but I really like the new design!

IMO it improves the ergonomics of the cli and simplifies configuration and testing.
I think it also gives us more space to extend the tools available in the cli in the future!  🎉 

> As the design mentions, we could add a hook that gok calls, where you can translate from whichever format you want (json with comments, TOML, CUE, …) into json.

+1 on the hooks too! I think having the ability to write config in, say, YAML (😬) would be nice.

I'm also happy to contribute to this, let me know if I can be of any help.

stapelberg, 2022-11-23:
> Seems something's odd:
> 
> ```
> go install github.com/gokrazy/tools/cmd/gok@config-redesign
> ```

No, you can’t use `go install` like that unfortunately. You really need to check out `internal` and `tools` on local disk, switch to the `config-redesign` branch, add a `replace` directive, and use `go install`.

stapelberg, 2022-11-23:
> As the design mentions, we could add a hook that gok calls

Ah, I should have updated this comment, too. Since then, I updated the design to remove the hook concept.

I have come to realize that “gokrazy reads the files as they are” is an important principle that allows for introspection/debugging by using “cat config.json && gok …”, for example in CI pipelines. 

Running any required custom tools manually before invoking gok shouldn’t be too much trouble. If a shell wrapper script turns out to be complicated, we can look into adding functionality to make that easier, but ideally not in the form of running a hook.

damdo, 2022-11-23:
> “gokrazy reads the files as they are” is an important principle that allows for introspection/debugging

I understand, yes it makes sense to me. I think we can do without any of that and try and extend it if we have the need  further on.

stapelberg, 2023-01-08:
Happy new year everyone!

The config re-design work has now reached a point where I’m happy to share it for wider testing. I’m not aware of any issues anymore, and I have successfully updated my devices (simple and more complicated ones) with the new code 🎉

---

To install the new `gokr-packer` and `gok` tools from the branch, use:

```
go install -v github.com/gokrazy/tools/cmd/...@config-redesign
```

To double-check that you’re getting the right version:

* `go install` should print a line like `go: downloading github.com/gokrazy/tools v0.0.0-20230107175926-8f9863a4c2b3`, where the last part of the version should be a commit id you can find at https://github.com/gokrazy/tools/commits/config-redesign
* run `gok version` to ensure you see that same commit

---

You can find an updated preview version of the gokrazy website (including documentation) at https://t.zekjur.net/2023-01-08-preview-gokrazy/

In particular, the [migration guide](https://t.zekjur.net/2023-01-08-preview-gokrazy/userguide/migration-guide/) and the [instance config reference](https://t.zekjur.net/2023-01-08-preview-gokrazy/userguide/instance-config/) might be of interest.

But also feel free to play around with the `gok new` and `gok add` commands; see [quickstart](https://t.zekjur.net/2023-01-08-preview-gokrazy/quickstart/) and [packages → caddy](https://t.zekjur.net/2023-01-08-preview-gokrazy/packages/caddy-http-server/), for example.

---

Unless significant new issues are uncovered in your tests, I plan to merge the branch in about 7 days from today, so please report any issues as early as you can :) 

Thanks in advance!

stapelberg, 2023-01-12:
I have been crossing t's and dotting i's, and everything should be nicely in shape now.

I’m planning to merge on Saturday, so if you want to give this a try to help with testing, please do so until then :) Thanks!

stapelberg, 2023-01-14:
Status update: This change is now implemented! 🎉 The documentation at https://gokrazy.org/ is up-to-date. See the [Migration Guide](https://gokrazy.org/userguide/migration-guide/) to learn how to switch from `gokr-packer` to `gok`.

---

I’ll leave this issue open for a while to make it easy for people to find, and as a place to allow for questions/feedback.

damdo, 2023-01-15:
Awesome! I've updated my devices and things work like a charm. 🙌 
The experience of using a centralised config file is great so far! Also very nice job with updating the documentation and writing a migration guide. Thank you very much 🎉

stapelberg, 2023-01-15:
I’m glad to hear everything went okay with the migration for you :) 

Thanks for reporting back! By the way, if you want, you could propose https://github.com/damdo/gokrazy-machine in a pull request now. Or would you prefer if I tried to merge it myself?

damdo, 2023-01-15:
> Thanks for reporting back!

No worries, happy to help.

> Thanks for reporting back! By the way, if you want, you could propose https://github.com/damdo/gokrazy-machine in a pull request now. Or would you prefer if I tried to merge it myself?

Ah yes that would be nice! Yes, I can put up a draft PR and then we can work together on that. I think there are a couple of TODOs in there for which I will need your help.

stapelberg, 2025-02-23:
I’ll close this issue as the instance-centric config has been the default for years by now. Feel free to open new issues if you have any questions or problems.

closed 2025-02-23 by stapelberg