gokrazy/gokrazy #149: invalid value for flag -ldflags: parameter may not start with quote character '

andig, 2022-10-17:
## Observed behavior

Building the image fails with `ldflags` parameter:

    invalid value "'-X github.com/evcc-io/evcc/server.Version=0.105.1 -X github.com/evcc-io/evcc/server.Commit=43d4221cf -s -w'" for flag -ldflags: parameter may not start with quote character '

## Expected behavior

It would help if the image succeeded. It should since `-ldflags='...'` works fine for normal build (shell quoting) and the flag can be passed as-is to the compiler.

This is the buildflags file:

  cat buildflags/github.com/evcc-io/evcc/buildflags.txt
  -tags=release,gokrazy
  -ldflags='-X github.com/evcc-io/evcc/server.Version=0.105.1 -X github.com/evcc-io/evcc/server.Commit=43d4221cf -s -w'

andig, 2022-10-17:
Nvm. Can be fixed in the build script. Root cause is probably, that the go compiler is invoked without intermediary shell.

closed 2022-10-17 by andig

stapelberg, 2022-10-17:
Yes, we intentionally pass each line directly as arguments to `go build`, without any shell escaping, quoting or other processing in between. This can be surprising when porting a shell-based invocation, but bottom line is the clearest way to implement it.

References: 

reading buildflags: https://github.com/gokrazy/tools/blob/c87c806562c3493bd11aa5af46ce5175a70ca95e/cmd/gokr-packer/packer.go#L246

applying buildflags: https://github.com/gokrazy/tools/blob/c87c806562c3493bd11aa5af46ce5175a70ca95e/packer/gotool.go#L292