gokrazy/gokrazy #150: Automate creating user partition

andig, 2022-10-17:
When I last used gokrazy, creating the user partition was cumbersome and involved using loopback devices. This is painful on MacOS and in Docker, both. I've noticed that the `gokr-packr` now advertises to do this without loopback:

     /sbin/mkfs.ext4 evcc_0.105.1.image -F -E offset=1157627904

This seems possible inside Docker. To automate the process, it would be nice if this step could be included in the build process, maybe by adding a command line parameter for a Golang template like:

    --userdata "mkfs.ext4 {{ .Image }} -F -E offset={{ .Offset }}

That would allow running automated builds even in CI and Docker without additional manual steps or parsing the `gokr-packr` output.

stapelberg, 2022-10-17:
I have good news for you: https://github.com/gokrazy/mkfs. Does that fit your use-case?

andig, 2022-10-17:
Wow. It just got even crazier than when I discovered WPA Wifi ;)

closed 2022-10-17 by andig

stapelberg, 2022-10-17:
> Wow. It just got even crazier than when I discovered WPA Wifi ;)

Let me make you aware of https://github.com/gokrazy/gokrazy/issues/147, then, which is another big development (still in progress) :)