gokrazy/gokrazy #153: Dynamic configuration with instance-centric redesign

andig, 2022-11-08:
Consider some, maybe typical, requirements for gokrazy applications:

	go install github.com/gokrazy/tools/cmd/gokr-packer@main

	mkdir -p flags/github.com/gokrazy/breakglass
	echo "-forward=private-network" > flags/github.com/gokrazy/breakglass/flags.txt

	mkdir -p flags/github.com/evcc-io/evcc
	echo "--sqlite /data/evcc.db" > flags/github.com/evcc-io/evcc/flags.txt

	mkdir -p buildflags/github.com/evcc-io/evcc
	echo "$(BUILD_TAGS),gokrazy" > buildflags/github.com/evcc-io/evcc/buildflags.txt
	echo "-ldflags=$(LD_FLAGS)" >> buildflags/github.com/evcc-io/evcc/buildflags.txt

	gokr-packer -overwrite=$(IMAGE_FILE) -target_storage_bytes=1258299392 $(IMAGE_OPTIONS)
	gzip -f $(IMAGE_FILE)

Basically all config is static and would lend itself well to being part of the git repo. I'm currently creating the config dynamically for the following reasons:

- config directory cannot be changed, i.e. tied to `flags` and `buildflags` which may not perfectly fit existing repo structure
- with instance.centric redesign, where `--instance_dir` can be chosen, dynamic config (`LD_FLAGS` above) is the remaining problem as it would require updating the `json` file on each build (and hence not suitable to commit).

Would it make sense to a allow a third option of configuration using per-package build arguments on the command line? Maybe something along the lines of 

    gokr-packer/gok --build-flags "github.com/evcc-io/evcc=-ldflags=$(LD_FLAGS)" ...

stapelberg, 2022-11-12:
Hey @andig, thanks for filing this issue!

> Basically all config is static and would lend itself well to being part of the git repo.

Agreed. The static parts of the configuration should best be versioned as a regular file in the git repository.

> * dynamic config (`LD_FLAGS` above) is the remaining problem as it would require updating the `json` file on each build (and hence not suitable to commit).

Updating a JSON file is simple enough using `jq`.

As an example, you could commit `evcc/_gokrazy/config.json.tmpl` to git with the following content:

```json
{
    "Packages": [
        "github.com/gokrazy/breakglass",
	"github.com/evcc-io/evcc"
    ],

    "PackageConfig": {
	"github.com/gokrazy/breakglass": {
	    "CommandLineFlags": [
		"-forward=private-network"
	    ]
	},
	"github.com/evcc-io/evcc": {
	    "GoBuildFlags": [
		"-ldflags=-X github.com/evcc-io/evcc/server/updater.Password=@UPDATER_PASSWORD@"
	    ],
	    "CommandLineFlags": [
		"--sqlite=/data/evcc.db"
	    ]
	}
    },

    "Hostname": "evcc"
}
```

And then, in your CI (GitHub Actions) steps, you run a shell script like this one, placed in `evcc/_gokrazy/build.sh`, from a throw-away build directory like `evcc/_build`:

```shell
#!/bin/sh

mkdir -p gokrazy-instance-dir/evcc

# Copy the existing go.mod as a starting point
builddir=gokrazy-instance-dir/evcc/builddir/github.com/evcc-io/evcc
mkdir -p "$builddir"
cp ../go.{mod,sum} "$builddir"
# gokrazy expects the module path gokrazy/build/evcc, and we need
# to instruct the go tool to find evcc in our working directory.
go mod edit \
   -module gokrazy/build/evcc \
   -replace github.com/evcc-io/evcc=$PWD/.. \
   "$builddir/go.mod"

# Take the _gokrazy/config.json.tmpl from git and derive a config.json,
# with the updater password placeholder replaced.
jq \
    --arg val "secretpassword" \
    '.PackageConfig["github.com/evcc-io/evcc"].GoBuildFlags[0] |= sub("@UPDATER_PASSWORD@"; $val)' \
    ../_gokrazy/config.json.tmpl > gokrazy-instance-dir/evcc/config.json

# For debugging:
cat gokrazy-instance-dir/evcc/config.json

# Write /tmp/root.squashfs only for quick testing
gok \
    --instance_dir=$PWD/gokrazy-instance-dir \
    overwrite \
    --instance=evcc \
    --root=/tmp/root.squashfs
```

This script creates a gokrazy instance that uses the exact same `go.mod`/`go.sum` as the working copy, then creates a `config.json` with the CI run-time parameters and builds a gokrazy image.

I realize that the builddir / go.mod handling is pretty cumbersome, and I’m interested in making it easier. Maybe the planned `gok new` could have a mode that creates such an instance based on the provided `go.mod`.

But hopefully that illustrates how you can best build a gokrazy instance in CI:
* versioning the largely-static part of your config in git as a regular file
* in your CI, create and work in a throw-away `_build` directory underneath your working copy (add `_build` to `.gitignore`)
* run a script that creates, configures and builds a gokrazy instance in the `_build` directory
* in that script, you can replace placeholders from your static config to derive the config you need

Does that make sense? Happy to answer any follow-up questions

stapelberg, 2022-11-12:
Regarding an additional flag configuration surface: I see several big problems with that approach that make me want to avoid it if at all possible:

1. Having a single JSON file is a very simple to understand conceptual architecture, which is a property I want to preserve. The lookup order for configuration is already very complex because we need to keep supporting existing setups (at least for a reasonable time frame), so I don’t want to add another override mechanism into the mix depicted below (in particular because the evaluation order goes against the rest of the image — for `gokr-packer`, you can derive a `config.json` from flags, but as soon as you use `config.json`, flags are ignored, not treated as an override):

    [image]

    Keep in mind that all the possible ways of configuration need to be documented and explained with examples. The mental complexity for all users drastically increases the more different ways we need to document.

2. Introducing a second way of parsing configuration (via command line flags) has the potential for hard-to-understand semantics in terms of quoting, and risks that the two configuration surfaces diverge in terms of what is possible in each of them.

andig, 2022-11-18:
Excellent approach combining Go templates. Thank you for the password example- issue solved.

stapelberg, 2022-11-18:
Thanks for reporting back. I’m glad the example helped :)

I’ll keep this issue open to track adding `gok add` to simplify the setup, and also to document typical CI setups better on the website.

stapelberg, 2023-01-14:
> I’ll keep this issue open to track adding `gok add` to simplify the setup, and also to document typical CI setups better on the website.

`gok add` is now available, and CI setups are now documented in our userguide: https://gokrazy.org/userguide/automation/#githubactions

closed 2023-01-14 by stapelberg