gokrazy/gokrazy #155: How to configure resolution? fresh install shows invalid mode on monitor

WhiteHexagon, 2022-11-10:
## Platform

I’m using:
* [x] gokrazy/rpi4b

## Observed behavior
A couple hiccups with the Quickstart guide, but finally managed to boot the new image.  I see maybe a page and a half of small text fly up the left side of the screen, and then my dell monitor says there is a timing issue, and requests a change to 1920x1200@60Hz.  Same thing on Pi400.  Not sure where to check/change the resolution, and cant see anything obvious in the various repos or docu.  Also not sure how this is gonna connect to my wifi network, since it seems to expect ethernet?

## Expected behavior
some kind of successful boot (just the 'hello' app installed)

WhiteHexagon, 2022-11-11:
I tried video and photo but the text is only there maybe .5s 

Yes both other Pi normally work with this monitor, but I also now tried a TV, same blank screen (but no resolution warning)

I already accidentally tried without fbstatus, thinking it was something to do with social media :)

I tried a different Pi4 (since I recalled doing some 64bit upgrades earlier in the year...).  Same.  I also tried the second mini hdmi output.  Same.  But then I tried a Pi3 and it worked!! both the monitor and TV :)  So seems like the sdcard image is written out correctly, maybe just something specific to my Pi4s?

uname -r
5.15.32-v8+
I cant seem to discover the firmare version, but this is the Pi model:
Pi4 Model B Rev 1.1

stapelberg, 2022-11-11:
Very weird! I never have run into any problem like that, and have never heard from anyone about a similar problem.

You can try adjusting the config.txt as described here: http://wiki.sunfounder.cc/index.php?title=Adjust_Resolution_for_Raspberry_Pi

To modify config.txt, clone https://github.com/gokrazy/kernel and add a `replace` directive to the relevant `go.mod` file (`builddir/github.com/gokrazy/kernel/go.mod`) to point to your local directory. Then, subsequent `gokr-packer` runs should pick up your modified file. Double-check by mounting the first SD card partition and looking at the file :)

WhiteHexagon, 2022-11-12:
WiFi is working! (testing on the Pi3) it is allocated a private IP address :)  I used the option with -tls=self-signed, but when I try to access the IP address, I get security exceptions in both firefox and brave (assuming basic auth. username is gokrazy?).  Do I need to do something more?

I haven't been able to test the Pi4 tips so far, will let you know.

stapelberg, 2022-11-12:
> I used the option with -tls=self-signed

Why not try without TLS first? We have encrypted WiFi support now, so you can just use regular HTTP.

> I get security exceptions in both firefox and brave

Can you provide a screenshot? The exact error message is important.

> assuming basic auth. username is gokrazy?

Yes, it is.

> Do I need to do something more?

For self-signed TLS, you either need to import the CA certificate into your browser, or ignore the security warning about an untrusted certificate. I think in Chrome you need to type `thisisunsafe`.

WhiteHexagon, 2022-11-14:
I saw the TLS option as part of the WiFi guide, hence added it without thinking :)  but without that option I can finally access the web interface, how exciting :)  That is using secure WiFi too :)  I have been hunting around for a micro-kernal to run my Go code, but I think this offers a really nice 'in-between' environment, well done :)

So another curiosity, it seems this latest sdcard image (I always overwrite) still only boots Pi3, but now corrupts? the framebuffer.  So everything works, and the 'status' screen with the running 'top'? command is there, but the gokrazy gopher logo is missing, and a large border added.  I try to attach a picture.  Thanks for the support!

WhiteHexagon, 2022-11-14:
Is there somewhere I can email the picture please? my browser is not working fully with the new 'modern web' git.

WhiteHexagon, 2022-11-14:
[image]

WhiteHexagon, 2022-11-14:
I tried to zoom that area (and cut-out the bg to reduce file size)

[image]

stapelberg, 2022-11-20:
> I saw the TLS option as part of the WiFi guide, hence added it without thinking :)

Thanks, I updated the userguide to make it more clear that encrypted WiFi is preferred.

> So another curiosity, it seems this latest sdcard image (I always overwrite) still only boots Pi3, but now corrupts? the framebuffer. So everything works, and the 'status' screen with the running [...]

Yeah, the `fbstatus` program does not cope so well with different resolutions right now. I’ll transfer this issue to the fbstatus repository for tracking.