gokrazy/gokrazy #156: Task Scheduling

arky, 2022-11-19:
Is there a way to do task scheduling in gokrazy. I would like to launch a my [miniooni](https://github.com/ooni/probe-cli) to run every day at midnight. 

## Platform

I’m using:

* [x] gokrazy/rpi3b
* [x] gokrazy/rpi3b+

## Observed behavior
Currently the program runs continuous or with user launching it from web interface. 
 
## Expected behavior
The program is launched based on pre-set schedule.

stapelberg, 2022-11-19:
Hey @arky 👋

gokrazy doesn’t include any built-in facilities to do this, but you can easily write a few lines of Go code around your logic (or around starting a separate process) that takes care of scheduling. For an example, check out https://github.com/stapelberg/zkj-nas-tools/blob/master/dornroeschen/main.go

If you want to run an external process, you probably want to tell gokrazy to not start said process automatically, for which you can use the `dontstart` file, see https://dcpri.me/2022/08/31/restic-rest-server-gokrazy/ for an example. I realize this file is not currently documented, but that will be addressed as part of https://github.com/gokrazy/gokrazy/issues/147 

Hope this helps, and let me know if you have follow-up questions.

arky, 2022-11-19:
Thank you @stapelberg I'll follow your suggestions and contribute back some notes and docs to the project.  

Closing this issues as resolved for now.

closed 2022-11-19 by arky