gokrazy/gokrazy #157: Raspberry Pi Zero 2 doesn't boot further than the "rainbow screen"

eliasnaur, 2022-11-20:
NOTE: This is very likely an issue at my end, but I didn't find a mailing list or forum to seek help. 

## Platform

I’m using the Raspberry Pi Zero 2

## Observed behavior

Device boots to the "rainbow screen" on HDMI, but console doesn't come up 

## Expected behavior

Device boots further, and the kernel console log and/or the gokrazy hello program log is displayed on HDMI.

Some notes:

- Build with `gokr-packer -serial_console disabled -overwrite /dev/disk4 github.com/gokrazy/hello` on macOS and NixOS (`/dev/sda`).
- SD-card seems to be fine; I mounted the EFI and root volumes in NixOS and listed the various directories.
- The device boots https://seedsigner.com fine.
- The device has had its radios (bluetooth/wifi) physically disabled.

Also, the various documentation and tools says that only RPi 3+4 are supported, but https://github.com/gokrazy/gokrazy/issues/96 is completed. I assume Zero 2 is supported.

Any pointers to debugging this issue is much appreciated.

stapelberg, 2022-11-20:
Hey Elias, good to see you here! 👋

Yes, the Pi Zero 2 W is supported (if in doubt, https://gokrazy.org/platforms/ is canonical, though I am interested in cleaning up any places that are misleading), and I am using one in production usage. 

Can you share the full output of your `gokr-packer` run please (removing any passwords)? In particular, I’m curious if you are on the latest version of `github.com/gokrazy/kernel` and `github.com/gokrazy/firmware`.

I can upload a known-working image if you want, so that we could narrow it down to whether the hardware or software part is broken (likely software, but you never know), and to give you an opportunity to compare and bisect on a file-level?

BTW: I think on macOS, it’s better to use `/dev/rdisk4` instead of `/dev/disk4`, but that should only be a performance difference, not a correctness difference.

stapelberg, 2022-11-20:
To ensure things are working fine in principle, I grabbed my spare Raspberry Pi Zero 2 W, soldered pin headers onto it, and enabled the bootloader debug log:

```shell
# Be sure to start with an empty directory, and specifically no builddir/, otherwise you need to edit
# the go.mod files in the respective builddir subdirectory!
mkdir ~/gokrazy/elias
cd ~/gokrazy/elias
go mod init
go mod edit -replace github.com/gokrazy/kernel=/home/michael/go/src/github.com/gokrazy/kernel go.mod
go mod edit -replace github.com/gokrazy/firmware=/home/michael/go/src/github.com/gokrazy/firmware go.mod
# NOTE: no -serial_console=disabled flag here:
gokr-packer -overwrite=/dev/sda \                                                                       
  github.com/gokrazy/fbstatus \                                                                                                                                                                            
  github.com/gokrazy/hello \
  github.com/gokrazy/serial-busybox
```

The diff in the kernel repository is:
```patch
diff --git i/config.txt w/config.txt
index 1c19159..0ba9aaa 100644
--- i/config.txt
+++ w/config.txt
@@ -1,6 +1,6 @@
 # Switch the CPU from ARMv7 into ARMv8 (aarch64) mode
 arm_control=0x200
 
-enable_uart=0
+enable_uart=1
 
 kernel=vmlinuz
```

In the firmware repository, use:
```shell
sed -i -e "s/BOOT_UART=0/BOOT_UART=1/" bootcode.bin
```

Everything boots as expected. Here is a full successful boot log including the Pi bootloader and Linux kernel log: https://gist.github.com/stapelberg/d32215ce43de1494817361e74d88a138

I did this on my Linux machine, so I’ll give it a shot on macOS, too, just in case.

eliasnaur, 2022-11-20:
Hi Michael!

A known-good image would be good. Unfortunately I don't have an UART debugger, so I'll have to do with HDMI output.

`rdisk4` didn't work for me (macOS 12.6.1):

```
% gokr-packer -serial_console disabled -overwrite /dev/rdisk4 github.com/gokrazy/hello
gokrazy packer v0.0.0-20221008090041-c87c806562c3 on GOARCH=arm64 GOOS=darwin

Updating gokrazy installation on http://gokrazy

Build target: CGO_ENABLED=0 GOARCH=arm64 GOOS=linux
Build timestamp: 2022-11-20T06:49:52-06:00
Loading system CA certificates from bundled Mozilla CA list
Building 1 Go packages:

  github.com/gokrazy/hello

[done] in 0.40s

Including loadable kernel modules from:
/Users/a/go/pkg/mod/github.com/gokrazy/kernel@v0.0.0-20221117041835-70da94736ca9/lib/modules

Feature summary:
  use GPT: true
  use PARTUUID: true
  use GPT PARTUUID: true
2022/11/20 06:49:53 partitioning /dev/rdisk4 (GPT + Hybrid MBR)
2022/11/20 06:49:53 Using sudo to gain permission to format /dev/rdisk4
2022/11/20 06:49:53 If you prefer, cancel and use: sudo setfacl -m u:${USER}:rw /dev/rdisk4
Password:
2022/11/20 06:49:56 device holds 7935623168 bytes
2022/11/20 06:49:56 write /dev/rdisk4: invalid argument
2022/11/20 06:49:56 exit status 1
```

`disk4` did:

```
% gokr-packer -serial_console disabled -overwrite /dev/disk4 github.com/gokrazy/hello
gokrazy packer v0.0.0-20221008090041-c87c806562c3 on GOARCH=arm64 GOOS=darwin

Updating gokrazy installation on http://gokrazy

Build target: CGO_ENABLED=0 GOARCH=arm64 GOOS=linux
Build timestamp: 2022-11-20T06:54:04-06:00
Loading system CA certificates from bundled Mozilla CA list
Building 1 Go packages:

  github.com/gokrazy/hello

[done] in 0.40s

Including loadable kernel modules from:
/Users/a/go/pkg/mod/github.com/gokrazy/kernel@v0.0.0-20221117041835-70da94736ca9/lib/modules

Feature summary:
  use GPT: true
  use PARTUUID: true
  use GPT PARTUUID: true
2022/11/20 06:54:05 partitioning /dev/disk4 (GPT + Hybrid MBR)
2022/11/20 06:54:05 Using sudo to gain permission to format /dev/disk4
2022/11/20 06:54:05 If you prefer, cancel and use: sudo setfacl -m u:${USER}:rw /dev/disk4
2022/11/20 06:54:05 device holds 7935623168 bytes
2022/11/20 06:54:05 Re-reading partition table failed: gokrazy is currently missing code for re-reading partition tables on your operating system. Please see the README at https://github.com/gokrazy/tools for alternatives, and consider contributing code to fix this. Remember to unplug and re-plug the SD card before creating a file system for persistent data, if desired.

Creating boot file system
[creating boot file system]
Kernel directory: /Users/a/go/pkg/mod/github.com/gokrazy/kernel@v0.0.0-20221117041835-70da94736ca9
EEPROM update summary:
  pieeprom.upd (sig 76909e1d6f)
  recovery.bin
  vl805.bin (sig 6246230ecd)
[done] in 13.42s, 70 MiB
MBR summary:
  LBAs: vmlinuz=51781 cmdline.txt=142965
  PARTUUID: 2e18c40c

Creating root file system
[done] in 0.23s
If your applications need to store persistent data, unplug and re-plug the SD card, then create a file system using e.g.:

	mkfs.ext4 /dev/disk/by-partuuid/60c24cc1-f3f9-427a-8199-2e18c40c0004

To boot gokrazy, plug the SD card into a Raspberry Pi 3 or 4 (no other models supported)

Build complete!

To interact with the device, gokrazy provides a web interface reachable at:

	http://gokrazy:QKGueSl2WKoP62dcJvx8@gokrazy/

In addition, the following Linux consoles are set up:

	1. foreground Linux framebuffer console on HDMI

WARNING: if the above URL does not work, perhaps name resolution (DNS) is broken
in your local network? Resolving your hostname failed: lookup macbog.local: no such host
Did you maybe configure a DNS server other than your router?
```

stapelberg, 2022-11-20:
Okay, here you go: https://t.zekjur.net/2022-11-20-elias-full.img.bz2

This image was created using (without any replace directives):
```shell
% gokr-packer \
  -target_storage_bytes=1258299392 \
  -overwrite=/tmp/elias-full.img  \
  -serial_console=disabled \
  github.com/gokrazy/fbstatus \ 
  github.com/gokrazy/hello \
  github.com/gokrazy/serial-busybox
```

I have copied this image to an SD card on macOS using `sudo dd if=elias-full.img of=/dev/rdisk5 bs=5M`.

Because the image is 1.2G (uncompressed) and needs to be written in full, this will take comparatively long (30 seconds). When using the packer directly, it only writes what it needs (≈ 50 MB), which is much faster.

I verified this SD card boots in my Raspberry Pi Zero 2 W with HDMI connected.

Let me know if this works.

eliasnaur, 2022-11-20:
Thank you! Your image also doesn't boot further than the rainbow. In contrast, the SeedSigner image does boot and shows its console on HDMI.

That points to the disabled wifi/bluetooth somehow affecting the gokrazy boot process. I'll have to dig more.

Thank you for your help!

stapelberg, 2022-11-20:
Are you 100% sure that you have a Raspberry Pi Zero **2**? I’m asking because the seedsigner website talks about the Raspberry Pi Zero version 1.

The crucial difference is that the Pi Zero 2 is arm64, whereas the Pi Zero 1 is armhf.

If you have a Pi Zero 1, you can try using the community-supported Raspberry Pi Foundation kernel instead: https://gokrazy.org/platforms/#community

eliasnaur, 2022-11-20:
OMG, yes it's a Raspberry Pi Zero v1.1, not a Zero 2. I'm very, very sorry for my wasting of your time!

The reason for my confusion is that because of the supply shortage, recent seedsigner devices are built on RPi Zero 2 with radios disabled. My excuse is that I bought my device directly from SS recently, and assumed it was a RPi Zero 2.

I'll try the community kernel.

closed 2022-11-20 by eliasnaur

stapelberg, 2022-11-20:
No worries :)

I’m glad we could get to the bottom of this.

eliasnaur, 2022-11-20:
For posterity, this command worked on the RPi Zero 1:

```
$ GOARCH=arm gokr-packer -kernel_package=github.com/gokrazy-community/kernel-rpi-os-32/dist -overwrite /dev/disk4 github.com/gokrazy/hello
```